espressif/esp-matter #1: Failed to resolve compile flags for chip.c

aSaintjusteLAB714, 2022-06-21:
**Environment**

Module or chip used: [ESP32-WROOM-32-E]
IDF version :  v4.4.1
Build System: [idf.py]

**Compiler version**
xtensa-esp32-elf: esp-2021r2-patch3-8.4.0
riscv32-esp-elf: esp-2021r2-patch3-8.4.0
esp32ulp-elf: 2.28.51-esp-20191205
cmake: 3.20.3 
openocd-esp32:  v0.11.0-esp32-20211220
ninja: 1.10.2

**Operating System**: [Window]
**Using an IDE?**: [Yes]

**Problem Description**
Build failed for  light example in the latest esp-matter version.

**Detailed problem description** 
After pulling the git repo and installing esp idf version 4.4.1 I am unable to build the light example project. I have tried build other example projects in the esp-matter project and those also have failed with the same error. 

**Expected Behavior**
Default light example build should pass

**Actual Behavior**
Build failing due to  'Failed to resolve compile flags for %s', 'C:/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/chip.c'

**Steps to Reproduce**
step 1: Install the toolchain with version esp-idf 4.4.1 from git repository
step 2: Clone the esp-matter repo, make sure to have use recursive to get the submodules
step 3: Set the build target to esp32
step 4: Build the project

**Build log**:
```
C:\esp-matter\examples\light>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\esp-matter\examples\light\build
Executing "ninja all"...
[2/674] Generating args.gn
FAILED: esp-idf/chip/args.gn
cmd.exe /C "cd /D C:\esp-matter\examples\light\build\esp-idf\chip && C:\Users\astel\.espressif\python_env\idf4.4_py3.9_env\Scripts\python.exe C:/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/create_args_gn.py C:/esp-matter/examples/light/build C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1 C:/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/chip.c C:/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/chip.cpp C:/esp-matter/examples/light/build/esp-idf/chip/args.gn.in C:/esp-matter/examples/light/build/esp-idf/chip/args.gn --filter-out=-DHAVE_CONFIG_H;-DIDF_VER=\\\"v4.4.1\\\";-Iconfig"
Traceback (most recent call last):
  File "C:\esp-matter\connectedhomeip\connectedhomeip\config\esp32\components\chip\create_args_gn.py", line 70, in <module>
    c_flags = get_compile_flags(args.c_file)
  File "C:\esp-matter\connectedhomeip\connectedhomeip\config\esp32\components\chip\create_args_gn.py", line 51, in get_compile_flags
    raise Exception("Failed to resolve compile flags for %s", src_file)
Exception: ('Failed to resolve compile flags for %s', 'C:/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/chip.c')
[4/674] Performing build step for 'bootloader'
[1/97] Generating project_elf_src_esp32.c
[2/97] Building C object esp-idf/hal/CMakeFiles/__idf_hal.dir/mpu_hal.c.obj
[3/97] Building C object esp-idf/hal/CMakeFiles/__idf_hal.dir/wdt_hal_iram.c.obj
[4/97] Building C object esp-idf/hal/CMakeFiles/__idf_hal.dir/cpu_hal.c.obj
[5/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/adc_periph.c.obj
[6/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/soc_include_legacy_warn.c.obj
[7/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/lldesc.c.obj
[8/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/gpio_periph.c.obj
[9/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/dac_periph.c.obj
[10/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/i2s_periph.c.obj
[11/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/interrupts.c.obj
[12/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/i2c_periph.c.obj
[13/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/lcd_periph.c.obj
[14/97] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/cpu_util.c.obj
[15/97] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_init.c.obj
[16/97] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_clk_init.c.obj
[17/97] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/compare_set.c.obj
[18/97] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/rtc_pm.c.obj
[19/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/ledc_periph.c.obj
[... 62 lines cut ...]
[82/97] Linking C static library esp-idf\log\liblog.a
[83/97] Linking C static library esp-idf\esp_rom\libesp_rom.a
[84/97] Linking C static library esp-idf\esp_common\libesp_common.a
[85/97] Linking C static library esp-idf\xtensa\libxtensa.a
[86/97] Linking C static library esp-idf\esp_hw_support\libesp_hw_support.a
[87/97] Linking C static library esp-idf\esp_system\libesp_system.a
[88/97] Linking C static library esp-idf\efuse\libefuse.a
[89/97] Linking C static library esp-idf\bootloader_support\libbootloader_support.a
[90/97] Linking C static library esp-idf\spi_flash\libspi_flash.a
[91/97] Linking C static library esp-idf\micro-ecc\libmicro-ecc.a
[92/97] Linking C static library esp-idf\soc\libsoc.a
[93/97] Linking C static library esp-idf\hal\libhal.a
[94/97] Linking C static library esp-idf\main\libmain.a
[95/97] Linking C executable bootloader.elf
[96/97] Generating binary image from built executable
esptool.py v3.3-dev
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated C:/esp-matter/examples/light/build/bootloader/bootloader.bin
[97/97] cmd.exe /C "cd /D C:\esp-matter\examples\light\build\bootloader\esp-idf\esptool_py && C:\Users\astel\.espressif\python_env\idf4.4_py3.9_env\Scripts\python.exe C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/esp-matter/examples/light/build/bootloader/bootloader.bin"
Bootloader binary size 0x62c0 bytes. 0xd40 bytes (12%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

shubhamdp, 2022-06-22:
Basically issue is with the paths, paths on windows and linux/mac are different and comparison [here](https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/create_args_gn.py#L48) is failing.

I know how to fix this specific issue, but not sure what we would ran into after fixing this as this was never tested on Windows and we support only Linux and Mac as development hosts.

So, I would like to give this a try on Windows and it would take some time.

aSaintjusteLAB714, 2022-06-22:
@shubhamdp Thank you for you help. Please let me know what you find.

shubhamdp, 2022-06-23:
@aSaintjusteLAB714 If you'd like to give it a try, please apply the attached patch, [compare_paths.txt](https://github.com/espressif/esp-matter/files/8963536/compare_paths.txt).

```
cd path/to/esp-matter/connectedhomeip/connectedhomeip
git apply path/to/compare_paths.txt
```

and try to rebuild the example. This will fix the current error. If you encounter any more error please post them here

shubhamdp, 2022-06-24:
@aSaintjusteLAB714 did you get a chance to try out the patch?

aSaintjusteLAB714, 2022-06-24:
@shubhamdp  I did see that path issue was resolved, but I ran into another issue.

When I attempted to build the project it failed.
I see that in the console output that it was unable to load the "build_overrides/build.gn" file.
When I open "build_overrides" file it contain " ../../examples/build_overrides".

```
C:\esp-matter\examples\light>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\esp-matter\examples\light\build
Executing "ninja all"...
[29/1297] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
sec_cert,63,6,0xd000,12K,
nvs,data,nvs,0x10000,24K,
otadata,data,ota,0x16000,8K,
phy_init,data,phy,0x18000,4K,
ota_0,app,ota_0,0x20000,1920K,
ota_1,app,ota_1,0x200000,1920K,
fctry,data,nvs,0x3e0000,24K,
*******************************************************************************
[629/1297] Performing configure step for 'chip_gn'
FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure
cmd.exe /C "cd /D C:\esp-matter\examples\light\build\esp-idf\chip && C:\gn-windows-amd64\gn.exe --root=C:/esp-matter/connectedhomeip/connectedhomeip/config/esp32 gen --check --fail-on-unused-args C:/esp-matter/examples/light/build/esp-idf/chip && C:\Users\astel\.espressif\tools\cmake\3.20.3\bin\cmake.exe -E touch C:/esp-matter/examples/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure"
ERROR at //.gn:15:1: Unable to load "C:/esp-matter/connectedhomeip/connectedhomeip/config/esp32/build_overrides/build.gni".
import("//build_overrides/build.gni")
^-----------------------------------
[631/1297] Performing configure step for 'bootloader'
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.32.0.windows.2")
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/astel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/astel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/astel/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/esp-matter/examples/light/sdkconfig
-- Adding linker script C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/soc/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32 esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/bootloader C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/bootloader_support C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/efuse C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/esp32 C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/esp_common C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/esp_hw_support C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/esp_rom C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/esp_system C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/esptool_py C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/freertos C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/hal C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/log C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/bootloader/subproject/main C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/bootloader/subproject/components/micro-ecc C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/newlib C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/partition_table C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/soc C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/spi_flash C:/Users/astel/.espressif/frameworks/esp-idf-4.4.1/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/esp-matter/examples/light/build/bootloader
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

chshu, 2022-06-30:
@aSaintjusteLAB714 I'm afraid that Matter SDK doesn't support Windows build environment yet.

As documented in esp-matter docs: https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#host-setup, as well as CHIP upstream setup guide: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md, only Linux and MacOS are supported.

aSaintjusteLAB714, 2022-07-01:
@chshu Thank you for you assistance. I will be trying this on a MacOS

dhrishi, 2022-07-10:
>Will there be support for windows in the future?

Not sure. If it is supported then the upstream connectedhomeip documentation will be updated accordingly.

I will close this issue now

closed 2022-07-10 by dhrishi