espressif/esp-matter #2: chip-tool配网时报错

sunhuil, 2022-07-12:
使用idf-4.4.1分支
相关日志如下：
sevan@sevan-vm:out$ ./chip-tool pairing ble-wifi 0x7283 HiWiFi_2CC364 nekteck2014 20202021 3840
[1657608648.092822][8874:8874] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1657608648.092995][8874:8874] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1657608648.093048][8874:8874] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1657608648.093079][8874:8874] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1657608648.093225][8874:8874] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-xQ87fb)
[1657608648.093555][8874:8874] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1657608648.093641][8874:8874] CHIP:DL: NVS set: chip-counters/reboot-count = 3 (0x3)
[1657608648.093852][8874:8874] CHIP:DL: Got Ethernet interface: ens33
[1657608648.093940][8874:8874] CHIP:DL: Found the primary Ethernet interface:ens33
[1657608648.094026][8874:8874] CHIP:DL: Failed to get WiFi interface
[1657608648.094042][8874:8874] CHIP:DL: Failed to reset WiFi statistic counts
[1657608648.094069][8874:8874] CHIP:IN: UDP::Init bind&listen port=0
[1657608648.094101][8874:8874] CHIP:IN: UDP::Init bound to port=48842
[1657608648.094107][8874:8874] CHIP:IN: UDP::Init bind&listen port=0
[1657608648.094116][8874:8874] CHIP:IN: UDP::Init bound to port=34510
[1657608648.094119][8874:8874] CHIP:IN: BLEBase::Init - setting/overriding transport
[1657608648.094121][8874:8874] CHIP:IN: TransportMgr initialized
[1657608648.094130][8874:8874] CHIP:FP: Initializing FabricTable from persistent storage
[1657608648.094176][8874:8874] CHIP:TS: Last Known Good Time: 2022-07-12T14:36:08
[1657608648.094387][8874:8874] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x49791F832B287728, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1657608648.094493][8874:8874] CHIP:FP: Fabric index 0x2 was retrieved from storage. Compressed FabricId 0xAACB5FE89D69EC0C, FabricId 0x0000000000000002, NodeId 0x000000000001B669, VendorId 0xFFF1
[1657608648.094609][8874:8874] CHIP:FP: Fabric index 0x3 was retrieved from storage. Compressed FabricId 0xFD732189D1939D1B, FabricId 0x0000000000000003, NodeId 0x000000000001B669, VendorId 0xFFF1
[1657608648.095595][8874:8874] CHIP:ZCL: Using ZAP configuration...
[1657608648.095946][8874:8874] CHIP:CTL: System State Initialized...
[1657608648.095970][8874:8874] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1657608648.095980][8874:8874] CHIP:CTL: Setting attestation nonce to random value
[1657608648.095991][8874:8874] CHIP:CTL: Setting CSR nonce to random value
[1657608648.096022][8874:8874] CHIP:IN: UDP::Init bind&listen port=5550
[1657608648.096036][8874:8874] CHIP:IN: UDP::Init bound to port=5550
[1657608648.096040][8874:8874] CHIP:IN: UDP::Init bind&listen port=5550
[1657608648.096047][8874:8874] CHIP:IN: UDP::Init bound to port=5550
[1657608648.096050][8874:8874] CHIP:IN: TransportMgr initialized
[1657608648.096067][8874:8874] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1657608648.096071][8874:8874] CHIP:CTL: Setting attestation nonce to random value
[1657608648.096075][8874:8874] CHIP:CTL: Setting CSR nonce to random value
[1657608648.097026][8874:8874] CHIP:CTL: Generating NOC
[1657608648.097412][8874:8874] CHIP:FP: Validating NOC chain
[1657608648.097819][8874:8874] CHIP:FP: NOC chain validation successful
[1657608648.097881][8874:8874] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1657608648.097899][8874:8874] CHIP:TS: Last Known Good Time: 2022-07-12T14:36:08
[1657608648.097902][8874:8874] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1657608648.097904][8874:8874] CHIP:TS: Retaining current Last Known Good Time
[1657608648.099201][8874:8874] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1657608648.100169][8874:8874] CHIP:TS: Committing Last Known Good Time to storage: 2022-07-12T14:36:08
[1657608648.101050][8874:8874] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1657608648.101078][8874:8874] CHIP:IN: UDP::Init bind&listen port=5550
[1657608648.101104][8874:8874] CHIP:IN: UDP::Init bound to port=5550
[1657608648.101107][8874:8874] CHIP:IN: UDP::Init bind&listen port=5550
[1657608648.101116][8874:8874] CHIP:IN: UDP::Init bound to port=5550
[1657608648.101119][8874:8874] CHIP:IN: TransportMgr initialized
[1657608648.101156][8874:8874] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1657608648.101169][8874:8874] CHIP:CTL: Setting attestation nonce to random value
[1657608648.101185][8874:8874] CHIP:CTL: Setting CSR nonce to random value
[1657608648.101426][8874:8874] CHIP:CTL: Generating NOC
[1657608648.101742][8874:8874] CHIP:FP: Validating NOC chain
[1657608648.102099][8874:8874] CHIP:FP: NOC chain validation successful
[1657608648.102154][8874:8874] CHIP:FP: Updated fabric at index: 0x2, Node ID: 0x000000000001B669
[1657608648.102163][8874:8874] CHIP:TS: Last Known Good Time: 2022-07-12T14:36:08
[1657608648.102165][8874:8874] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1657608648.102167][8874:8874] CHIP:TS: Retaining current Last Known Good Time
[1657608648.103147][8874:8874] CHIP:FP: Metadata for Fabric 0x2 persisted to storage.
[1657608648.104017][8874:8874] CHIP:TS: Committing Last Known Good Time to storage: 2022-07-12T14:36:08
[1657608648.105269][8874:8874] CHIP:CTL: Joined the fabric at index 2. Compressed fabric ID is: 0x0000000000000000
[1657608648.105298][8874:8874] CHIP:IN: UDP::Init bind&listen port=5550
[1657608648.105324][8874:8874] CHIP:IN: UDP::Init bound to port=5550
[1657608648.105328][8874:8874] CHIP:IN: UDP::Init bind&listen port=5550
[1657608648.105338][8874:8874] CHIP:IN: UDP::Init bound to port=5550
[... 55 lines cut ...]
[1657608678.124272][8874:8874] CHIP:FP: Forgetting fabric 0x2
[1657608678.124275][8874:8874] CHIP:TS: Pending Last Known Good Time: 2022-07-12T14:36:08
[1657608678.124294][8874:8874] CHIP:TS: Previous Last Known Good Time: 2022-07-12T14:36:08
[1657608678.124297][8874:8874] CHIP:TS: Reverted Last Known Good Time to previous value
[1657608678.124303][8874:8874] CHIP:CTL: Shutting down the commissioner
[1657608678.124313][8874:8874] CHIP:CTL: Shutting down the controller
[1657608678.124324][8874:8874] CHIP:IN: Expiring all connections for fabric 3!!
[1657608678.124327][8874:8874] CHIP:FP: Forgetting fabric 0x3
[1657608678.124329][8874:8874] CHIP:TS: Pending Last Known Good Time: 2022-07-12T14:36:08
[1657608678.124347][8874:8874] CHIP:TS: Previous Last Known Good Time: 2022-07-12T14:36:08
[1657608678.124358][8874:8874] CHIP:TS: Reverted Last Known Good Time to previous value
[1657608678.124364][8874:8874] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1657608678.124388][8874:8874] CHIP:DMG: IM WH moving to [Uninitialized]
[1657608678.124400][8874:8874] CHIP:DMG: IM WH moving to [Uninitialized]
[1657608678.124402][8874:8874] CHIP:DMG: IM WH moving to [Uninitialized]
[1657608678.124404][8874:8874] CHIP:DMG: IM WH moving to [Uninitialized]
[1657608678.124406][8874:8874] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1657608678.124419][8874:8874] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1657608678.124530][8874:8874] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-4fj80d)
[1657608678.124844][8874:8874] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1657608678.124881][8874:8874] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1657608678.124886][8874:8874] CHIP:DL: Inet Layer shutdown
[1657608678.124887][8874:8874] CHIP:DL: BLE shutdown
[1657608678.125049][8874:8874] CHIP:DL: System Layer shutdown
[1657608678.125094][8874:8874] CHIP:TOO: Run command failure: ../third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:244: CHIP Error 0x00000032: Timeout

dhrishi, 2022-07-12:
@sunhuil Please provide details about the setup. Host is Linux/Mac? Device is ESP32 or ESP32-C3 or any other? Also, does this issue happen everytime? From the logs, it looks like a timeout. Please share the logs on the device side too.

dhrishi, 2022-07-25:
@sunhuil Do you have any updates to share?

dhrishi, 2022-07-26:
Closing this due to lack of response. Feel free to re-open with the details.

closed 2022-07-26 by dhrishi

sidhoo-root, 2022-10-28:
i meet this issue too. can u help me
details:

ubuntu 20.04 on vitual meachine
device is esp32c3, 
1. flash the manufacturing binary [https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool#readme](url),
2. run device, log:
>> ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
I (30) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (30) boot: compile time 14:24:49
I (30) boot: chip revision: 3I (37) boot.esp32c3: SPI Mode       : DIO
I (41) boot.esp32c3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (70) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (77) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (85) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (92) boot:  4 ota_0            OTA app          00 10 00020000 001e0000
I (100) boot:  5 ota_1            OTA app          00 11 00200000 001e0000
I (107) boot:  6 fctry            WiFi data        01 02 003e0000 00006000
I (115) boot: End of partition table
I (119) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=2f3f8h (193528) map
I (158) esp_image: segment 1: paddr=0004f420 vaddr=3fc90400 size=00bf8h (  3064) load
I (159) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=fb500h (1029376) map
I (324) esp_image: segment 3: paddr=0014b528 vaddr=3fc90ff8 size=02ad0h ( 10960) load
I (327) esp_image: segment 4: paddr=0014e000 vaddr=40380000 size=1033ch ( 66364) load
I (342) esp_image: segment 5: paddr=0015e344 vaddr=50000010 size=00010h (    16) load
I (348) boot: Loaded app from partition at offset 0x20000
I (348) boot: Disabling RNG early entropy source...
I (363) cpu_start: Pro cpu up.
I (371) cpu_start: Pro cpu start user code
I (371) cpu_start: cpu freq: 160000000
I (371) cpu_start: Application information:
I (374) cpu_start: Project name:     light
I (379) cpu_start: App version:      v1.0
I (383) cpu_start: Compile time:     Oct 28 2022 16:11:32
I (389) cpu_start: ELF file SHA256:  e7c0ddc45aab754c...
I (395) cpu_start: ESP-IDF:          v4.4.2
I (400) heap_init: Initializing. RAM available for dynamic allocation:
I (407) heap_init: At 3FCA46E0 len 0001B920 (110 KiB): DRAM
I (414) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (420) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM0;32mI (427) spi_flash: detected chip: generic
I (432) spi_flash: flash io: dio
I (437) sleep: Configure to isolate all GPIO pins in sleep state
I (442) sleep: Enable automatic switching of GPIO sleep configuration
I (450) coexist: coexist rom version 9387209
I (454) cpu_start: Starting scheduler.
I (467) led_driver_ws2812: Initializing light driver
I (467) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (487) app_main: Light created with endpoint_id 1
E (487) esp_matter_core: Command 0x0047 on cluster 0x0300 already exists. Not creating again.
I (497) pp: pp rom version: 9387209
I (497) net80211: net80211 rom version: 9387209
I (507) wifi:wifi driver task: 3fcb0920, prio:23, stack:6656, core=0
I (507) system_api: Base MAC address is not set
I (517) system_api: read default base MAC address from EFUSE
I (517) wifi:wifi firmware version: eeaa27d
I (527) wifi:wifi certification version: v7.0
I (527) wifi:config NVS flash: enabled
I (527) wifi:config nano formating: disabled
[... 132 lines cut ...]
I (1807) phy: chan:13,max_power:80
I (1807) phy: chan:14,max_power:80
W (1817) wifi:Haven't to connect to a suitable AP now!
I (1817) chip[DL]: Done driving station state, nothing else to do...
I (1827) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name MATTER-2481)
I (1837) chip[DL]: Device already advertising, stop active advertisement and restart
I (1847) NimBLE: GAP procedure initiated: stop advertising.

I (1857) NimBLE: GAP procedure initiated: advertise; 
I (1857) NimBLE: disc_mode=2
I (1867) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (1877) NimBLE: 

I (1877) app_main: Commissioning window opened
I (1877) led_driver_ws2812: led set r:0, g:0, b:0
I (1887) led_driver_ws2812: led set r:0, g:0, b:0
I (1887) led_driver_ws2812: led set r:25, g:20, b:16
> I (31447) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable, device name MATTER-2481)
I (31447) chip[DL]: Device alrea32mI (31457) NimBLE: GAP procedure initiated: stop advertising.

I (31467) NimBLE: GAP procedure initiated: advertise; 
I (31467) NimBLE: disc_mode=2
I (31467) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (31477) NimBLE: 
3. run chip-tool cmd :`./chip-tool pairing ble-wifi 1234 Redmi_huan 87654321 38802273 2481`