espressif/esp-matter #4: [Feature request] PlatformIO integration

Diegorro98, 2022-07-13:
Maybe it's too soon to ask for it, but would be great to have the multiple features that PlatformIO offers to developers, and recently [Espressif collaborates with PlatformIO Labs](https://www.espressif.com/en/news/PlatformIO_ESP_collaboration).

I have tried to migrate the light example to a esp-idf framework based PlatformIO project, but it has been impossible for me. As I don't have too much experience with cmake and ESP-IDF, maybe someone more experienced is capable of migrating an example project.

Diegorro98, 2022-07-14:
I'm to create this issue at [PlatformIO espressif32 repo](https://github.com/platformio/platform-espressif32), because I think that is an issue that should be in both repo.

Massimiliano-solutiontech, 2023-09-25:
Any update on this?

HamzaHajeir, 2024-05-26:
Any updates, so far?