espressif/esp-matter #5: Custom device network possibilities with esp-matter framework

thecashit, 2022-07-21:
Hi there,

1) Are there possible to launch esp-httpd server simultaneously with matter?
2) Are there possible to setup wi-fi credentials code behind?
3) Are there possible to add custom BLE Services and Characteristics simultaneously with stack that esp-matter exposed?

Thnx

chshu, 2022-07-22:
@thecashit  It's definitely possible to integrate Matter with your existing application, both are application tasks based on esp-idf.

> 1. Are there possible to launch esp-httpd server simultaneously with matter?

Yes, Matter runs in a separate task, you can launch esp-httpd in main task or separate task as well.

> 2. Are there possible to setup wi-fi credentials code behind?

Wi-Fi provisioning is part of the normal Matter commissoning process. But if you alreay have your own way to setup Wi-Fi credentials, it's also doable using Matter on-network commissioning method, which commissions a on-network device (already connected to IP network) to Matter fabric.

> 3. Are there possible to add custom BLE Services and Characteristics simultaneously with stack that esp-matter exposed?

[blemesh_bridge](https://github.com/espressif/esp-matter/tree/main/examples/blemesh_bridge) is a good example for your reference, it enables the external platform feature, and customizes the BLE service in addition to the Matter usage.

> Or maybe we can start matter core simultaneously with our current firmware?

It should a single integrated firmware runs on the SoC.

thecashit, 2022-07-27:
@chshu Thank your very much for detailed explanation. 
I was checked information and it is really seems like it it achievable task.
The one thing i cannot find - how to provide Wi-Fi credentials to Matter and get ones after Matter provisioning.
Could you please provide a little more information?

chshu, 2022-07-27:
> how to provide Wi-Fi credentials to Matter and get ones after Matter provisioning.

There maybe some misunderstanding on the word "provide Wi-Fi credentials to Matter".
Say a device need to get the Wi-Fi credentials to connect to Wi-Fi AP / IP network, it can either:
1. use matter commissiong via BLE, once done, the device will get Wi-Fi credentials, connect to the Wi-Fi network, and join the Matter fabric;
2. use some other non-matter method to get Wi-Fi credentials (it's out of Matter scope), then use the Matter on-network commissioning to join the Matter fabric.

thecashit, 2022-07-29:
@chshu  Thank you for clarification.
But while investigate Matter scope i really can't understand 2 things. 
That'se will be great if you will help with that.

Let's see what we will try to achieve.

1) Firmware start and check Wi-Fi hotspots nearby
2) Firmware check saved Wi-Fi credentials list and if there is match should start Matter scope with SSID credentials provided. So - now we have SSID and Password,now we should start Matter with this credentials

Point 1 - i really cannot understand how to start esp_matter and provide SSID/Password to connect to.
You mention that it is available via  Matter on-network commissioning, am i right?
Another guess that i have is to save this credentials to NVS where esp_matter held this data and after that start Matter task

2) If there is no match - we can start Matter without any credentials with built-in commissioning via BLE

3) User can setup device via Matter provisioning or via our internal provisioning (for backward capability at least)

Point 2: if the user provide credentials via Mater BLE scope how can we get this credentials to save in our firmware to use in step 2 described before

Point 3: if the user use our proprietary commissioning and setup wi-fi credentials via non-Matter scope how can we restart Matter with that credentials (similar with Point 1)

I understand that is is unusual questions but it will be excellent if we will be able to made such mechanism.

Thank you

chshu, 2022-08-02:
@thecashit Understood the mechanism you want to achieve, it's indeed a customized use case. A quick answer to your Points  is that all Wi-Fi related operation can be done using ESP's Wi-Fi APIs, Matter application and other non-Matter applicaiton (including you own code logic) could coexist, they all call into Wi-Fi APIs to configure Wi-Fi and register the callback to receive the Wi-Fi events (like connected, disconnected).

The current Matter logic is as follow:
* Matter Application calls ESP's APIs to configure the Wi-Fi connection, ESP SDK handles the Wi-Fi credential storge and reconnection after reboot, each time only one Wi-Fi SSID/Passcode pair is stored. (refer to the code: [wifi connect](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/NetworkCommissioningDriver.cpp#L164))
* For the on-network commissioning case, the other Non-Matter application calls ESP's APIs to provide SSID/Password and connect to Wi-Fi, then an event will trigger the Matter application to start mDNS over IP network, so it can be discovered by commissoner and do the matter on-network commissioning. (refer to the code: [dns start](https://github.com/project-chip/connectedhomeip/blob/master/examples/platform/esp32/common/CommonDeviceCallbacks.cpp#L148)).

To achieve your goal, you need to store Wi-Fi credentials in nvs, and write your own code to do the wifi scan, and choose the one conenct to (this can be done using ESP's nvs and wifi APIs, nothing matter related). But I'm confused on the real use case, since a matter fabric always stay in single local network, normally it should not switch wifi network.

thecashit, 2022-08-04:
@chshu Thank you very much for the explanation, everything became clear.
I'll try to figure it out soon and run some tests.

The use case is quite simple - the user can move the device between different locations, each with its own SSID ... But yes, in this case, most likely Matter will not be able to connect to the controller, since the controllers will be different.

Thanks again for the clarifications. I close the issue, if there are any problems, I will create a new one.

closed 2022-08-04 by thecashit