espressif/esp-matter #10: How to deal with multiple Indentify typedefs

jonsmirl, 2022-07-29:
So if I make two end points like this, Indentify becomes ambiguous.
Should I just make multiple wrappers for Identify or is there a better way to merge these types to a single function?

    dimmable_light::config_t light_config;
    light_config.on_off.on_off = DEFAULT_POWER;
    light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
    endpoint_t *endpoint = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE);

    dimmer_switch::config_t switch_config;
    endpoint_t *endpoint = dimmer_switch::create(node, &switch_config, ENDPOINT_FLAG_NONE);

-----------------
type Identify is ambiguous here...
static void on_identify_trigger_effect(Identify *identify)
{
    ESP_LOGI(TAG, "currenr identify trigger effect id: %d", identify->mCurrentEffectIdentifier);
}

jonsmirl, 2022-07-30:
This is part of the problem:

static void on_identify_trigger_effect(Identify *identify)
{
    ESP_LOGI(TAG, "currenr identify trigger effect id: %d", identify->mCurrentEffectIdentifier);
}
using namespace chip::app::Clusters;

If you declare the namespace after on_identify_trigger_effect() it works. Move it in front and it becomes ambiguous.  I have not dug deep enough into this to understand why it is ambiguous.

using namespace chip::app::Clusters;
static void on_identify_trigger_effect(Identify *identify)
{
    ESP_LOGI(TAG, "currenr identify trigger effect id: %d", identify->mCurrentEffectIdentifier);
}

chiragatal, 2022-08-01:
Ideally the namespace should have been at the top with others. I think this would be a problem of not rebasing before merging.

Regarding the ambiguity, I am not sure where it could be coming from. esp-matter has `identify` namespace for the cluster and attribute, but it is all lower case.

jonsmirl, 2022-08-01:
This name space, 'using namespace chip::app::Clusters;' is also declaring Identity.

Move the that line above the trigger function and you can't compile anymore.

```
/home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/cluster-enums.h:30:11: note: candidates are: 'namespace chip::app::Clusters::Identify { }'
 namespace Identify {
           ^~~~~~~~
In file included from /home/jonsmirl/aosp/esp-matter/components/esp_matter_identify/esp_matter_identify.h:18,
                 from /home/jonsmirl/aosp/esp-matter/examples/jds/main/app_main.cpp:15:
/home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/identify-server/identify-server.h:23:8: note:                 'struct Identify'
 struct Identify
        ^~~~~~~~
/home/jonsmirl/aosp/esp-matter/examples/jds/main/app_main.cpp:35:50: error: 'identify' was not declared in this scope
 static void on_identify_trigger_effect(Identify *identify)
                                                  ^~~~~~~~
/home/jonsmirl/aosp/esp-matter/examples/jds/main/app_main.cpp:35:50: note: maximum limit of 1000 namespaces searched for 'identify'
/home/jonsmirl/aosp/esp-matter/examples/jds/main/app_main.cpp:35:50: note: suggested alternative:
In file included from /home/jonsmirl/aosp/esp-matter/components/esp_matter/esp_matter.h:22,
                 from /home/jonsmirl/aosp/esp-matter/examples/jds/main/app_main.cpp:13:
/home/jonsmirl/aosp/esp-matter/components/esp_matter/esp_matter_attribute.h:194:11: note:   'esp_matter::cluster::identify'
 namespace identify {
           ^~~~~~~~
/home/jonsmirl/aosp/esp-matter/examples/jds/main/app_main.cpp: In function 'void app_main()':
/home/jonsmirl/aosp/esp-matter/examples/jds/main/app_main.cpp:111:30: error: 'on_identify_trigger_effect' was not declared in this scope
                              on_identify_trigger_effect);
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jonsmirl/aosp/esp-matter/examples/jds/main/app_main.cpp:111:30: note: maximum limit of 1000 namespaces searched for 'on_identify_trigger_effect'
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

dhrishi, 2022-08-04:
Can you please confirm if you face issues with the latest code? Please close the issue if you don't.

closed 2022-08-04 by jonsmirl