espressif/esp-matter #6: Switches in esp_matter_endpoint.h missing groups cluster

jonsmirl, 2022-07-29:
namespace dimmer_switch {
typedef struct config {
    cluster::identify::config_t identify;
    cluster::binding::config_t binding;
} config_t;

uint32_t get_device_type_id();
endpoint_t *create(node_t *node, config_t *config, uint8_t flags);
} /* dimmer_switch */

namespace color_dimmer_switch {
typedef struct config {
    cluster::identify::config_t identify;
    cluster::binding::config_t binding;
} config_t;

Should be...
It is also missing scenes, but I am not sure if that is needed.
Groups is required because a switch can be bound to a group of lights.

namespace dimmer_switch {
typedef struct config {
    cluster::identify::config_t identify;
    cluster::groups::config_t groups;
    cluster::binding::config_t binding;
} config_t;

uint32_t get_device_type_id();
endpoint_t *create(node_t *node, config_t *config, uint8_t flags);
} /* dimmer_switch */

namespace color_dimmer_switch {
typedef struct config {
    cluster::identify::config_t identify;
    cluster::groups::config_t groups;
    cluster::binding::config_t binding;
} config_t;

chiragatal, 2022-07-29:
The groups cluster is optional. In the default `<devce_type>::create()` APIs, only the mandatory clusters are added. The endpoint_t handle can then be used to add any optional clusters, for example `groups::create()` in this case.

(This is also true for any clusters and its features; only the mandatory attributes and commands are added.)

chiragatal, 2022-07-29:
Actually, the optional groups cluster is being added currently: https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_endpoint.cpp#L216. Since client clusters don't have any attributes, the config for groups this is not needed in the switch config.

But since this is optional, I think it should be removed from here.

(From this, it seems like there should be some documentation in the header file too, mentioning which clusters are actually being added.)

jonsmirl, 2022-07-29:
The goal is to get this use case working....
https://github.com/project-chip/connectedhomeip/issues/21243

To do that the Switches need to be group clients and the lights group servers.

dhrishi, 2022-07-29:
You can enable any of the optional clusters/attributes/commands as per your usecase in the application code. The provision to handle them would be made in the SDK but they won't be added to the default cluster APIs.

jonsmirl, 2022-07-29:
Nice to have a paragraph in esp-matter docs explaining that you don't differentiate between client and server clusters. In your model the full client and server API is always available and then the linker decides whether to include the methods or not.  This will need to be explained to people coming off from SiLabs where since they will be searching for client/server modes.

jonsmirl, 2022-08-04:
Add the bit about client/server to the docs?

dhrishi, 2022-08-09:
Yes, will see if we can add the documentation at some appropriate location. Closing the original issue as it is addressed.

closed 2022-08-09 by dhrishi