espressif/esp-matter #8: Inclusion of Thread diagnostics in Root node

jonsmirl, 2022-07-29:
This should have IFDEF and be conditional on Thread being enabled.
Similarly for wifi diagnostics.

chiragatal, 2022-07-29:
This is already being done in `root_node::create()`: https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_endpoint.cpp#L77

jonsmirl, 2022-07-29:
it is wasting ram here....

namespace endpoint {
namespace root_node {
typedef struct config {
    cluster::basic::config_t basic;
    cluster::ota_requestor::config_t ota_requestor;
    cluster::general_commissioning::config_t general_commissioning;
    cluster::network_commissioning::config_t network_commissioning;
    cluster::general_diagnostics::config_t general_diagnostics;
    cluster::administrator_commissioning::config_t administrator_commissioning;
    cluster::operational_credentials::config_t operational_credentials;
    cluster::diagnostics_network_wifi::config_t diagnostics_network_wifi;
    cluster::diagnostics_network_thread::config_t diagnostics_network_thread;
} config_t;

chiragatal, 2022-07-29:
Temporarily, yes. The node::config_t is currently a local variable in the examples, and so the memory used is freed up. But yes, it can be optimised further by just adding those ifdefs.

closed 2022-08-04 by jonsmirl