espressif/esp-matter #9: Node is too generic, use RootNode

jonsmirl, 2022-07-29:
These should be called RootNode to follow spec.

    /* Create a Matter node */
    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, NULL);

jonsmirl, 2022-07-29:
Maybe like this?

    /* Create a Matter RootNode endpoint */
    root_node::config_t root_node_config;
    node_t *node = node::create(&root_node_config, app_attribute_update_cb, NULL);

    /* add on/off switch endpoint */
    on_off_switch::config_t switch_config;
    endpoint_t *endpoint = on_off_switch::create(node, &switch_config, ENDPOINT_FLAG_NONE);

chiragatal, 2022-07-29:
Node here is more of a generic term. And when `node::create()` is called, it internally adds the mandatory root_node on endpoint 0.

jonsmirl, 2022-07-29:
Useful to explain the side effect of root_node creation via a comment in the example sources.

chiragatal, 2022-07-29:
Yeah, it makes sense, now that you mention it. It is easy to get confused between node and root_node.

jonsmirl, 2022-07-29:
Need to come up with different names for these various things...
BLE is using node too. Too many uses of the same word.

For a start...
node_t *node = node::get();
to
device_t *device = device::get();

Should that root_node be automatically created? How do you get a pointer to it to add optional clusters?

chiragatal, 2022-08-01:
The root_node is being automatically created just to keep the application code simple and cover most of the use-cases. But if optional clusters are always being added to root_node, we can move the creation of root_node also to the application instead of doing it internally.

With the current implementation, you can get the endpoint handle for the root node with this:
```
uint16_t root_node_endpoint_id = 0;
node_t *node = node::get();
endpoint_t *endpoint = endpoint::get(node, root_node_endpoint_id);
```

dhrishi, 2022-08-25:
Updated the comment in the application to reflect the creation of Root Node device type in create API

closed 2022-08-25 by dhrishi