espressif/esp-matter #11: blemesh example almost working,  proxy issue?

jonsmirl, 2022-07-30:
I have blemesh example all compiled, installed and 98% working on a pair of C3s.

Is it proxying the state inside the bridge? Because I appear to be able to get the bulb and the bridge out of sync in what state they are in.  If proxy is correct I should not be able to get these messages....

I (1427464) EXAMPLE: event 0x00, opcode 0x8202, src 0x0001, dst 0x0005
I (1427464) EXAMPLE: ESP_BLE_MESH_GENERIC_SERVER_STATE_CHANGE_EVT
I (1427464) EXAMPLE: onoff 0x01
W (1427474) BOARD: led red is already on
I (1483794) EXAMPLE: event 0x00, opcode 0x8202, src 0x0001, dst 0x0005
I (1483794) EXAMPLE: ESP_BLE_MESH_GENERIC_SERVER_STATE_CHANGE_EVT
I (1483794) EXAMPLE: onoff 0x01
W (1483804) BOARD: led red is already on
I (1505974) EXAMPLE: event 0x00, opcode 0x8202, src 0x0001, dst 0x0005
I (1505974) EXAMPLE: ESP_BLE_MESH_GENERIC_SERVER_STATE_CHANGE_EVT
I (1505974) EXAMPLE: onoff 0x00

Once I manage to get it back in sync, it works as expected.

InfiniteYuan, 2022-07-31:
From your description, I am still not clear about the problems you encountered in using it.

There are following concept:
1. **BLE Mesh Proxy Feature**: The Proxy feature is used to relay/forward Network PDUs received by a node between GATT and advertising bearers. It is suitable for BLE Mesh Node.
2. **Matter Proxy(Subscription Proxy)**: A subscription proxy is a type of Node that is capable of multiplexing subscriptions from multiple subscribers onto a single subscription to a given publisher.
3. **Matter Bridge**: A Bridge serves to allow the use of non-CHIP IoT devices (e.g. devices on a Zigbee or Z-Wave network, or any other non-CHIP connectivity technology) in a CHIP Fabric, with the goal to enable the consumer to keep using these non-CHIP devices together with their CHIP devices.

In the [blemesh_bridge](https://github.com/espressif/esp-matter/tree/main/examples/blemesh_bridge) example, including BLE Mesh Provisioner Role and Matter Bridge Feature.

jonsmirl, 2022-07-31:
I have the entire example working in general.

What I believe is going on....the state of the light bulb (on/off) is stored in two places: the bulb and a proxy value in the bridge. If everything is working correctly those two states are always in agreement.  But if things aren't working correctly the state in proxy is different than the state in the bulb.  When these states get out of sync (not in agreement) I see the "led red is already on" messages until I manage to get them back in sync.

InfiniteYuan, 2022-08-01:
`W (1483804) BOARD: led red is already on` This log informs that the red LED has been turned on. It happens when continuous control turns on the lights.

closed 2022-08-04 by jonsmirl