espressif/esp-matter #13: USB JTAG on C3, timeout errors

jonsmirl, 2022-07-31:
I can't get the USB JTAG on the C3 to work on my ELF file. Likely because it is 50MB. I keep getting timeout errors.

jonsmirl@ares:~$ ls -l /home/jonsmirl/aosp/esp-matter/examples/jds/build/blemesh_bridge.elf
-rwxrwxr-x 1 jonsmirl jonsmirl 47235876 Jul 30 20:47 /home/jonsmirl/aosp/esp-matter/examples/jds/build/blemesh_bridge.elf
jonsmirl@ares:~$ 

So how do I get rid of all of the internal symbols from the CHIP library?
Any other ideas on how to get JTAG working?

jonsmirl, 2022-08-03:
If you build openocd from source, https://github.com/espressif/openocd-esp32, it will work.
The openocd in IDF 4.4.1 does not work on this large C3 file.

jonsmirl, 2022-08-03:
This needs to be added to the docs for this project. 

You need to make up a version number for bootstrap before it will run....
I randomly picked 9.5, probably some script to do this, don't know what it is.
diff --git a/configure.ac b/configure.ac
index f0324a651..a146f1962 100644
--- a/configure.ac
+++ b/configure.ac
@@ -1,7 +1,7 @@
 # SPDX-License-Identifier: GPL-2.0-or-later
 
 AC_PREREQ(2.64)
-AC_INIT([openocd], [ ],
+AC_INIT([openocd], [9.5 ],
   [OpenOCD Mailing List <openocd-devel@lists.sourceforge.net>])
 AC_CONFIG_SRCDIR([src/openocd.c])
 AC_CONFIG_AUX_DIR([.])

jonsmirl, 2022-08-03:
Maybe you can push out an IDF 4.4.2 with this fixed openocd so other people don't need to spend a day debugging this.

igrr, 2022-08-04:
@jonsmirl have just pushed 4.4.2 yesterday: https://github.com/espressif/esp-idf/tree/v4.4.2, it comes with the newer OpenOCD version (v0.11.0-esp32-20220411).

dhrishi, 2022-08-04:
Thanks @igrr. We will officially move to v4.4.2 soon, for Matter.

dhrishi, 2022-08-05:
Will close this issue as the original problem is resolved

closed 2022-08-05 by dhrishi