espressif/esp-matter #16: blemesh bridge and Sengled (Telink) BLE mesh light

jonsmirl, 2022-07-31:
I'm new to BLE Mesh, can you explain to me how to pair the Sengled bulb?

I'm able to pair and control Sengled from rRF Mesh app. I just need to pick "No OOB" when provisioning. I don't need to enter a key on the Sengled, just click ok to "No OOB'. Then bind with key the app gives me and I can turn it on-off.  I can also use the nRF Mesh app to pair with the Espressif on_off server. ESP does not prompt with OOB. 

When I boot the Espressif on_off app the bridge sees it and pairs with it.
When I boot the Sengled bulb I don't see anything on the bridge.

So the bridge is not see the bulb advertise pairing mode, but the nRF mesh does see it.

So why does the Mesh Bridge not see the unpaired Sengled bulb while the nRF App can see it? Nothing is printed on the console when I put the Sengled in to pairing mode.

InfiniteYuan, 2022-08-01:
There some code about uuid match: [esp_ble_mesh_provisioner_set_dev_uuid_match](https://github.com/espressif/esp-matter/blob/main/examples/blemesh_bridge/main/app_blemesh.c#L638-L643). Please take a look

jonsmirl, 2022-08-01:
The UUID is 0x85ea vs 0xdddd. I set match length to zero and it can now see the bulb.  Is the point of this matching to retrieve different security keys for the device? I have the security key, it is printed on the box, but the device also works without it.

Now I am at composition failing.
I (9883) app_ble_mesh: ble_mesh_config_client_cb, error_code = 0x00, event = 0x00, addr: 0x0008, opcode: 0x8008
I (9893) app_ble_mesh: composition data b40808000300690007000000140400000200030000120112001002100410061007100312041206120712001301130313041307130813710100fe710100ff7101
W (9913) blemesh_bridge: This isn't an unexpected device ...

So I need to learn more to get past this issue.

jonsmirl, 2022-08-01:
I disabled the Composition check and allowed everything into the block. Now I can pair and control these Sengled mesh BLE bulbs.  It would be useful if there was a check on UUID like there is on Composition which prints out the mesh UIDs seen and not matched. That would have pointed me at the UUID issue earlier.

I would recommend adding a section to the blemesh docs explaining about UUIDs and Composition data. Then explain how to change it so that other vendor's product can be accessed.

jonsmirl, 2022-08-04:
I understand now that modifications to UUID and Composition are needed before I can the Sengled bulbs. Once you make those mods the bulbs work. I might suggest that the example app loosen up it constraints to allow more BLE Mesh devices to work (especially the UUID one). Showing how to do that would make the example more useful.

closed 2022-08-04 by jonsmirl