espressif/esp-matter #17: When creating endpoints allow passing in user data

jonsmirl, 2022-08-01:
In our keypad I need to create eight identical dimmable_light endpoints,  eight switch endpoints and eight generic switch endpoints. So when I am iterating in app_driver_attribute_set_defaults I need to be able to map the endpoint to a pointer into my data structure. This will let me pass the same pointer into the triplets of (light, switch, generic_sw) and associate them with each other. 

    endpoint_t *endpoint1 = on_off_switch::create(node, &switch_config, ENDPOINT_FLAG_NONE, (void *)user_data);

    endpoint_t *endpoint = endpoint::get_first(node);
    while (endpoint) {
        uint16_t endpoint_id = endpoint::get_id(endpoint);
        mydata_t *mydata = endpoint::get_userdata(endpoint);
        cluster_t *cluster = cluster::get_first(endpoint);

chiragatal, 2022-08-01:
Is your end goal to just have a user_data or like a led_handle (similar to the current button_handle) and use it in the driver? Or do you have some other requirement/use-case with the user_data as a part of the endpoint?

jonsmirl, 2022-08-01:
When I create 8 identical end points I need some mechanism to tell which copy of my internal data is associated with the end point.  In the existing samples this is not an issue because you aren't creating multiple, identical endpoints - there is a single data structure. 

A handle and user data pointer are basically the same thing.   I'm trying to store a pointer to the data structure I allocated to support each copy of the end point. I've allocated 8 copies of this structure, how do I associate them with the end points? If don't want to use a loop and search my structures for matching endpoint numbers, better to let me store a handle/user_pointer in your endpoint structure.

dhrishi, 2022-08-04:
Thanks! This is useful and will be applicable in many of the usecases. We have added a provision to pass user data to the endpoints

closed 2022-08-04 by dhrishi