espressif/esp-matter #18: What causes these long pauses in Matter commands?

jonsmirl, 2022-08-01:
Turning my BLE Mesh bulb on/off with Matter is taking 5.1 seconds. Turning the same bulb on/off with nRF Mesh app is so fast I can't see a delay. The delay is in the Matter code, not the BLE code.

There is a 3.1 second gap between these messages:
```
I (377033) chip[SC]: Received Sigma3 msg
E (377103) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (380203) chip[IN]: Prepared unauthenticated message 0x3fc97f24 to 0x0000000000000000 (0)  of type 0x40 and protocolId (0, 0) on exchange 2611r with MessageCounter:9712725.
I (380213) chip[IN]: Sending unauthenticated msg 0x3fc97f24 with MessageCounter:9712725 to 0x0000000000000000 at monotonic time: 000000000005CD27 msec
```

And a 1.37 second gap between these two:
```
I (375563) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x0000000000007283
I (376933) chip[IN]: Prepared unauthenticated message 0x3fc97f24 to 0x0000000000000000 (0)  of type 0x31 and protocolId (0, 0) o
```

file:///home/jonsmirl/pause.txt

jonsmirl, 2022-08-01:
I wired my ATECC608B into a WROOM but it doesn't want to work. Maybe it is bad, I have not used it in five years so I ordered some new ones.  I tried to get examples/peripherals/secure_element/atecc608_ecdsa working. If I'm reading things correctly this chip support works with MBED TLS so Matter will pick it up automatically once it is working. If you can test on your end before my new parts arrive that would be useful.

jonsmirl, 2022-08-02:
I set my ESP32 to use 240Mhz clock.  Now it runs in 2.6 seconds. This really needs to be under one second to make consumers happy. Hopefully when my ATECC608B arrives it will be enough to get down to one second.  

I still think one second is still ridiculously slow for an IOT device. The encryption being used in Matter is excessive.

jonsmirl, 2022-08-02:
How do I get the idf to build a release version of CHIP in esp-matter?

jonsmirl, 2022-08-02:
This is not as dire as I though it was. It is accurate for the first interaction with the device, if you keep the session open to the device subsequent interactions are faster.

Please adjust the documentation to reflect using chip-tool in interactive mode:
./chip-tool interactive start

Also explain that first time session is made it takes about 5 seconds, then if you keep the session open subsequent command run in 1.1 to 1.5 seconds. This needs to be explained to ensure customers come to the right conclusion on ESP Matter performance.

jonsmirl, 2022-08-03:
Chip tool prints timestamps on its log messages, you can just read it off from the screen.

In interactive mode, to toggle bridged BLE light:
ESP32 240Mhz -- 1.14 seconds
ESP32C3 -- 1.52 seconds

No bridge
ESP32C3 -- 1.2 seconds

These may not be totally accurate, I'm not sure what chip-tool is doing for 0.8 seconds before it sends the first packet. This performance is a lot better, but I would still like to see it faster. Consumers are not going to be happy until it is under one second.

dhrishi, 2022-08-03:
I see, for multiple attempts, the time required is around 0.3 sec for the light example on ESP32-C3 (using the same chip-tool timestamp prints)

jonsmirl, 2022-08-03:
I would not be surprised if the 1.3s I was see in the evening yesterday was 2.4Ghz saturation around me from people streaming TV. I am get 0.3 this morning when everyone has gone to work.

dhrishi, 2022-08-03:
This is an experiment worth doing

jonsmirl, 2022-08-03:
I was getting these messages on the ESP last night, none this morning,
```
I (14979567) wifi:bcn_timout,ap_probe_send_start
```

dhrishi, 2022-08-04:
This message is seen when STA misses the beacon. Generally, it can happen when the AP is far away or sometimes when the Wi-Fi network is crowded

jonsmirl, 2022-08-04:
I don't see any huge problem now with encryption speed. If you wanted to reduce that initial 5 seconds I suspect an ATECC608B would help.  The other bug references documenting how to properly test this.

closed 2022-08-04 by jonsmirl