espressif/esp-matter #19: Missing features in light example after commissioning at Home Assistant

Diegorro98, 2022-08-02:
I've been testing commissioning and controlling devices in Home Assistant, the [given examples](https://nabucasa.github.io/matter-example-apps/) given by Nabu Casa works, but flashing the light example from this repo, results in missing features on the device.

This how the given example looks like: (It has Product name, vendor name, hardware version, and the light control)
[image]

And this is how the example from this repo looks like (Product name, vendor name, hardware version have placeholder values and there isn't any control for the light)
[image]

How can placeholder values be changed? And about the controls, maybe the example doesn't report correctly some information so that home assistant doesn't know that is a light, so I think that this is a problem.

dhrishi, 2022-08-03:
@Diegorro98 If you are trying with the esp-matter `light` example then it creates a `color temperature light` device type which is currently not supported with Home Assistant. See here: https://developers.home.google.com/matter/supported-devices

You can simply change the example code to make the light as a `dimmable light` (similar to the upstream lighting example you've mentioned above). With that, ideally the Home Assistant should be able to identify the device. Can you please give it a try and let us know if it helps?

`
    /* Create a Matter node */
    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, NULL);

    dimmable_light::config_t light_config;
    light_config.on_off.on_off = DEFAULT_POWER;
    light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
    endpoint_t *endpoint = dimmable_light::create(node, &light_config, ENDPOINT_FLAG_NONE);

    /* These node and endpoint handles can be used to create/add other endpoints and clusters. */
    if (!node || !endpoint) {
        ESP_LOGE(TAG, "Matter node creation failed");
    }

    /* Initialize identify */
`

@shubhamdp Can you please help about the 2nd part of the question: Changing the product name, vendor name, hardware version etc.

Diegorro98, 2022-08-03:
I tried the changes you suggested me, at first didn't work. But after erasing the flash and clean build the example it worked!
I can control the light and the bright. Thank you!

dhrishi, 2022-08-03:
@Diegorro98 Glad to know. Erasing flash is required as the commissioning with the correct device type needs to happen again.

I will still keep this issue open until we have a resolution on the 2nd part.

shubhamdp, 2022-08-03:
For second part i.e setting device instance information:

Can you please apply [0001-Set-device-instance-info-provider-to-read-data-from-.txt](https://github.com/espressif/esp-matter/files/9253988/0001-Set-device-instance-info-provider-to-read-data-from-.txt). We'll also release this patch sooner.

Enable `CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER` and `CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER` sdkconfig options and rebuild and flash the firmware.

To set the configs you have specified, please use [generate_esp32_chip_factory_bin.py](https://github.com/project-chip/connectedhomeip/blob/master/scripts/tools/generate_esp32_chip_factory_bin.py) script to generate the factory partition. Please check `-h/--help` for more options.

Below is the sample how you can use this script. Please make sure the VID/PID for the test certs that you will be using matches with the one you'll be providing in `--vendor-id` and `--product-id`  arguments.
```
./generate_esp32_chip_factory_bin.py -p 20202021 -d 3840 \
    --dac-cert connectedhomeip/credentials/test/attestation/Chip-Test-DAC-FFF2-8001-0008-Cert.der \
    --dac-key connectedhomeip/credentials/test/attestation/Chip-Test-DAC-FFF2-8001-0008-Key.der \
    --pai-cert connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.der \
    --cd connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der \
    --vendor-id 0xFFF2 --product-id 0x8001 \
    --product-name "Light Switch" --vendor-name "Espressif Systems" \
    --hw-ver 3 --hw-ver-str "ESP32C3-DevKitM1"
```

Now, flash the generated `factory_partition.bin` at deafult nvs partition address.
```
esptool.py -p (PORT) write_flash 0x1000 path/to/factory_partition.bin
```

dhrishi, 2022-08-05:
@Diegorro98 Did you get a chance to try it out

Diegorro98, 2022-08-06:
Hi! Sorry for the delay, I've been disconnected for some days and I'm testing it.
First, I should mention that flash address given in this comment

> Now, flash the generated factory_partition.bin at deafult nvs partition address.
> ```
> esptool.py -p (PORT) write_flash 0x1000 path/to/factory_partition.bin
> ```
maybe is wrong, because I tried and the output at the console was the following:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```
I checked `partitions.csv` at the example and the nvs partition seems to have 0x10000 offset, so I specified to write on that flash address after erasing the whole flash and flashing the example just in case, and it seemed to work. 

But now I'm facing another issue that doesn't let me test, at the console, I receive some error logs about app_qrcode:
```
E (2299) chip[DL]: Long dispatch time: 340 ms, for event type 2
E (2299) app_qrcode: Couldn't get discriminator: Error Device Layer:0x000002
E (2309) app_qrcode: Error getting payload -1
E (2319) app_qrcode: Error getting short manual code text
E (2329) app_qrcode: Error getting long manual code text
E (2339) app_qrcode: Error getting qrcode text
```
and if y wrote this command at the console, I receive errors too:
```
> matter onboardingcodes ble
QRCode:            E (40850) chip[SVR]: GetCommissionableDataProvider()->GetSetupPasscode() failed: 2d
I (40850) chip[SVR]: *** Using default EXAMPLE passcode 20202021 ***
E (40860) chip[SVR]: GetCommissionableDataProvider()->GetSetupDiscriminator() failed: 201
E (40860) chip[SVR]: GetPayloadContents() failed: 201
```
I've doing some testing and seems to be related to the mentioned sdkconfigs options. If I go back to the defaults, the problem is solved.

Diegorro98, 2022-08-06:
Sorry, I completely forgot to apply the changes at `esp_matter_core.cpp`. I just tested with them, there are still errors, but now are different. At boot:
```
E (2336) chip[DL]: Long dispatch time: 326 ms, for event type 2
E (2336) app_qrcode: Couldn't get setup_pin_code: Error CHIP:0x0000002D
E (2346) app_qrcode: Error getting payload -1
E (2366) app_qrcode: Error getting short manual code text
E (2376) app_qrcode: Error getting long manual code text
E (2376) app_qrcode: Error getting qrcode text
```

Running command at console:
```
> matter onboardingcodes ble
QRCode:            E (130206) chip[SVR]: GetCommissionableDataProvider()->GetSetupPasscode() failed: 2d
I (130206) chip[SVR]: *** Using default EXAMPLE passcode 20202021 ***
MT:634J042C00KA0648G00
QRCodeUrl:         E (130216) chip[SVR]: GetCommissionableDataProvider()->GetSetupPasscode() failed: 2d
I (130236) chip[SVR]: *** Using default EXAMPLE passcode 20202021 ***
https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A634J042C00KA0648G00
ManualPairingCode: E (130246) chip[SVR]: GetCommissionableDataProvider()->GetSetupPasscode() failed: 2d
I (130256) chip[SVR]: *** Using default EXAMPLE passcode 20202021 ***
34970112332
Done
```

Although that, it worked and I have been able to set product name, vendor name and hardware version and its visible at Home Assistant!

I'm going to leave the issue open just in case that the error logs are something that need to be solved, if you prefer to close it because its already known or because whatever reason, feel free to close the issue!

shubhamdp, 2022-08-08:
Specs mentions few mandates on how to store the passcode and verifier on the device and that has not been implemented yet. And not storing passcode in the device is also a valid case and it that we have implemented.

So the error logs you posted are not an actual issue.

closed 2022-08-08 by Diegorro98

Diegorro98, 2022-12-19:
@shubhamdp I can't do this anymore because it requires spake2p to be added at the PATH, I found that this tool is located at `connectedhomeip/src/tools/spake2p`, I tried to add it just like that to the PATH but it didn't work. I think I should build it but I don't know how to.
I've seen that since https://github.com/project-chip/connectedhomeip/commit/09704264d1dcf49e8bb30409aa19783e0423c548 commit exists a python implementation, and later, at https://github.com/project-chip/connectedhomeip/commit/1a56789d5f343e490d909ad1f943460b62175ad8, you implemented this in `generate_esp32_chip_factory_bin.py`. However these commits are not yet at the Matter submodule in this repository.

Should I build spake2p, apply these commits or wait to this commits to be released?

shubhamdp, 2022-12-19:
You will have to build `spake2p`, guide for using factory data is mentioned here: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/factory_data.md#dependency

Diegorro98, 2022-12-19:
Thank you so much!

shubhamdp, 2022-12-19:
@Diegorro98 This building host tools will be done through `install.sh` script, changes are up and they will be out soon.