espressif/esp-matter #27: Sending commands

jonsmirl, 2022-08-06:
Why do you trigger this update when nothing has been changed?
```
static void app_driver_button_toggle_cb(void *arg)
{
    ESP_LOGI(TAG, "Toggle button pressed");
    uint16_t endpoint_id = switch_endpoint_id;
    uint32_t cluster_id = OnOff::Id;
    uint32_t command_id = OnOff::Commands::Toggle::Id;

    g_cluster_id = cluster_id;
    g_command_id = command_id;
    lock::chip_stack_lock(portMAX_DELAY);
    client::cluster_update(endpoint_id, cluster_id);
    lock::chip_stack_unlock();
}
```
and then end up in this routine with the problem of passing g_cluster/g_command
```
void app_driver_client_command_callback(client::peer_device_t *peer_device, uint16_t remote_endpoint_id,
                                        void *priv_data)
{
    /** TODO: Find a better way to get the cluster_id and command_id.
    Once done, move the console commands to esp_matter_client. */
    if (g_cluster_id == OnOff::Id) {
        if (g_command_id == OnOff::Commands::Off::Id) {
            on_off::command::send_off(peer_device, remote_endpoint_id);
        } else if (g_command_id == OnOff::Commands::On::Id) {
            on_off::command::send_on(peer_device, remote_endpoint_id);
        } else if (g_command_id == OnOff::Commands::Toggle::Id) {
            on_off::command::send_toggle(peer_device, remote_endpoint_id);
        }
    }
}
```
why didn't you just directly send the command?
```
static void app_driver_button_toggle_cb(void *arg)
{
    ESP_LOGI(TAG, "Toggle button pressed");
    on_off::command::send_toggle();
}
```
And then internal to the stack figure out (peer_device, remote_endpoint_id) from the binding table?

chiragatal, 2022-08-08:
Our idea was to reuse the command_send APIs for bound devices (which uses the binding table) and also for the controller use-case. The same flow (cluster_update followed by command_callback) can be used for both.

This is also helpful when the other device is not already connected. The command_callback then happens asynchronously.

jonsmirl, 2022-08-08:
```
static void app_driver_button_toggle_cb(void *arg)
{
    ESP_LOGI(TAG, "Toggle button pressed");
    on_off::command::send_toggle();
}
```
Would go into the library code wait for the library to sort things out -- like connect if needed. app_driver_client_command_callback would be in the library instead of the app. Hide this stuff from the user application.
The app does not need to know how you are achieving this.
```
void app_driver_client_command_callback(client::peer_device_t *peer_device, uint16_t remote_endpoint_id,
                                        void *priv_data)
{
    /** TODO: Find a better way to get the cluster_id and command_id.
    Once done, move the console commands to esp_matter_client. */
    if (g_cluster_id == OnOff::Id) {
        if (g_command_id == OnOff::Commands::Off::Id) {
            on_off::command::send_off(peer_device, remote_endpoint_id);
        } else if (g_command_id == OnOff::Commands::On::Id) {
            on_off::command::send_on(peer_device, remote_endpoint_id);
        } else if (g_command_id == OnOff::Commands::Toggle::Id) {
            on_off::command::send_toggle(peer_device, remote_endpoint_id);
        }
    }
}
```

jonsmirl, 2022-08-08:
And for this... pass in a pointer to the function instead of the command ID. Then you don't need the if-then.
```
void app_driver_client_command_callback(client::peer_device_t *peer_device, uint16_t remote_endpoint_id,
                                        void *priv_data)
{
    /** TODO: Find a better way to get the cluster_id and command_id.
    Once done, move the console commands to esp_matter_client. */
    if (g_cluster_id == OnOff::Id) {
        if (g_command_id == OnOff::Commands::Off::Id) {
            on_off::command::send_off(peer_device, remote_endpoint_id);
        } else if (g_command_id == OnOff::Commands::On::Id) {
            on_off::command::send_on(peer_device, remote_endpoint_id);
        } else if (g_command_id == OnOff::Commands::Toggle::Id) {
            on_off::command::send_toggle(peer_device, remote_endpoint_id);
        }
    }
}
```

jonsmirl, 2022-08-08:
I believe the problem you are running into with g_command_id/g_cluster_id is that they can't live on the stack. So use this
```
on_off::command::send_toggle();
on_off::command::send_setparams(x, ,y ,z);
```
To call into the esp-matter stack and allocate a structure..
```
   struct {
        pointer to command
        pointer to copy of args
        pointer to suspended thread
   }
```
Then inside the esp-matter stack pass this message around in a queue or something waiting for connect/etc to finish. Then when everything is set up you can call the command and delete the struct.  . You may need to suspend the user thread and wait for results from this before waking the user thread back up. The user app does not need to know what is going on.

Another approach would be to return NOT_READY errors from a send() command. And then if you got NOT_READY you'd have to connect/etc first and then redo the send.

closed 2022-10-28 by jonsmirl