espressif/esp-matter #28: toogle in Light

jonsmirl, 2022-08-07:
Doesn't toggle need to look like this to trigger subscription notifications?
```
static void app_driver_button_toggle_cb(void *arg)
{
    ESP_LOGI(TAG, "Toggle button pressed");
    uint16_t endpoint_id = light_endpoint_id;
    uint32_t cluster_id = OnOff::Id;
    uint32_t attribute_id = OnOff::Attributes::OnOff::Id;

    node_t *node = node::get();
    endpoint_t *endpoint = endpoint::get(node, endpoint_id);
    cluster_t *cluster = cluster::get(endpoint, cluster_id);
    attribute_t *attribute = attribute::get(cluster, attribute_id);

    esp_matter_attr_val_t val = esp_matter_invalid(NULL);
    attribute::get_val(attribute, &val);
    val.val.b = !val.val.b;
    attribute::update(endpoint_id, cluster_id, attribute_id, &val);

    lock::chip_stack_lock(portMAX_DELAY);
    client::cluster_update(endpoint_id, cluster_id);
    lock::chip_stack_unlock();
}
```

Diegorro98, 2022-08-07:
Is this the way to trigger subscription notifications?
```
lock::chip_stack_lock(portMAX_DELAY);
client::cluster_update(endpoint_id, cluster_id);
lock::chip_stack_unlock();
```
I want to use in my project to notify changes at the attributes, but I didn't know how to do it

dhrishi, 2022-08-09:
The upstream SDK code takes care of providing the subscriber with the reports and the application does not need to worry about that. In short, the current light example (without any changes) here, does report data when subscribed. Steps to try out:

```
//Commission
$ ./out/debug/chip-tool interactive start
>>> pairing ble-wifi node-id TESTSSID TESTPASSWD 20202021 3840

//Subscribe to on-off attribute
>>> onoff subscribe on-off min-interval max-interval node-id 1

```
After this, if you locally update the on-off attribute on the light using the push button or the CLI, then you will see the subscription report being received by the chip-tool at the min-interval. I have tested this with the current codebase and it seems to work fine. Please give it a try.

Diegorro98, 2022-08-10:
In my case I needed to notify the update avoiding using `attribute::update`, because using that caused an unwanted behavior, so I explored how it worked and I found that `MatterReportingAttributeChangeCallback` is the function that notifies the changes.
I explored how it was called and this the final solution that I achieved:
```
void CallReportingCallback(intptr_t closure) {
    MatterReportingAttributeChangeCallback(endpoint_id, clusterId, attributeId);
}

void function_that_triggers_update(void *arg) {
    chip::DeviceLayer::PlatformMgr().ScheduleWork(CallReportingCallback, 0);
}
```
I used a 0 for the second argument because I didn't need send anything, but you can pass a ConcreteAttributePath by casting it to a `intptr_t` and then cast back and use it as an `MatterReportingAttributeChangeCallback` argument.

I've tried and it works like a charm when subscribing to that attribute, but I would like to ask you, @dhrishi, if this is a good practice and if something else is needed.

dhrishi, 2022-08-11:
Hi @Diegorro98 I didn't clearly understand this -
> In my case I needed to notify the update avoiding using attribute::update, because using that caused an unwanted behavior

Currently, whenever any attribute is updated (through any means) or an event is generated then the subscriber gets a report. Do you mean that you need to explicitly notify the update without updating the attributes? If yes, I don't know if that will be possible

jonsmirl, 2022-08-11:
This may be the problem.... when you first set up the subscription do you get an immediate status reply? It should do that so that you know the beginning state of the attribute.  Technically that is a status change -- the attribute went form a state of unknown to a known state.

dhrishi, 2022-08-11:
>when you first set up the subscription do you get an immediate status reply?

Yes

Diegorro98, 2022-08-11:
> Do you mean that you need to explicitly notify the update without updating the attributes or generating an event?

Yes, that's what I need. But it is not that the attribute isn't updated, the attribute is already updated but not by `attribute::update`. 

I'm going to try to explain what I'm doing and why I need it in my project:
I'm doing some testing with devices like window covering or barrier control, I know that they are not currently on the esp-matter data model, but I wrote the code by my own.
So, when I run the commands to set the position, at [`window-covering-server.cpp`](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/window-covering-server/window-covering-server.cpp) the application sets the position attribute to the requested position (in [`barrier-control-server.cpp`](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/barrier-control-server/barrier-control-server.cpp) doesn't but I modified to replicate the functionality from window covering because I thought that the cluster was unfinished because it didn't do anything at attribute callback set at `app_main.cpp`) so the driver movement has to be set on the attribute update callback.
Then, if I want to report the current position as the ZCL specification requests, which is different from the target position, I can't use `attribute::update` because it would change the target position to the current position.
And as I manage the attribute reading callback I can give the current position, avoiding using `attribute::update`. But when the position changes and it should be reported, I can't notify it. So that is the reason why I need this.

Other way to resolve this could be to override the command callback, so I can avoid using the attribute update callback and I can use for reporting a change, but I found a way that maybe wasn't the cleanest; using `esp_matter::command::destroy` and then `esp_matter::command::create` with the desired callback.

jonsmirl, 2022-08-11:
The window covering cluster has different attributes for target and current position.

Hrishikesh...   I don't think the ZCL in the tree matches the spec (I can't see the spec anymore).  The ZCL appears to be missing some fields indicating current and target positions.   zcl/data-model/chip/window-covering.xml

    
    <attribute side="server" writable="false" code="0x000B" define="WC_TARGET_POSITION_LIFT_PERCENT100THS"        type="Percent100ths" min="0" max="10000"  isNullable="true" optional="true" reportable="true" >TargetPositionLiftPercent100ths</attribute>
    
    <attribute side="server" writable="false" code="0x000C" define="WC_TARGET_POSITION_TILT_PERCENT100THS"        type="Percent100ths" min="0" max="10000"  isNullable="true" optional="true" reportable="true" >TargetPositionTiltPercent100ths</attribute>

    
    <attribute side="server" writable="false" code="0x0003" define="WC_CURRENT_POSITION_LIFT"                     type="INT16U"   min="0x0000" max="0xFFFF" isNullable="true" optional="true">CurrentPositionLift</attribute>
    
    <attribute side="server" writable="false" code="0x0004" define="WC_CURRENT_POSITION_TILT"                     type="INT16U"   min="0x0000" max="0xFFFF" isNullable="true" optional="true">CurrentPositionTilt</attribute>

PSONALl, 2022-08-28:
@Diegorro98 according to matter specification once the movement of window cover has successfully finished, the current position attribute should be updated and there is no mention of notifying the attribute 
Also, the current position attribute's value is not changed during movement, It just gets updated after the movement according to spec, but if you want to notify a current position attribute then you can use functionality you have implemented

dhrishi, 2022-09-02:
As the original question from @jonsmirl and the issue from @Diegorro98 is addressed, can we close this?