espressif/esp-matter #29: Replace broken device API

jonsmirl, 2022-08-07:
Suppose my device dies and I buy another identical device. In my current home system you would commission the new device and then use the "Replace Device" button to copy the state of the broken device into the new device.  So how do you save the state of a Matter device so that you can restore it later?  The key part here is saving multiple fabrics and device IDs and then restoring them. But there is a lot of other state - bindings, scenes, subscriptions, default values, etc. All per fabric.

dhrishi, 2022-08-11:
>The key part here is saving multiple fabrics and device IDs and then restoring them.

Where do you save the state? Matter talks only about local control and hence there may be some devices that do not have internet connectivity at all.

jonsmirl, 2022-08-11:
A controller would save the state. ie the app with the "Replace Device" button. 

I have this button in my existing home automation system and I have used it at least 30 times to replace dead devices. Consider how much of a pain it is to replace a dead device without it. You will have to manually recommission back onto each fabric and then individually rebuild the device state for each fabric. Very user unfriendly. 

Since Matter has punted on this, the vendor apps have to implement it.

jonsmirl, 2022-08-11:
I also use that button when my devices are behaving flaky.  I manually factory reset them, then use the 'Replace Device' button to restore their state.  I think this is undoing corruption in the flash.

jonsmirl, 2022-08-11:
Since Matter has punted on this, the vendor apps have to implement it.  ... as in Espressif makes me a nice vendor API I can call to get the device state as a blob, and another API which I can use to restore it.  And some way to know if it has changed since last save call.

dhrishi, 2022-08-22:
If a device is a part of multiple fabrics, then this implies that the device will have to share information of the other fabrics (like operational cert, private key) to the controller of some other fabric, which doesn't seem right?

jonsmirl, 2022-08-22:
The blob the device passes back can be encrypted. 

The people designing Matter are laser focused on the "Alexa, turn the light off" use case with banker level security.  They are not paying any attention to the problems they are causing the owners of the devices.  For example they have ignored scenes which are probably the number one feature in a real home automation system.  And one of my big complaints, they are ignoring 3-way light switches which is probably a fifth of all light switches. They are ignoring the fact that devices die and have to be replaced. I can list a hundred things they are ignoring. And they put WAY too much effort into security.

Multi-admin is not a management system, it just creates a VPN for each controller allowing each controller to completely ignore any other controllers. Consider the impact of this on the home owner. The home owner design scenes, but given the way multi-admin is designed those scenes can't be shared among admins. The home owner is going to have to reenter them into each controller.  What about automations?  Again they can't be shared.

Your objection is correct, the problem is that Matter has not planned for use cases as simple as "Replace Device".

Here's another question for you. Has Matter tested opening a commissioning window and going through a minute of setup transactions while another controller is hitting the same device with normal commands?

dhrishi, 2022-08-22:
>The blob the device passes back can be encrypted.

I had thought of this. But then again, which key do we use for encryption? If it is specific to the device, then how and with whom do we share this?

>Here's another question for you. Has Matter tested opening a commissioning window and going through a minute of setup transactions while another controller is hitting the same device with normal commands?

This can be tested

jonsmirl, 2022-08-22:
The only key you have available is the key for the controller doing the backup.

dhrishi, 2022-08-23:
So, the same problem, multiple ecosystem data with the controller now.

dhrishi, 2022-08-26:
As per the discussion above, I don't see a solution to this for now.

closed 2022-08-26 by dhrishi