espressif/esp-matter #32: Multiple lights exposed by BLE bridge

jonsmirl, 2022-08-07:
If my switch is on the same device with the BLE Bridge, and I group the lights and then bind the switch to the group, is it going to work? Or is it going to hit the loopback problem. It is easy to merge the BLE bridge and light_switch example code to set this up for testing.

dhrishi, 2022-08-23:
If it is not facilitated in the upstream SDK then it won't work as we don't have any custom changes on top of it. I think, currently, it will fail.

jonsmirl, 2022-08-23:
I think the spec has a big blank spot on bindings between bridge nodes. I don't believe anyone has considered it.

This going to make consumers with lots of Zigbee devices unhappy.

dhrishi, 2022-08-25:
I will close this now as the original question is answered

closed 2022-08-25 by dhrishi