espressif/esp-matter #36: install.sh failed (didn't find openssl installation)

jkl2e, 2022-08-15:
Mac OS 12.4

> sh install.sh 
> 
> Running Matter Setup
> 
> -e 
>   WELCOME TO...
> 
> -e 
>         █
>         █
>     ▄   █   ▄                                 █     ▌
>     ▀▀█████▀▀     ▄▓▀▀▀▄,▄▀▀▀▀▄   ╓▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀  ,▄▀▀▀▀▄    ▄▀▀
>   ▀█▄       ▄█▀   █     █     █  ▐▌      █    █     █   .█▄▄▄▄▄▄█⌐ ▐▌
>     ▀█▄   ▄█▀     █     █     █  ▐▌      █    █     █    █         ▐▌
>  ▄██▀▀█   █▀▀██▄  █     █     █   ╙▀▄▄▄Φ▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀^  ▐▌
> ▀▀    █   █    ▀▀
> 
> -e   BOOTSTRAP! Bootstrap may take a few minutes; please be patient.
> 
> Downloading and installing packages into local source directory:
> 
>   Setting up CIPD package manager...done (2.7s)
>   Setting up Python environment.....[-]
> ERROR at //build/config/linux/pkg_config.gni:116:17: Script returned non-zero exit code.
>     pkgresult = exec_script(pkg_config_script, args, "value")
>                 ^----------
> Current dir: /Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/
> Command: python3 /Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/build/config/linux/pkg-config.py openssl
> Returned 1.
> stderr:
> 
> Package openssl was not found in the pkg-config search path.
> Perhaps you should add the directory containing `openssl.pc'
> to the PKG_CONFIG_PATH environment variable
> No package 'openssl' found
> Could not run pkg-config.
> 
> See //src/crypto/BUILD.gn:53:3: whence it was called.
>   pkg_config("openssl_config") {
>   ^-----------------------------
> See //BUILD.gn:87:7: which caused the file to be included.
>       "${chip_root}/src/crypto",
>       ^------------------------
> ['gn', 'gen', '/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']
> 
> Traceback (most recent call last):
>   File "/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 307, in install_packages
>     subprocess.check_call(gn_cmd,
>   File "/usr/local/Cellar/python@3.9/3.9.2_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
>     raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 793, in <module>
>     sys.exit(main())
>   File "/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 785, in main
>     return EnvSetup(**vars(parse())).setup()
>   File "/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 457, in setup
>     result = step(spin)
>   File "/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 606, in virtualenv
>     if not virtualenv_setup.install(
>   File "/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 339, in install
>     install_packages(gn_target)
>   File "/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 314, in install_packages
>     raise subprocess.CalledProcessError(err.returncode, err.cmd,
> subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/johannes/esp32/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

But openssl installed on my mac

> openssl version
> LibreSSL 2.8.3

jkl2e, 2022-08-15:
switching opensell version causes the same error...

> openssl version
> OpenSSL 1.1.1j  16 Feb 2021

dhrishi, 2022-08-16:
@jkl2e Can you please check if the steps [here](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#installing-prerequisites-on-macos) help you with your issue?

dhrishi, 2022-08-23:
@jkl2e Do you have any updates to share?

dhrishi, 2022-08-25:
Closing this now. Feel free to re-open if you still face the issue

closed 2022-08-25 by dhrishi

svenehlers, 2023-03-03:
I got a similar error and trying to fix it by following the suggested steps. I will report if these helped.

I am using Ubuntu 22.04 with WSL on WIndows 10.

> WELCOME TO...
> 
>          █
>          █
>      ▄   █   ▄                                █     █
>      ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
>    ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █
>      ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █
>   ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █
>  ▀▀    █   █    ▀▀
> 
>   BOOTSTRAP! Bootstrap may take a few minutes; please be patient.
> 
> Downloading and installing packages into local source directory:
> 
>   Setting up CIPD package manager...done (1m40.7s)
>   Setting up Python environment.....[\]
> ERROR at //build/config/linux/pkg_config.gni:116:17: Script returned non-zero exit code.
>     pkgresult = exec_script(pkg_config_script, args, "value")
>                 ^----------
> Current dir: /mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/
> Command: python3 /mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/build/config/linux/pkg-config.py openssl
> Returned 1.
> stderr:
> 
> Could not run pkg-config.
> 
> See //src/crypto/BUILD.gn:84:3: whence it was called.
>   pkg_config("openssl_config") {
>   ^-----------------------------
> See //BUILD.gn:152:7: which caused the file to be included.
>       "${chip_root}/src/crypto",
>       ^------------------------
> ['gn', 'gen', '/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']
> 
> Traceback (most recent call last):
>   File "/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 335, in install_packages
>     subprocess.check_call(
>   File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
>     raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['gn', 'gen', '/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 857, in <module>
>     sys.exit(main())
>   File "/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 849, in main
>     return EnvSetup(**vars(parse())).setup()
>   File "/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 501, in setup
>     result = step(spin)
>   File "/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 661, in virtualenv
>     if not virtualenv_setup.install(
>   File "/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 370, in install
>     install_packages(gn_target)
>   File "/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 343, in install_packages
>     raise subprocess.CalledProcessError(
> subprocess.CalledProcessError: Command '['gn', 'gen', '/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/mnt/c/users/ma28/Desktop/XXX/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

vishesh-varma, 2023-08-27:
Same Issue, Using WSL

vishesh-varma, 2023-08-27:
#282 fixes everything!