espressif/esp-matter #37: BLE bridge using mesh advertising

jonsmirl, 2022-08-18:
The BLE bridge example is advertising using the BLE mesh protocol.  Matter setup apps can't see the advertisement since they are looking for a normal BLE advertisement.   How is this supposed to work? Can I use normal BLE and mesh simultaneously? Or do I need to be in normal BLE mode while commissioning and then start the mesh later?

jonsmirl, 2022-08-18:
I rolled back this commit and it advertises in nomral mode, but it still won't pair.
d9416458a3660b725ec572a32cfe015725ea4c4b

InfiniteYuan, 2022-08-18:
Can you provide the commit id of the esp-matter you are using?

jonsmirl, 2022-08-18:
I was on d380670056ca9320a9860c2a128eb41cc037669a before I rolled back. I just went forward and picked up a CHIP update. I will rebuild again to incorporate the new chip.

jonsmirl, 2022-08-18:
i am on e0c4b5db757b4daa70c2b5d09791de2430d437d2 now. It is still a BLE mesh advertising packet.

InfiniteYuan, 2022-08-18:
```
E (971) CHIP[DL]: Failed to set random address err: 524
E (981) chip[DL]: Disabling CHIPoBLE service due to error: Error CHIP:0x000000AC
E (991) chip[DL]: Long dispatch time: 303 ms, for event type 2
``

I found the error log here.

InfiniteYuan, 2022-08-18:
Please test it again with [blemesh_bridge.diff.zip](https://github.com/espressif/esp-matter/files/9367970/blemesh_bridge.diff.zip)

jonsmirl, 2022-08-18:
I applied the patch, deleted build, rebuilt. It is a BLE Mesh advertisement, not a normal BLE type. chip-tool can't see BLE Mesh advertising packets since it filters them out.  Look in the log, it is advertising in mesh mode... I (1456) CHIP[DL]: ESP_BLE_MESH_START_BLE_ADVERTISING_COMP_EVT, index 0, err_code 0
```
I (896) app_ble_mesh: ESP_BLE_MESH_PROVISIONER_ADD_LOCAL_APP_KEY_COMP_EVT, err_code 0
I (906) app_ble_mesh: ESP_BLE_MESH_PROVISIONER_BIND_APP_KEY_TO_MODEL_COMP_EVT, err_code 0
I (916) app_ble_mesh: BLE Mesh Provisioner initialized
I (1426) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name MATTER-3840)
I (1426) NimBLE: GAP procedure initiated: advertise; 
I (1426) NimBLE: disc_mode=2
I (1426) NimBLE:  direct_addr_type=0 direct_addr=
I (1436) NimBLE: 00:00:00:00:00:00
I (1436) NimBLE:  adv_channel_map=7 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (1446) NimBLE: 

I (1456) CHIP[DL]: ESP_BLE_MESH_START_BLE_ADVERTISING_COMP_EVT, index 0, err_code 0
I (1466) chip[DL]: CHIPoBLE advertising started
E (1466) chip[DL]: Long dispatch time: 829 ms, for event 
```

jonsmirl, 2022-08-18:
Run the nRF Connect app on your phone, then it is easy to see the difference.
https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_US&gl=US

jonsmirl, 2022-08-18:
Forget all of that, the patch is working.

I forgot to fully erase the flash from the other code I was working on so it booted already commissioned.

dhrishi, 2022-08-24:
The changes have been merged to the main branch now

closed 2022-08-24 by dhrishi