espressif/esp-matter #38: halCommonSetIndexedToken, halCommonGetIndexedToken

jonsmirl, 2022-08-19:
Are these implemented from cluster/scenes/scene.h?
EMBER_AF_PLUGIN_SCENES_USE_TOKENS
To make the OnWithRecallGlobalScene, otherwise it stores the state to RAM and then forgets it.

Not clear if anyone in the Matter world has ever turned this command on.

jonsmirl, 2022-08-19:
https://github.com/project-chip/connectedhomeip/issues/3275

Looks like they are punting on this.

dhrishi, 2022-08-19:
Scenes cluster is marked as provisional for v1.0 Matter release. So there's a chance that it is not supported right now.

jonsmirl, 2022-08-19:
Not implementing halCommonGetToken/halCommonSetToken also disables part of the on/off cluster.
OnWithRecallGlobalScene and OffWithEffect

jonsmirl, 2022-09-03:
Closing since this is a core Matter problem. They are not going to look at this until after 1.0

closed 2022-09-03 by jonsmirl