espressif/esp-matter #39: Lighter Matter repo download

Diegorro98, 2022-08-19:
I have found that if you simply clone this repo, then initiating only the Matter submodule, you can download less Matter submodules by executing this script:
```
$ path/to/esp-matter/connectedhomeip/connectedhomeip/scripts/checkout_submodules.py --shallow --platform esp32
```
Problems might come on Matter bootstrap, because it tries to download all the submodules if you run `bootstrap.sh` from the Matter root folder. If you run it from another folder it will try to download submodules from the repo you are in, if you are. That's why running this repo's `install.sh` doesn't mess up Matter submodules, because bootstrap is launched from this repo's root folder. 

This is a problem with esp-idf extension for vscode because when you install matter from an existing repository, it tries to bootstrap from the matter root folder, so it mess up with submodules. Depending on the answer to this issue I will open an issue into the vscode extension.

**Size comparation:**
|    | Normal | Light |
|----|--------|-------|
| GB |   ~18  |   ~6  |

I have tried to build all the examples present at this repo and all were successfully built.

There is any reason why I should not use this method? Because is an huge difference at download time and space saving.

shubhamdp, 2022-08-19:
@Diegorro98 Please take a look at this documentation, it does tell user to download the lighter version of repository and submodule: https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#getting-the-repositories
This clones all the submodules shallowly.

If you use the `checkout_submodules.py` then it will only pull only ESP32 specific submodules. But, sourcing `bootstrap.sh` will mess it up by cloning the complete history of the remaining submodules.

If you want to use the `checkout_submodules.py` script then after running this script, rather than sourcing bootstrap.sh first and then `activate.sh`, you can source `activate.sh` which will do the bootstrap for you but won't pull the remaining submodules.

Diegorro98, 2022-08-19:
@shubhamdp 
> Please take a look at this documentation, it does tell user to download the lighter version of repository and submodule: https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#getting-the-repositories
This clones all the submodules shallowly.

I know, and that's how I usually download the repo, maybe `git clone --depth 1 --shallow-submodules --recursive` can be faster (I didn't tested) than using `git clone --recursive`, but in terms of disk space, is just ~1 GB difference. So checking out ESP32 submodules only saves much more space that the documented methods. 

> you can source `activate.sh` which will do the bootstrap for you but won't pull the remaining submodules.

I have tried `activate.sh` and it worked.
So, if I'm going to use that Matter repo only for ESP Matter framework, could there be problems in not pulling on the remaining submodules?

shubhamdp, 2022-08-19:
Nope, there won't be any problem and I will make the above suggested changes in the docs and `install.sh` script.

jonsmirl, 2022-08-19:
Once everything is stable and released you won't need to download the Matter repository at all, right? It will just be a library you link to.

closed 2022-10-07 by espressif-bot