espressif/esp-matter #40: Implement device type 0x000F generic switch

jonsmirl, 2022-08-23:
Can you please implement device type 0x000F generic switch? Looks like you already have the Switch Cluster implemented. I would like to test and see if Apple has implemented support for this in IOS.   BTW, there are a lot more device types missing.

    <deviceType>
        <name>MA-genericswitch</name>
        <domain>CHIP</domain>
        <typeName>Matter Generic Switch</typeName>
        <profileId editable="false">0x0103</profileId>
        <deviceId editable="false">0x000f</deviceId>
        <clusters lockOthers="true">
            <include cluster="Switch" client="false" server="true" clientLocked="true" serverLocked="true"></include>
            <include cluster="Fixed Label" client="false" server="false" clientLocked="true" serverLocked="false"></include>
            <include cluster="User Label" client="false" server="false" clientLocked="true" serverLocked="false"></include>
        </clusters>
    </deviceType>

dhrishi, 2022-08-23:
>Can you please implement device type 0x000F generic switch? 

Yes, we will implement that.

>BTW, there are a lot more device types missing.

Currently, the SDK supports a few device types. More device types will be added incrementally.

jonsmirl, 2022-08-25:
If you look in the Matter spec, Generic Switch has very specific events it needs to emit for single click, double click, multi-click etc. To make things easy for Matter implementers, the Espresif button library should emit those exact same events so that there is a 1:1 mapping onto the Matter API. Currently the events are similar, but not exactly the same.

The Espressif button library should also implement rocker switch (this is a switch type in the Matter generic switch spec). A rocker switch does not make contact when idle, then if you push on the top or the bottom it makes contact. Rocker switches have three wires - middle common, top button, bottom button.  Rocker switches are commonly used for window covering controls.

Maybe you can pass the Generic Switch spec over to the IOT Button group and suggest they copy it? Or make an esp-matter specific button library that exactly matches the spec?

jonsmirl, 2022-09-06:
Note that generic switch support has been added to the switch examples in the matter repository.

jonsmirl, 2022-09-07:
This is not critical to me now, I have a work around. Still need to be fixed so that you implement the spec.
```
    dimmer_switch::config_t switch_config;
    endpoint_t *endpoint2 = dimmer_switch::create(node, &switch_config, ENDPOINT_FLAG_NONE, switch_handle);

    cluster::switch_cluster::create(endpoint2, NULL, CLUSTER_FLAG_CLIENT);
    cluster::fixed_label::create(endpoint2, NULL, CLUSTER_FLAG_SERVER);
    cluster::user_label::create(endpoint2, NULL, CLUSTER_FLAG_SERVER);
    endpoint::add_device_type_id(endpoint2, 0x000F);  // Generic switch endpoint
```

PSONALl, 2022-09-07:
@jonsmirl we are already working on a generic switch device type, We will add it to esp-matter soon.

jonsmirl, 2022-09-18:
I see the new generic switch code... doesn't also need...
    cluster::binding::create(endpoint3, &(switch_config.binding), CLUSTER_FLAG_SERVER);
    cluster::fixed_label::create(endpoint3, NULL, CLUSTER_FLAG_SERVER);
    cluster::user_label::create(endpoint3, NULL, CLUSTER_FLAG_SERVER);

PSONALl, 2022-09-27:
> I see the new generic switch code... doesn't also need... cluster::binding::create(endpoint3, &(switch_config.binding), CLUSTER_FLAG_SERVER); cluster::fixed_label::create(endpoint3, NULL, CLUSTER_FLAG_SERVER); cluster::user_label::create(endpoint3, NULL, CLUSTER_FLAG_SERVER);

@jonsmirl you are right! these clusters are not mandatory for generic switch

closed 2022-10-28 by jonsmirl