espressif/esp-matter #41: How do you get multiple controllers onto the same fabric?

jonsmirl, 2022-08-26:
Say I have an iPhone controlling my house, how do I get my wife's iPhone onto the same fabric so that she can also control the devices? Does wife's phone open a commissioning window thus allowing my phone to commission it? If so, then what is the API call to let my wife's phone switch onto this newly commissioned fabric?

Or do I need to pull the fabric keys out of my phone's database and send them to her through some other mechanism that I implement?

Does she become an Admin or something else?

Is this process of getting a second controller onto a fabric described anywhere?

dhrishi, 2022-08-29:
>how do I get my wife's iPhone onto the same fabric so that she can also control the devices?

IMO, your wife's iPhone should have the NOC from the same root of trust. This should be managed by the fabric/ecosystem that you are on. Then, you as the current admin, can configure appropriate ACL privileges on the device(s) to control by adding an entry of the NOC(node id) of your wife's phone. After that, she would be able to control them.

>Does wife's phone open a commissioning window thus allowing my phone to commission it? 

It is the exact opposite way for the multi-admin scenario. The existing administrator opens the commissioning window on the device and then the new administrator can commission the device.

jonsmirl, 2022-08-29:
Can you demo this in your IOS app?

dhrishi, 2022-08-29:
>Can you demo this in your IOS app?

No, not yet. As the app doesn't create/manage an actual fabric of its own. It is based on the upstream CHIP framework.

dhrishi, 2022-08-30:
You can verify this with the host-based chip-tool:

```
Commission using the first commissioner (default commissioner node-id 112233)
$ ./out/debug/chip-tool pairing ble-wifi 135246 TEST_SSID TEST_PASSWD 20202021 3840

Control using the same commissioner - success
$ ./out/debug/chip-tool onoff toggle 135246 1

Try to control through some other controller (node-id 445566) in the same fabric - failure, unsupported access
$ ./out/debug/chip-tool onoff toggle 135246 1 --commissioner-nodeid 445566

Set access control on the device from the first commissioner for the new controller (assigned the privilege as admin here) 
$ ./out/debug/chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [ 112233, 445566 ], "targets": null}]' 135246 0

Try to control through some other controller in the same fabric - success
$ ./out/debug/chip-tool onoff toggle 135246 1 --commissioner-nodeid 445566
```

closed 2022-08-30 by dhrishi