espressif/esp-matter #42: Need help to build a Matter-WiFi controller application as per the requirement

VivekPalle, 2022-08-26:
- We need your clarification for the below application and supported matter sample codes.
We want to develop a Wi-Fi Matter Controller device that can join the existing Matter network at Home with commissioning. It should be able to turn as a controller to control the end devices over matter/through bridges and also should be able to communicate with our cloud platforms and mobile applications to see the end devices status and also to control the end devices.
 
Please help with the step by step approach and sample codes that help us develop this Matter Controller that meets our application requirements.

- We need ESP32-H2 development boards (EVB's) - 5 no's to start working on Matter-Thread applications. Where do we get the same? These were not there online.
We have all other ESP32 and ESP32-C3 modules which support Matter-WiFi, because we have been working on these for years.

dhrishi, 2022-08-26:
>We want to develop a Wi-Fi Matter Controller device that can join the existing Matter network at Home with commissioning.

We have an example for a light_switch (onoff client cluster) that can control a light (onoff server cluster). You can find it [here](https://github.com/espressif/esp-matter/tree/main/examples/light_switch#light-switch). This will serve as a reference for the usecase you are looking for. Are there any specific device types that you want to control? Or will you have all of the possible client clusters on your device to control any end device?

>our cloud platforms and mobile applications to see the end devices status and also to control the end devices.

You can do your application specific stuff after you receive the Wi-Fi station connected event in the application or may be after the [commissioning is complete](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L40).

>control the end devices over matter/through bridges 

Didn't quite understand the bridges part here. Which type of bridge are you talking about? A Matter bridge?

>Where do we get the same?

Please reach out to sales@espressif.com

VivekPalle, 2022-08-29:
Dear Hrishikesh,

Thanks for your reply.

We want to develop a Wi-Fi Matter Controller device that can join the
existing Matter network at Home with commissioning.

We have an example for a light_switch (onoff client cluster) that can
control a light (onoff server cluster). You can find it here
<https://github.com/espressif/esp-matter/tree/main/examples/light_switch#light-switch>.
This will serve as a reference for the usecase you are looking for. Are
there any specific device types that you want to control? Or will you have
all of the possible client clusters on your device to control any end
device?

- Ok. We will workout this example and get back. Our concern is how a newly
joined matter controller takes control over existing matter fabrics.

Yes, we are expecting our matter controller should have all the possible
client clusters. Suggest, if any sample code having all the required matter
clusters.

our cloud platforms and mobile applications to see the end devices status
and also to control the end devices.

You can do your application specific stuff after you receive the Wi-Fi
station connected event in the application or may be after the commissioning
is complete
<https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L40>
.

- Ok. Got it.

control the end devices over matter/through bridges

Didn't quite understand the bridges part here. Which type of bridge are you
talking about? A Matter bridge?

- Yes, I am talking about the Matter Bridge and Thread Border router.
Expecting our matter controller should be able to control all the devices
in the existing matter ecosystem (same network).

Thanks.

VivekPalle, 2022-09-02:
Dear Hrishikesh,

We worked out the light and switch examples and understood how it is functioning.

Now we want to execute our application requirement, by adding a new Matter-Controller (WiFi) in the same local network. This Matter-Controller should be able to listen the control/data commands between the Light and Switch devices and also can be able to control the Light device and able to read the switch state.

Please suggest the example code that works for our above requirement.

Thanks.

dhrishi, 2022-09-02:
>This Matter-Controller should be able to listen the control/data commands between the Light and Switch devices

That isn't possible as the communication between them will be secure. Why do you want to listen? To know what changed (on the light)? In that case, you can subscribe to the attributes of your interest and they will be reported when there's a change.

>and able to read the switch state.

The light_switch example is also a type of minimal Matter controller as it can control devices that have OnOff server cluster. It has OnOff client cluster and does not have a state or attributes.

Basically, the controller:
i. Is a Matter node that is commissioned into the same Matter network/fabric
ii. Has all the client clusters corresponding to the server clusters of the Matter nodes it wants to control
iii. Can subscribe to attributes and events of different Matter nodes and receive reports

Is this the functionality you are looking for?

VivekPalle, 2022-09-05:
@dhrishi Thanks for the inputs. It's clear.

I started debugging the examples and I can able to control the servers. Subscribed to on-off attribute of Light example from the chip-tool, but can't able to see the reporting from the Light to chip-tool when controlled the Light from the Switch example.

(./chip-tool onoff subscribe on-off 1 2 29315 1) 
Here the Switch and Light are bonded with binding command and controlling.

I understood that the chip-tool helps to send the commands and exit immediately. How do I make the chip-tool to run continuously to receive the updates from the subscriptions?

I also want to run the ALL-CLUSTER-APP example as a client to control the supported cluster servers and also subscribe to receive the notifications. I can't find the documentation that helps me go through the command format to send between example codes. But the commands documentation for the chip-tool is very clear to understand.
Can you help to provide the links of documentation that helps to understand the command format and flow.

dhrishi, 2022-09-07:
>Here the Switch and Light are bonded with binding command and controlling.

Is the ACL on light updated too?
>I understood that the chip-tool helps to send the commands and exit immediately. How do I make the chip-tool to run continuously to receive the updates from the subscriptions?

chip-tool has an interactive mode that can be used here. See this for reference:
https://github.com/espressif/esp-matter/issues/28#issuecomment-1208896957

> I also want to run the ALL-CLUSTER-APP example as a client to control the supported cluster servers and also subscribe to receive the notifications. I can't find the documentation that helps me go [...]

all-clusters-app is an app in the upstream connectedhomeip repository and basically has all the cluster servers. Currently, there is no example in upstream connectedhomeip as well as this repository that supports  all the client clusters on ESP32 and acts like a full fledged controller.

VivekPalle, 2022-09-09:
@dhrishi Thanks for your information. Interactive mode is working fine.

It's very clear that, below the tools that we need to use as controllers. These only help us to create the matter fabric, commissioning and interact (control) with the examples.
- Python-based Device Controller
- CHIPTool
- Android CHIPTool
- Darwin CHIPTool

Is it possible to control a matter device from two different matter controllers (Android CHIPTool and Linux CHIPTool)?

> Here the Switch and Light are bonded with binding command and controlling.
> Is the ACL on light updated too?

Yes, it is updated.

dhrishi, 2022-09-13:
>Is it possible to control a matter device from two different matter controllers (Android CHIPTool and Linux CHIPTool)?

@VivekPalle Yes. It is possible. This is a part of multi-admin feature of Matter. You can find more details here: https://github.com/espressif/esp-matter/issues/41

VivekPalle, 2022-09-19:
@dhrishi Thanks for your great support.

Now we are very clear with the controller and server examples and got hands on with the same.

Now we are working on the source code to implement Smart Socket features with metering. We need below support.
- Where do we found the cluster list document that describes cluster syntax, parameters and cluster ID's? We want to add metering related cluster to lighting-app example.
- Do we have any code flow document for these examples from Espressif, that helps us to integrate our application code in a proper way?

dhrishi, 2022-09-27:
@VivekPalle I see these questions in a separate issue. Will close this one.

closed 2022-09-27 by dhrishi