espressif/esp-matter #43: Memory consumption during commissioning

jonsmirl, 2022-08-27:
Where did all of the memory go? The gui task is idle, it is just logging the free memory in a loop.
I had 32,800 free when I started commissioning.

Things weren't so bad until after mDNS started and then 30K disappeared almost instantly when CASE messages started.
Excessive number of packet buffers? TCP window too large?

[commission.log](https://github.com/espressif/esp-matter/files/9437202/commission.log)

jonsmirl, 2022-09-03:
I have not really investigated this, but I am running into issues where the Matter task is not running in time to avoid timeouts. So I suspect the priority of the Matter task should be increased. I have to run all of my tasks at priority 0 to get things to work, and it is still not totally stable. This may be the problem behind that other bug where all of the memory disappeared, maybe the Matter task was not running quickly enough to free things?

Combined and closed
https://github.com/espressif/esp-matter/issues/44

dhrishi, 2022-09-04:
For the memory error `PacketBuffer: pool EMPTY`, please provide more details -
1. Is it with ESP32/ESP32-C3?
2. esp-matter HEAD commit used (as the connectedhomeip submodule is updated a few times recently)
3. Example used or a specific scenario that can help us reproduce this issue
4. Does this happen everytime?

jonsmirl, 2022-09-04:
1) using C3
2) I was on 32ae38aa9506c523a9fb564db6a8eaa981be739e, sept 1
3) I have other task running at priority zero. Those task are not time critical.
4) The typical failure is a timeout, the log was one of the more dramatic failures

What I think is going on is that my tasks are starving the Matter task of time. But my task are not critical, I don't want that to happen. I want the Matter task to have top priority.

Uncomment the graphics demo in this test app and it should happen.  The graphics demo is another task running in a loop.
https://github.com/espressif/esp-matter/issues/48#issuecomment-1235746910

jonsmirl, 2023-03-24:
I believe this has been sorted out now. It was a side effect of running out of memory. FreeRTOS boosts the priority of tasks waiting on semaphores. As I ran out of memory my other tasks got stuck trying to allocate and FreeRTOS kept boosting their priority.