espressif/esp-matter #45: Link gpio's whit multiple clusters.

giandub, 2022-08-28:
The esp32 guide does not explain how to connect multiple clusters to their respective gpio. Thank you.

dhrishi, 2022-08-29:
@giandub Is this for using the led driver? If yes, first you should be using the GPIO led_driver as [here](https://github.com/espressif/esp-matter/blob/main/device_hal/device/esp32_devkit_c/esp_matter_device.cmake#L7)

Then, if you have multiple GPIOs then you can call [this](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_driver.cpp#L152) function from the application with a different led configuration every time. You will get a handle in return which can then be used to make calls to `led_driver_set_power` wherever required.

Similar approach can be followed if you want to use buttons in this way too.

giandub, 2022-08-30:
Ok.thank you.

giandub, 2022-08-30:
Hi, I'm sorry but I'm not able to do that. I tried to edit the following lines, the toggle messages to ids 1 and 2 seem to work but the LED on id 2 is not responding.

In the app_main.cpp:
       Changed this
   /* Initialize driver */
    app_driver_handle_t light_handle[3];
        for (int x = 1; x < 3; x ++){
   light_handle[x]= app_driver_light_init(x);
}
   
  Add this:

endpoint_t *endpoint1 = extended_color_light::create(node, &light_co>
    light1_endpoint_id = endpoint::get_id(endpoint1);

In the app_driver.cpp 

Changed
app_driver_handle_t app_driver_light_init(int light_id)
{
    /* Initialize led */
    led_driver_config_t config = led_driver_get_config(light_id);
    led_driver_handle_t handle = led_driver_init(&config);
    return (app_driver_handle_t)handle;
}

In 
~/esp-matter/device_hal/led_driver/gpio/led_driver.c
Changed 

 int gpiopin[3] = {0,13,12};
  int ledchannel[3] = {2,0,1};

led_driver_config_t led_driver_get_config(int light_id)
{
    led_driver_config_t config = {
        .gpio = gpiopin[light_id],
        .channel = ledchannel[light_id],
    };
    return config;

This command turn  led 1:

> matter esp attribute set 0x1 0x6 0x0 1
I (330075) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (330085) esp_matter_core: strore attribute in nvs: endpoint_id-0x1, cluster_id-0x6, attribute_id-0x0
Done

This command don't turn nothing:

> matter esp attribute set 0x2 0x6 0x0 1
I (358735) esp_matter_attribute: ********** Endpoint 0x0002's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (358745) esp_matter_core: strore attribute in nvs: endpoint_id-0x2, cluster_id-0x6, attribute_id-0x0
Done

Thank you.

dhrishi, 2022-09-02:
@giandub Which ESP32/C3 devkit are you using? Have you physically connected external leds to the GPIOs you have mentioned above?

Also, ideally, you should not be changing the code in the device_hal. Instead, you can change the app_main.cpp to not use `app_driver_light_init` and directly call `led_driver_init` with your configuration each time. It is okay, if you don't use the function `led_driver_get_config` which is for default cases.

giandub, 2022-09-02:
Hi, i have an esp32devkit-c with wroom-32U.
Yes i have connected 2 led on pin 12 and 13.

giandub, 2022-09-02:
Hello, I tried to make changes only app_main.cpp as you suggested. I had the same behavior: the led 2 does not respond.

   /* Initialize driver */
    led_driver_config_t config;
    config.gpio = 12;
    config.channel = 0;
    led_driver_handle_t handle = led_driver_init(&config);
    config.gpio =13;
    config.channel = 1;
    led_driver_handle_t handle1 = led_driver_init(&config);

 I think that the LED drivers initialize correctly because the following function works correctly: 

    /* Starting driver with default values */
   app_driver_light_set_defaults(light_endpoint_id);
 app_driver_light_set_defaults(light1_endpoint_id);

It's probably an error on endpoint configuration. I added this lines to create endpoint id 2:

endpoint_t *endpoint1 = extended_color_light::create(node, &light_config, ENDPOINT_FLAG_NONE, handle1);
    light1_endpoint_id = endpoint::get_id(endpoint1);

Thank you for your help.have a great day.

dhrishi, 2022-09-02:
I connected external LEDs to GPIOs 12 and 13 and with changes in the application, I could get this working. I can control both the LEDs through the `matter esp attribute set` commands

I think this could be missing in your implementation. Comment or remove this `if block`: https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_driver.cpp#L77
By default, it will handle only endpoint 1 (light_endpoint_id). The example assumes only 1 light endpoint and hence the check.

dhrishi, 2022-09-02:
For your reference, I have added a patch (rough set of changes) for the working functionality here: https://gist.github.com/dhrishi/8f70564ce48d89da49c640049a6a6dc4

Please take a look and let me know in case of anything

giandub, 2022-09-03:
Ok, it work...thank you..

closed 2022-09-03 by giandub

surjeetk, 2023-05-07:
Hi All,

I followed the above patch. Only issue I'm facing is **on press of one button, both the LEDs are getting switched on/off instead of only button specific GPIO.** 

Any suggestions please?

Thanks in advance.

dhrishi, 2023-05-08:
@surjeetk Please share your patch on top of a standard Matter example

surjeetk, 2023-05-13:
> @surjeetk , Please assign different channels to the corresponding LEDs in your config [like this](https://gist.github.com/dhrishi/8f70564ce48d89da49c640049a6a6dc4#file-matter_light_multiple_gpio-patch-L72).
> 
> ```
>       led_driver_config_t config_led_1 = {.gpio = 12, .channel = 0};
>       led_driver_config_t config_led_2 = {.gpio = 13, .channel = 1};
> ```
> 
>     * The [led_driver](https://github.com/espressif/esp-matter/blob/main/device_hal/led_driver/gpio/led_driver.c#L24) code uses `led_driver_handle_t led_driver_init(led_driver_config_t *config)` which returns `(led_driver_handle_t)(config->channel + 1);`.
> 
>     * When using the same channel ([or not mentioning channels in the config](https://github.com/surjeetk/docs/blob/master/1b95fcf66793c2ead7df933d8d80eefd7f8f6b95.patch#L102)), both the light endpoints are being essentially passed with the same address of the _priv_data_ (as both EP have the same device type) as led handlers. (i.e. same handler objects passed)
> 
>     * Changing this solves the issue, as then different handler objects will be passed.

It worked by assigning different channels. Thank you @dhairyashah1 and @dhrishi