espressif/esp-matter #46: esp-matter/examples/blemesh_bridge --> Build error 'Failed to resolve component 'esp_adc_cal'

lboue, 2022-08-30:
Hello,

I am facing an issue with esp-matter examplesblemesh_bridge.
I am getty a Build error _'Failed to resolve component 'esp_adc_cal'_ error.

```
pi@RP4:~/esp-matter/examples/blemesh_bridge$ idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/pi/esp-matter/examples/blemesh_bridge/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/pi/esp-matter/examples/blemesh_bridge/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 /home/pi/esp-matter/examples/blemesh_bridge"...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- Component directory /home/pi/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Solving dependencies requirements
Updating lock file at /home/pi/esp-matter/examples/blemesh_bridge/dependencies.lock
Processing 3 dependencies:
[1/3] espressif/button (2.0.1)
[2/3] espressif/qrcode (0.1.0)
[3/3] idf (5.1.0)
-- Found Python3: /home/pi/.espressif/python_env/idf5.1_py3.9_env/bin/python3.9 (found version "3.9.2") found components: Interpreter 
CMake Error at /home/pi/esp/esp-idf/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'esp_adc_cal'.
Call Stack (most recent call first):
  /home/pi/esp/esp-idf/tools/cmake/build.cmake:287 (__build_resolve_and_add_req)
  /home/pi/esp/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  /home/pi/esp/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  /home/pi/esp/esp-idf/tools/cmake/build.cmake:288 (__build_expand_requirements)
  /home/pi/esp/esp-idf/tools/cmake/build.cmake:574 (__build_expand_requirements)
  /home/pi/esp/esp-idf/tools/cmake/project.cmake:434 (idf_build_process)
  CMakeLists.txt:36 (project)

-- Configuring incomplete, errors occurred!
See also "/home/pi/esp-matter/examples/blemesh_bridge/build/CMakeFiles/CMakeOutput.log".
HINT: The component esp_adc_cal has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details.
cmake failed with exit code 1, output of the command is in the /home/pi/esp-matter/examples/blemesh_bridge/build/log/idf_py_stderr_output_3230 and /home/pi/esp-matter/examples/blemesh_bridge/build/log/idf_py_stdout_output_3230
```

I tried to run _idf.py add-dependency esp_adc_cal_  but it doesn't fix this issue.
Could you help me to fix that?

Rerards
Ludovic

InfiniteYuan, 2022-08-31:
OK, Let me try.

lboue, 2022-09-01:
> OK, Let me try.
Did you manage to build blemesh_bridge examples for esp32?

chshu, 2022-09-02:
Could you use IDF v4.4.2 to build blemesh_bridge example? Seems you are using the latest IDF v5.0.

dhrishi, 2022-09-03:
@lboue Did you get a chance to try with ESP-IDF v4.4.2?

lboue, 2022-09-04:
I will try today. I will keep you informed of the result.

lboue, 2022-09-04:
I downloaded the ESP-IDF v4.4.2 archive but it didn't work. Here are the details below:

## Switch to esp-idf 4.4.2
```
cd ~/esp
rm -rf esp-idf/
wget https://github.com/espressif/esp-idf/archive/refs/tags/v4.4.2.zip
unzip v4.4.2.zip
cd esp-idf-4.4.2
./install.sh
. ./export.sh
```

## Building ESP-BLE-MESH - ESP32 bridge example
```
export ESP_MATTER_PATH=/home/pi/esp-matter/connectedhomeip/connectedhomeip/src/platform/ESP32
cd ~/esp-matter/examples/blemesh_bridge
idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/pi/esp-matter/examples/blemesh_bridge/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/pi/esp-matter/examples/blemesh_bridge/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/pi/esp-matter/examples/blemesh_bridge"...
-- Found Git: /usr/bin/git (found version "2.30.2") 
fatal: not a git repository (or any of the parent directories): .git
CMake Error at CMakeLists.txt:27 (include):
  include could not find load file:

    /home/pi/esp-matter/connectedhomeip/connectedhomeip/src/platform/ESP32/device_hal/device/esp32_devkit_c/esp_matter_device.cmake

-- Component directory /home/pi/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Solving dependencies requirements
Updating lock file at /home/pi/esp-matter/examples/blemesh_bridge/dependencies.lock
Deleting 1 unused components
 espressif__button
Processing 2 dependencies:
[1/2] espressif/qrcode (0.1.0)
[2/2] idf (4.4.2)
CMake Error at /home/pi/esp/esp-idf-4.4.2/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'device'.
Call Stack (most recent call first):
  /home/pi/esp/esp-idf-4.4.2/tools/cmake/build.cmake:232 (__build_resolve_and_add_req)
  /home/pi/esp/esp-idf-4.4.2/tools/cmake/build.cmake:509 (__build_expand_requirements)
  /home/pi/esp/esp-idf-4.4.2/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:36 (project)

-- Configuring incomplete, errors occurred!
See also "/home/pi/esp-matter/examples/blemesh_bridge/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

What can I do this time?

Regards

dhrishi, 2022-09-04:
> export ESP_MATTER_PATH=/home/pi/espmatter/connectedhomeip/connectedhomeip/src/platform/ESP32

Why is this executed? The ESP_MATTER_PATH should be the path to the esp-matter repo and not the ESP32 platform in the connectedhomeip submodule

Steps to be followed:

```
$ cd /path/to/esp-idf (v4.4.2)
$ ./install.sh (only required the first time)
$ . ./export.sh (required everytime you open a new terminal)
$ cd /path/to/esp-matter
$ ./install.sh (only required the first time)
$ . ./export.sh (required everytime you open a new terminal - This step assigns the ESP_MATTER_PATH internally)
$ cd examples/blemesh_bridge
$ rm -rf sdkconfig sdkconfig.old build/ (remove stage config, if any)
$ idf.py build (no need to set-target explicitly as default is esp32)
```
Detailed instructions for compilation: https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#developing-with-the-sdk

lboue, 2022-09-04:
> Why is this executed? The ESP_MATTER_PATH should be the path to the esp-matter repo and not the ESP32 platform in the connectedhomeip submodule
>
> Steps to be followed:
>
> ```
> $ cd /path/to/esp-idf [...]

 /home/pi/esp-matter/ is my esp-matter repo :

IDF repo	=> /home/pi/esp-idf/
Matter repo	=> /home/pi/esp-matter/

dhrishi, 2022-09-04:
@lboue You don't need to set the path explicitly. Please follow the steps I've mentioned above and see if the build is successful

jonsmirl, 2022-09-07:
This builds for me without any issues.

lboue, 2022-09-07:
Hello,

This time I a facing an issue with _**nlassert**_ package :
```
pi@RP4:~/esp-matter/examples/blemesh_bridge $ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/pi/esp-matter/examples/blemesh_bridge/build
Executing "ninja all"...
[6/1382] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
sec_cert,63,6,0xd000,12K,
nvs,data,nvs,0x10000,24K,
otadata,data,ota,0x16000,8K,
phy_init,data,phy,0x18000,4K,
ota_0,app,ota_0,0x20000,1856K,
ota_1,app,ota_1,0x1f0000,1856K,
fctry,data,nvs,0x3c0000,24K,
*******************************************************************************
[168/1382] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/pi/esp-matter/examples/blemesh_bridge/sdkconfig
-- Adding linker script /home/pi/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/pi/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/pi/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/pi/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/pi/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/pi/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script /home/pi/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32 esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /home/pi/esp-idf/components/bootloader /home/pi/esp-idf/components/bootloader_support /home/pi/esp-idf/components/efuse /home/pi/esp-idf/components/esp32 /home/pi/esp-idf/components/esp_common /home/pi/esp-idf/components/esp_hw_support /home/pi/esp-idf/components/esp_rom /home/pi/esp-idf/components/esp_system /home/pi/esp-idf/components/esptool_py /home/pi/esp-idf/components/freertos /home/pi/esp-idf/components/hal /home/pi/esp-idf/components/log /home/pi/esp-idf/components/bootloader/subproject/main /home/pi/esp-idf/components/bootloader/subproject/components/micro-ecc /home/pi/esp-idf/components/newlib /home/pi/esp-idf/components/partition_table /home/pi/esp-idf/components/soc /home/pi/esp-idf/components/spi_flash /home/pi/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/esp-matter/examples/blemesh_bridge/build/bootloader
[244/1382] Performing build step for 'bootloader'
[1/97] Generating project_elf_src_esp32.c
[2/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/soc_include_legacy_warn.c.obj
[3/97] Building C object CMakeFiles/bootloader.elf.dir/project_elf_src_esp32.c.obj
[4/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/lldesc.c.obj
[5/97] Building C object esp-idf/hal/CMakeFiles/__idf_hal.dir/mpu_hal.c.obj
[6/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/adc_periph.c.obj
[7/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/dac_periph.c.obj
[8/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/gpio_periph.c.obj
[9/97] Building C object esp-idf/hal/CMakeFiles/__idf_hal.dir/cpu_hal.c.obj
[10/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/i2c_periph.c.obj
[11/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/interrupts.c.obj
[12/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/i2s_periph.c.obj
[13/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/lcd_periph.c.obj
[14/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/ledc_periph.c.obj
[15/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/mcpwm_periph.c.obj
[16/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/pcnt_periph.c.obj
[17/97] Building C object esp-idf/hal/CMakeFiles/__idf_hal.dir/wdt_hal_iram.c.obj
[18/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rmt_periph.c.obj
[... 228 lines cut ...]
has a source file:
  //third_party/connectedhomeip/third_party/nlunit-test/repo/src/nlunit-test.h
which was not found.
___________________
ERROR at //third_party/connectedhomeip/third_party/nlio/BUILD.gn:19:1: Source file not found.
source_set("nlio") {
^-------------------
The target:
[... cut ...]
which was not found.
___________________
ERROR at //third_party/connectedhomeip/third_party/nlio/BUILD.gn:19:1: Source file not found.
source_set("nlio") {
^-------------------
The target:
  //third_party/connectedhomeip/third_party/nlio:nlio
has a source file:
  //third_party/connectedhomeip/third_party/nlio/repo/include/nlbyteorder-big.h
which was not found.
[642/1382] Linking C static library esp-idf/newlib/libnewlib.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

dhrishi, 2022-09-08:
@lboue Can you please confirm if you have followed all the steps mentioned in the documentation? Is your `./install.sh` in esp-matter repo successful? If it is still an issue, please provide details about your environment like the OS, its version etc.

lboue, 2022-09-08:
After several attempts, it works!

```
1381/1382] Generating binary image from built executable
esptool.py v3.3-dev
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
Generated /home/pi/esp-matter/examples/blemesh_bridge/build/blemesh_bridge.bin
[1382/1382] cd /home/pi/esp-matter/examples/blemesh_bridge/build/esp-idf/esptool_py ...ition-table.bin /home/pi/esp-matter/examples/blemesh_bridge/build/blemesh_bridge.bi
blemesh_bridge.bin binary size 0x162a80 bytes. Smallest app partition is 0x1d0000 bytes. 0x6d580 bytes (24%) free.

Project build complete. To flash, run this command:
/home/pi/.espressif/python_env/idf4.4_py3.9_env/bin/python ../../../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x16000 build/ota_data_initial.bin 0x20000 build/blemesh_bridge.bin
or run 'idf.py -p (PORT) flash'
pi@RP4:~/esp-matter/examples/blemesh_bridge $ 
pi@RP4:~/esp-matter/examples/blemesh_bridge $ idf.py flash monitor
Executing action: flash
Serial port /dev/ttyUSB0
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Running ninja in directory /home/pi/esp-matter/examples/blemesh_bridge/build
Executing "ninja flash"...
[1/10] Performing build step for 'chip_gn'
ninja: no work to do.
[2/6] Performing build step for 'bootloader'
[1/1] cd /home/pi/esp-matter/examples/blemesh_bridge/build/bootloader/esp-idf/esptool_py && /home/pi/.espressif/python_env/idf4.4_py3.9_env/bin/python /home/pi/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/pi/esp-matter/examples/blemesh_bridge/build/bootloader/bootloader.bin
Bootloader binary size 0x62c0 bytes. 0xd40 bytes (12%) free.
[3/4] cd /home/pi/esp-matter/examples/blemesh_bridge/build/esp-idf/esptool_py && /ho...ition-table.bin /home/pi/esp-matter/examples/blemesh_bridge/build/blemesh_bridge.bi
blemesh_bridge.bin binary size 0x162a80 bytes. Smallest app partition is 0x1d0000 bytes. 0x6d580 bytes (24%) free.
[3/4] cd /home/pi/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH="/home...lemesh_bridge/build" -P /home/pi/esp-idf/components/esptool_py/run_serial_tool.cmak
esptool.py esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 bootloader/bootloader.bin 0x20000 blemesh_bridge.bin 0x8000 partition_table/partition-table.bin 0x16000 ota_data_initial.bin
esptool.py v3.3-dev
Serial port /dev/ttyUSB0
Connecting.....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:20:cd:44
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00020000 to 0x00182fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00016000 to 0x00017fff...
Compressed 25280 bytes to 15807...
Writing at 0x00001000... (100 %)
Wrote 25280 bytes (15807 compressed) at 0x00001000 in 0.7 seconds (effective 288.0 kbit/s)...
Hash of data verified.
Compressed 1452672 bytes to 920601...
Writing at 0x00020000... (1 %)
Writing at 0x0002bf76... (3 %)
Writing at 0x0003c775... (5 %)
Writing at 0x000463f3... (7 %)
Writing at 0x00051bd8... (8 %)
Writing at 0x00060303... (10 %)
Writing at 0x0006625a... (12 %)
Writing at 0x0006b324... (14 %)
Writing at 0x00070dd3... (15 %)
Writing at 0x0007618b... (17 %)
Writing at 0x0007b770... (19 %)
Writing at 0x0008165a... (21 %)
Writing at 0x00086b7c... (22 %)
Writing at 0x0008cde4... (24 %)
Writing at 0x00092806... (26 %)
Writing at 0x0009811b... (28 %)
Writing at 0x0009df6d... (29 %)
Writing at 0x000a4a70... (31 %)
Writing at 0x000aa8bf... (33 %)
Writing at 0x000b04bc... (35 %)
Writing at 0x000b59dd... (36 %)
Writing at 0x000bb202... (38 %)
Writing at 0x000c1060... (40 %)
Writing at 0x000c7459... (42 %)
Writing at 0x000cd227... (43 %)
Writing at 0x000d35bb... (45 %)
Writing at 0x000d8afc... (47 %)
Writing at 0x000de418... (49 %)
Writing at 0x000e4919... (50 %)
Writing at 0x000e9a94... (52 %)
Writing at 0x000ef324... (54 %)
Writing at 0x000f459f... (56 %)
Writing at 0x000f96ee... (57 %)
Writing at 0x000ff201... (59 %)
Writing at 0x00104870... (61 %)
Writing at 0x00109ee5... (63 %)
Writing at 0x0010f2cd... (64 %)
Writing at 0x00114204... (66 %)
Writing at 0x00119234... (68 %)
Writing at 0x0011e332... (70 %)
Writing at 0x00123872... (71 %)
Writing at 0x00128aab... (73 %)
Writing at 0x0012defa... (75 %)
Writing at 0x00133321... (77 %)
Writing at 0x0013893a... (78 %)
Writing at 0x0013e259... (80 %)
Writing at 0x00143860... (82 %)
Writing at 0x00148bde... (84 %)
Writing at 0x0014e4e7... (85 %)
Writing at 0x00156d0d... (87 %)
Writing at 0x0015f5de... (89 %)
Writing at 0x001659bd... (91 %)
Writing at 0x0016ba96... (92 %)
Writing at 0x001715ec... (94 %)
Writing at 0x00176cc3... (96 %)
Writing at 0x0017c492... (98 %)
Writing at 0x001818e3... (100 %)
Wrote 1452672 bytes (920601 compressed) at 0x00020000 in 21.7 seconds (effective 534.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 159...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (159 compressed) at 0x00008000 in 0.1 seconds (effective 481.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Writing at 0x00016000... (100 %)
Wrote 8192 bytes (31 compressed) at 0x00016000 in 0.1 seconds (effective 908.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

closed 2022-09-08 by lboue