espressif/esp-matter #47: Group multicast is not restored on reboot.

jonsmirl, 2022-08-31:
Using the esp-matter light demo unchanged...
[group.zip](https://github.com/espressif/esp-matter/files/9465087/group.zip)

It is restoring the multicast group on line 149/150
But wifi does not manage to connect to the AP until line 327.
A related question, why is my ESP having so much trouble connecting to the AP? There are several APs in the area with the same SSID.

I have chip-tool interactive session running. I create the group as in the end of this bug report.
https://github.com/project-chip/connectedhomeip/issues/20979
I verified that I can toggle the LED on/off a few time.
Then I rebooted the ESP device
After reboot you get the log from above.
Trying to toggle the light bulb does not generate any output on the ESP device.
I suspect this is because the Multicast group is not there.

jonsmirl, 2022-09-01:
[group2.zip](https://github.com/espressif/esp-matter/files/9465864/group2.zip)

Log from second test.  I can see the packet on Wireshark, but ESP does not see it.

----
# Packet 5 from /tmp/wireshark_wlp2s0NSD9Q1.pcapng
- 828
- 99.352372317
- 2600:4040:51b3:fb00:cbcd:199d:793a:dd0f
- ff35:40:fd00::100:1
- UDP
- 128
- 58055 → 5540 Len=66

jonsmirl, 2022-09-01:
Something about rebooting with multiple APs is breaking the ESP listening for multicast.

jadhavrohit924, 2022-09-01:
I could able to use group multicast after reboot. Also I can see logs on ESP device.
[groups.txt](https://github.com/espressif/esp-matter/files/9469414/groups.txt)

jonsmirl, 2022-09-01:
I have no trouble with multicast where there is single AP nearby.  

The failure is when there is network of APs. In my case the ESP is able to see four or five APs with the same SSID and varying RSSI. Something in the process of deciding which of these AP to connect to seems to break multicast.

dhrishi, 2022-09-01:
This is a known issue in the Wi-Fi code, discussed [here](https://github.com/espressif/esp-idf/issues/8269) and being worked upon.

jonsmirl, 2022-09-01:
There may be layers to this problem. Once they have fixed the wifi code to connect to the strongest AP, you will need to verify that the AP selection process is not breaking that Multicast listen which was set up before the AP was chosen.  From what I am observing it appears that the AP selection process is not preserving the Multicast setup.

jonsmirl, 2022-09-01:
Note that these multi-AP networks are common in US houses now due to Amazon (eero) and Ubiquiti selling home mesh networks. This has to work correctly before ESP Matter devices can be shipped.

https://www.amazon.com/Introducing-eero-mesh-WiFi-system-3-pack-/dp/B07WMLPSRL/ref=sr_1_1_sspa

jonsmirl, 2022-09-02:
How do I tell? They are all on a single IPv4 subnet.  I think they are all on single IPv6 subnet too. They are not routing, they are bridged APs.

jonsmirl, 2022-09-02:
If IPv6 broken on router, Matter should fallback to IPv4.  Normal consumer does not understand this level of problem.

chshu, 2022-09-02:
You may try this: connect PC to one of the APs, and use wireshark to sniffer the packets, then check whether the PC could receivce the multicast message from the chip-tool under another AP.

jonsmirl, 2022-09-02:
It is not simple to tell which AP these devices are attaching to since they all have the same SSID.  But I believe the PC had attached to the one 4ft away from me, and the ESP had attached to the one 100ft away.  All of my APs are not identical, so there could be issue at the one 100ft away. When I changed the SSID of the one 4ft away so that the ESP will use it, multicast works.

Another complication, all the Linux development machines are wired Ethernet desktops.   I need to build CHIP onto a Windows or Mac laptop, or Rasp Pi to be able to move it around.

jonsmirl, 2022-09-02:
I'm in the process of building CHIP on a RaspPi so that I can move it around.   A better solution for testing this would be an Android/IOS phone app.   I run something on my desktop injecting IPv6 multicast packets into the network over Ethernet. And then I walk around with the phone set to wifi mode and make sure I can receive them everywhere.

dhrishi, 2023-04-30:
I will close this as it looks like a generic issue and is added here https://github.com/project-chip/connectedhomeip/issues/22380

closed 2023-04-30 by dhrishi