espressif/esp-matter #48: Failed commission with invalid ssid or pw does not return to BLE advertising mode

jonsmirl, 2022-09-01:
Use the wrong ssid or pw to commission the demo light app. That will correctly fail because the wifi credentials are wrong. Now try to commission it again with the correct ssid/pw. It is not listening on BLE anymore.

dhrishi, 2022-09-02:
This works with the chip-tool in non-interactive mode. After a timeout, the fail-safe timer expires, commissioning window is opened again and BLE advertisement is started. Then you can commission the device with the correct credentials.
While the same thing fails with chip-tool in interactive mode. In that too, after the timeout, ESP32 restarts BLE advertisement. But, the chip-tool is in an incorrect state that it can't perform the commissioning.  If you are not seeing this, then can you please share the logs so that we can take a look?

In short, looks like an issue with the chip-tool in interactive mode?

jonsmirl, 2022-09-02:
Using command line, two commission commands  smirla does not exist, smirlx is correct.
./chip-tool pairing ble-wifi 1 smirla iebgener43 20202021 3840 --ble-adapter 1
./chip-tool pairing ble-wifi 1 smirlx iebgener43 20202021 3840 --ble-adapter 1
these two commands are the fail logs in the attachment.
I erased the flash and then did
./chip-tool pairing ble-wifi 1 smirlx iebgener43 20202021 3840 --ble-adapter 1
Which works as expected.

Note this is not exactly one of your samples, I merged the light sample and the ble-bridge sample together. This is a simple merger to do, there are not conflicts, just copy the pieces together.

[invalid SSID.zip](https://github.com/espressif/esp-matter/files/9479626/invalid.SSID.zip)

dhrishi, 2022-09-02:
Can you please try with the default light example and let me know what do you observe?

jonsmirl, 2022-09-02:
It works ok from light example.   So this is side effect from combining BLE bridge with light. 
Here's my test app, there is very little to it. The GUI piece is disabled so you can ignore that part.
//xTaskCreate(ST7789, "ST7789", 1024*6, NULL, 0, NULL);

[lowpan.zip](https://github.com/espressif/esp-matter/files/9479970/lowpan.zip)
I need BLE Bridge combined with Matter Bulb and Switch working for product. 

I also need this working..
https://github.com/espressif/esp-matter/issues/24

and this one:
https://github.com/espressif/esp-matter/issues/40

dhrishi, 2022-12-04:
@jonsmirl Is the incorrect SSID/passphrase issue resolved now for your app?

dhrishi, 2023-02-22:
I will close this now. Please reopen in case of any updates

closed 2023-02-22 by dhrishi