espressif/esp-matter #50: OnOff OffWithEffect not working

jonsmirl, 2022-09-05:
OffWithEffect does not appear to be working. OnWithTimedOff does appear to work.

OffWithEffect needs to coordinate with LevelControl to do the Fade to off.  Maybe some disconnect between your code and the Matter cluster code?

I tried:
onoff off-with-effect 0 0 1 1
onoff off-with-effect 0 2 1 1
onoff off-with-effect 1 0 1 1

OnWithRecallGlobalScene is not going to work because scenes aren't implemented.

jonsmirl, 2022-09-05:
Somewhere this construct was supposed to be called... 
`
OnOffEffect(
        chip::EndpointId endpoint, OffWithEffectTriggerCommand offWithEffectTrigger,
        chip::app::Clusters::OnOff::OnOffEffectIdentifier effectIdentifier = EMBER_ZCL_ON_OFF_EFFECT_IDENTIFIER_DELAYED_ALL_OFF,

        /*
         * effectVariant's type depends on the effect effectIdentifier so we don't know the type at compile time.
         * Casting to uint8_t for more flexibility since the type can be OnOffDelayedAllOffEffectVariant or
         * OnOffDelayedAllOffEffectVariant
         */
        uint8_t effectVariant = static_cast<uint8_t>(EMBER_ZCL_ON_OFF_DELAYED_ALL_OFF_EFFECT_VARIANT_FADE_TO_OFF_IN_0P8_SECONDS));
`

jonsmirl, 2022-09-05:
If you look in the erf32 lighting-app they have implemented this.
Not sure how this works...
```
OnOffEffect gEffect = {
    chip::EndpointId{ 1 },
    LightMgr().OnTriggerOffWithEffect,
    EMBER_ZCL_ON_OFF_EFFECT_IDENTIFIER_DELAYED_ALL_OFF,
    static_cast<uint8_t>(EMBER_ZCL_ON_OFF_DELAYED_ALL_OFF_EFFECT_VARIANT_FADE_TO_OFF_IN_0P8_SECONDS),
};
```

jonsmirl, 2022-09-05:
That looks like part of the solution, what about...
```
Table 23. Values of the EffectVariant Field of the OffWithEffect Command
EffectIdentifier Field ValueEffectVariant Field ValueDescription
0x000x00 (default)Fade to off in 0.8 seconds
0x01No fade
0x0250% dim down in 0.8 seconds
then fade to off in 12 seconds
0x01
0x03 to 0xffReserved
0x00 (default)20% dim up in 0.5s then fade to
off in 1 second
0x02 to 0xff
0x01 to 0xffReserved
0x00 to 0xffReserved
```

shubhamdp, 2026-10-29:
Closing as a stale, please re-open if required

closed 2026-10-29 by shubhamdp