espressif/esp-matter #51: How do group bindings work?

jonsmirl, 2022-09-06:
I used this binding:
binding write binding '[{"fabricIndex": 1, "group": 1}]' 1 2

0x42013f2e in esp_matter::cluster::level_control::command::send_move (remote_device=0x0, remote_endpoint_id=36532, move_mode=0 '\000', rate=1 '\001', option_mask=0 '\000', option_override=0 '\000') at /home/jonsmirl/aosp/esp-matter/components/esp_matter/esp_matter_client.cpp:169

When it got to line 175 of esp_matter_client.cpp, faults because remote_device is null.
    chip::Controller::LevelControlCluster cluster(*remote_device->GetExchangeManager(), remote_device->GetSecureSession().Value(), remote_endpoint_id);

Can you add group example similar to this Nordic one at the bottom...
https://github.com/project-chip/connectedhomeip/tree/master/examples/light-switch-app/nrfconnect#binding-process

jonsmirl, 2022-09-06:
The architecture of this callback has problems...
```
void app_driver_client_command_callback(client::peer_device_t *peer_device, uint16_t remote_endpoint_id,
                                        void *priv_data)
{
    /** TODO: Find a better way to get the cluster_id and command_id.
    Once done, move the console commands to esp_matter_client. */
    if (g_cluster_id == OnOff::Id) {
        if (g_command_id == OnOff::Commands::Off::Id) {
            on_off::command::send_off(peer_device, remote_endpoint_id);
        } else if (g_command_id == OnOff::Commands::On::Id) {
            on_off::command::send_on(peer_device, remote_endpoint_id);
        } else if (g_command_id == OnOff::Commands::Toggle::Id) {
            on_off::command::send_toggle(peer_device, remote_endpoint_id);
        }
    } else if (g_cluster_id == LevelControl::Id) {
        if (g_command_id == LevelControl::Commands::Move::Id) {
            level_control::command::send_move(peer_device, remote_endpoint_id, g_mode, g_rate, 0, 0);
        }
    }
}
```
That *priv_data should be a function pointer to the command plus the vargs it needs. Then this callback pretty much goes away and becomes generic.  If code needs big if-then like this, something is architecturally wrong.

https://github.com/espressif/esp-matter/issues/27

dhrishi, 2022-09-27:
@jonsmirl We've updated the SDK to support group bindings. Added documentation [here](https://github.com/espressif/esp-matter/tree/main/examples/light_switch#22-bind-a-group-to-switch)

Please give it a try.

wqx6, 2022-09-28:
@jonsmirl The group commands for level control and color control will be added later.

chshu, 2022-11-09:
@jonsmirl  Could you try out the https://github.com/espressif/esp-matter/commit/84db3c810a508b4e793fdce9c3cd14add01afc5a for new group commands, please help to close the issue if it works for you.

dhrishi, 2022-12-06:
Closing this now

closed 2022-12-06 by dhrishi