espressif/esp-matter #54: Aggregator Endpoint

jonsmirl, 2022-09-07:
What is this and how do I use it?

So dimmer has EPs:
1 Dimmable Light
2 Dimmer Switch
3 Generic Switch

Apparently it also needs to say these are pieces of single device.
Aggregator EP  (1, 2, 3)

chshu, 2022-09-08:
Please refer to spec 9.12, Aggregator is a spec device type for bridge device, which has a Descriptor cluster with a PartsList attribute containing all the endpoints representing those Bridged Devices.

The example:

[image]

jonsmirl, 2022-09-08:
Boris telling me to use this in a non-bridge device.  This is a new use of aggregator I am not familiar with.

jonsmirl, 2022-10-27:
This is the actual issue
https://github.com/espressif/esp-matter/issues/73

closed 2022-10-27 by jonsmirl