espressif/esp-matter #55: MATTER_BRIGHTNESS

jonsmirl, 2022-09-08:
In the lighting example, why is MATTER_BRIGHTNESS set to 254? Is this is the spec somewhere?

Apple Home sends out brightness up to 255.

jonsmirl, 2022-09-08:
When Apple home sends 255 it is not setting the Brightness attribute
```
I (76575) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:183370842 on exchange 10962r
I (76585) esp_matter_command: Received command 0x0004 for enpoint 0x0001's cluster 0x00000008
I (76595) chip[ZCL]: 0x3c13e2d0MOVE_TO_LEVEL_WITH_ON_OFF ff  0 0 0
I (76595) chip[IN]: Prepared secure message 0x3fc9991c to 0xD4AD2570BF0490EE (1)  of type 0x9 and protocolId (0, 1) on exchange 10962r with MessageCounter:261734723.
I (76615) chip[IN]: Sending encrypted msg 0x3fc9991c with MessageCounter:261734723 to 0xD4AD2570BF0490EE (1) at monotonic time: 0000000000012B40 msec
I (76625) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:183370843 on exchange 10963r
I (76635) esp_matter_command: Received command 0x0001 for enpoint 0x0001's cluster 0x00000006
I (76645) chip[ZCL]: On/Off set value: 1 1
I (76655) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (76665) chip[ZCL]: On/off already set to new value
I (76665) chip[IN]: Prepared secure message 0x3fc99934 to 0xD4AD2570BF0490EE (1)  of type 0x9 and protocolId (0, 1) on exchange 10963r with MessageCounter:261734724.
I (76685) chip[IN]: Sending encrypted msg 0x3fc99934 with MessageCounter:261734724 to 0xD4AD2570BF0490EE (1) at monotonic time: 0000000000012B87 msec
I (76705) chip[EM]: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:183370844 on exchange 10962r
I (76715) chip[EM]: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:183370845 on exchange 10963r
I (76725) app_main: Free 68856
I (77225) app_main: Free 68856
```
Looks like this when brightness is set correctly...
```
I (67035) app_main: Free 68856
I (67365) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:183370832 on exchange 10958r
I (67365) esp_matter_command: Received command 0x0004 for enpoint 0x0001's cluster 0x00000008
I (67375) chip[ZCL]: 0x3c13e2d0MOVE_TO_LEVEL_WITH_ON_OFF 70  0 0 0
I (67385) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x0000 is 64 **********
I (67395) chip[ZCL]: Setting on/off to 0x3c13e61c due to level change
I (67395) chip[ZCL]: On/Off set value: 1 1
I (67405) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (67415) chip[ZCL]: On/off already set to new value
I (67415) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0xFFFC is 3 **********
I (67435) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x4000 is 1 **********
I (67445) chip[IN]: Prepared secure message 0x3fc9991c to 0xD4AD2570BF0490EE (1)  of type 0x9 and protocolId (0, 1) on exchange 10958r with MessageCounter:261734714.
I (67455) chip[IN]: Sending encrypted msg 0x3fc9991c with MessageCounter:261734714 to 0xD4AD2570BF0490EE (1) at monotonic time: 000000000001077C msec
E (67475) chip[DL]: Long dispatch time: 110 ms, for event type 3
I (67475) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:183370833 on exchange 10959r
I (67495) esp_matter_command: Received command 0x0001 for enpoint 0x0001's cluster 0x00000006
I (67495) chip[ZCL]: On/Off set value: 1 1
I (67505) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (67515) chip[ZCL]: On/off already set to new value
I (67525) chip[IN]: Prepared secure message 0x3fc99934 to 0xD4AD2570BF0490EE (1)  of type 0x9 and protocolId (0, 1) on exchange 10959r with MessageCounter:261734715.
I (67535) chip[IN]: Sending encrypted msg 0x3fc99934 with MessageCounter:261734715 to 0xD4AD2570BF0490EE (1) at monotonic time: 00000000000107CA msec
I (67555) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x4000 is 1 **********
I (67565) chip[IN]: Prepared secure message 0x3fc9994c to 0xD4AD2570BF0490EE (1)  of type 0x5 and protocolId (0, 1) on exchange 8465i with MessageCounter:261734716.
I (67575) chip[IN]: Sending encrypted msg 0x3fc9994c with MessageCounter:261734716 to 0xD4AD2570BF0490EE (1) at monotonic time: 00000000000107F2 msec
I (67585) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x0000 is 64 **********
I (67595) chip[ZCL]: Event: move from 64
I (67605) chip[ZCL]:  to 112 
I (67605) chip[ZCL]: (diff +1)
I (67615) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x0000 is 112 **********
I (67625) app_driver: JDS n 3 b 50 step 32
I (67625) led_driver_ws2812: led set r:0, g:0, b:0
I (67635) esp_matter_core: strore attribute in nvs: endpoint_id-0x1, cluster_id-0x8, attribute_id-0x0
I (67645) app_main: Free 56384
I (67655) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x000F is 0 **********
I (67655) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0005's Attribute 0x0003 is 0 **********
I (67665) chip[ZCL]: Setting on/off to 0x3c13e61c due to level change
I (67675) chip[ZCL]: On/Off set value: 1 1
I (67675) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0006's Attribute 0x0000 is 1 **********
I (67685) chip[ZCL]: On/off already set to new value
I (67695) esp_matter_attribute: ********** Endpoint 0x0001's Cluster 0x0008's Attribute 0x0001 is 0 **********
[... cut ...]
I (68115) chip[EM]: Received message of type 0x1 with protocolId (0, 1) and MessageCounter:183370838 on exchange 8465i
I (68125) chip[IN]: Prepared secure message 0x3fcbc848 to 0xD4AD2570BF0490EE (1)  of type 0x10 and protocolId (0, 0) on exchange 8465i with MessageCounter:261734720.
I (68135) chip[IN]: Sending encrypted msg 0x3fcbc848 with MessageCounter:261734720 to 0xD4AD2570BF0490EE (1) at monotonic time: 0000000000010A23 msec
I (68155) app_main: Free 68620
I (68655) app_main: Free 68856
I (69155) app_main: Fr
```

dhrishi, 2022-09-08:
The maximum value is 254 for the CurrentLevel attribute of the Level Control cluster.

closed 2022-09-08 by jonsmirl