espressif/esp-matter #57: Development boards question.

giandub, 2022-09-11:
Hello, I want to know if development boards like devkit-c will be compatible with the entire matter ecosystem as a finished product. I have seen that devices must have some certifications such as device attestation certificate and others. Currently I have some the devkit-c flashed with my own firmware in my home and I wanted to know if flashing these cards with the examples of this repo the cards will be compatible with the entire ecosystem matter.thank you.

jonsmirl, 2022-09-18:
This is more of a question for the devices doing the commissioning. In the current Apple beta it will display a popup say "Add uncertified device?" and it will add it if you say 'yes'. Google beta does the same thing.   I don't have access to the Amazon beta.   I believe the commissioners will keep this capability, but they don't have to.

giandub, 2022-09-19:
Hello, so it would be preferable to manage the development boards with matter/non matter bridge? In that case the bridge would still be a non-certified device? What would be the limitations in using a nn certified device?

jonsmirl, 2022-09-19:
Each commissioner can set their own policy for allowing uncertified devices onto the network. The main three commissioners are Google, Apple, Amazon. For sure Google and Apple are allowing uncertified devices. I suspect Amazon will too, but I don't have access to their betas.  When you add an uncertified device you see a popup -- "This device is not Matter certified, do you want to add it anyway?"  This applies to bridged devices too, it is the bridge that is supplying the certification.

I believe Google, Apple, Amazon will all continue allowing this in their production releases --- but --- you may have to put the device (Homepod, Google Home, Echo) into developer mode before they will allow it.  Matter is designed this way to stop people from shipping 1000s of uncertified devices.   The decision on how to support uncertified devices is under the control of Google, Apple, Amazon we just have to live with whatever they decide to do.  

Other than the popup about being uncertified, everything else works the same exact for OTA updates. Uncertified devices can't get OTA updates from the Mater infrastructure. If you are planning on wide spread distribution you I believe you can get certified for as little at $1,000 which includes all of the certification testing. Then you can use OTA to update.

closed 2022-09-20 by giandub