espressif/esp-matter #58: Unable to add accessory

AchimPieters, 2022-09-16:
When I try to connect trough HomeKit, I get the message: Unable to add accessory, pairing with accessory failed. See image.
At the monitor in the terminal, it shows this.

```
I (116391) chip[DL]: CHIPoBLE advertising started
I (178011) chip[DL]: BLE GAP connection established (con 0)
I (178011) chip[DL]: CHIPoBLE advertising stopped
I (178661) chip[DL]: BLE GAP connection terminated (con 0 reason 0x213)
I (178661) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name MATTER-3840)
I (178681) NimBLE: GAP procedure initiated: advertise; 
I (178681) NimBLE: disc_mode=2
I (178681) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (178691) NimBLE: 
```

[image]

dhrishi, 2022-09-18:
Can you please confirm the iOS version you are using on the phone as well as the gateway (HomePod or Apple TV). We have it working on the latest iOS release v16.1 beta1. 

Please give it a try and let us know in case of any issues.

AchimPieters, 2022-09-18:
iPhone XR (MRY92ZD/A) IOS16.1 (20B5045d)

Gateway's:
HomePod mini (9MY5G2FN/A) OS 16.1 Latest
Apple TV 4K  OS 16.1 Latest
Apple TV HD OS 16.1 Latest

jonsmirl, 2022-09-18:
You also need to install a profile for the Matter support...
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md

Once you get everything installed correctly it should work.

AchimPieters, 2022-09-18:
@jonsmirl I have installed the profiles and the latest IOS Beta (16.1).
[image]

jonsmirl, 2022-09-18:
I also have this installed
https://download.developer.apple.com/iOS/iOS_Logs/EnableMatter_Instructions.pdf

dhrishi, 2022-09-19:
@AchimPieters I am on https://github.com/espressif/esp-matter/commit/960517dc14ad797c1f54f87982e46b59a4d095b3 and I can commission and control Matter light with iOS 16.1 (20B5045d)

jonsmirl, 2022-09-22:
I've confirmed that Apple broke IOS paring for a while. The update from Apple that I received yesterday fixed pairing.

dhrishi, 2022-09-23:
Thanks @jonsmirl 

@AchimPieters Can you please try with the latest update? iOS 16.1 beta2 - https://developer.apple.com/documentation/ios-ipados-release-notes/ios-16_1-release-notes

AchimPieters, 2022-09-24:
@dhrishi solved!

closed 2022-09-24 by AchimPieters