espressif/esp-matter #59: Problem with the application code integration with Matter examples(ESP32-C3-MINI-1)

VivekPalle, 2022-09-21:
Hi
We started to implement our application code with Matter ESP32 examples. Please clarify the below.

- Do you have any code flow document for these examples from Espressif, that helps us to understand and integrate our application code in a proper way? If yes, please share the same.
I tried to add my application code of energy metering part of a socket in the main function as a task. But it is not working as expected. It is a task loop, but it executed only once. Please help in this.

- Where do we found the cluster list document that describes cluster syntax, parameters and cluster ID's? We want to add metering related cluster to lighting-app example.

- Is Matter environment works in Ubuntu v22.04? It is working for v20.04, we have PC's with 22.04. Just want to know it supports or we will change the version to 20.04.

jonsmirl, 2022-09-21:
```
void memTask(void *pvParameter) {

    while (1) {
        /* Delay 1 tick (assumes FreeRTOS tick is 10ms */
        vTaskDelay(pdMS_TO_TICKS(1000));

       ESP_LOGI(TAG, "Free %d", xPortGetFreeHeapSize());
    }

    /* A task should NEVER return */
    vTaskDelete(NULL);
}
xTaskCreatePinnedToCore(memTask, "gui", 4096*2, NULL, 0, NULL, 1);
```

To get the doc you have to either wait until Matter ships, or pay $20,000 to the CSA. This has nothing to do with Espressif.
https://csa-iot.org/become-member/

There is not much point in adding metering support right now since it is not in the Matter standard.  Nothing except your own app will be able to read it.

Matter works on v22.04 now, they have finally fixed Python 3.10 support.

dhrishi, 2022-09-27:
@VivekPalle The Matter specification is not into the open source yet. Although, you can take a look at the [supported clusters into the upstream repo](https://github.com/project-chip/connectedhomeip/tree/master/src/app/clusters/) to get some details. Other questions are answered by @jonsmirl

VivekPalle, 2022-10-19:
@dhrishi @jonsmirl 
Thanks for your support. The task is running perfectly. 
Any more updates on metering cluster (to send Voltage, Current, Power and Energy parameters) in these days?

I am facing a challenge while testing my code (Socket application), I observed in many cases when I controlled the switching of Relay ON/OFF from the chip-tool the control is happening twice and the load is coming back to same state without toggling it's state from ON to OFF or OFF to ON.

Please find the attached logs for your reference. Please help me with the solution.
[image]
[image]

VivekPalle, 2022-10-20:
- Chip-tool "Unpair" command is not working to clear the details in the matter device. We are always erasing and flashing the firmware. Please suggest how to remove and re-add the device.
- Seems the debounce time is not implemented in the sample code (lighting-app) for push or toggle button. Sometimes the state change is not happening correctly. It is changing the state twice or more. Please help if you have any suggestions.

VivekPalle, 2022-10-21:
@dhrishi @jonsmirl 

I came out of the below issues after implementing the de-bounce time for push button switch.
- load flickering with button de-bouncing.
- Also the device controlling twice when toggling the state from chip-tool. When onoff command sent from the chiptool, the device controlling once and there is an interrupt from the switch due to relay switching that might causing some de bounce in the voltage.

These two issues are resolved. 

Below the open points for which I need your support.
- Chip-tool "Unpair" command is not working to clear the details in the matter device. We are always erasing and flashing the firmware. Please suggest how to remove and re-add the device.
- In the "light-switch" example, how do I see the notifications from the device (when button pressed) in the chip-tool. In my application, going to implement multiple switches to control multiple loads. I just want to understand whether my implemented switches are sending notifications or not.
- Also want to know how do I enable the multiple endpoints to the multiple switches in the examples code.

VivekPalle, 2022-10-25:
Hi,

Awaiting your response on this? This helps us to move further.

jonsmirl, 2022-10-25:
In the "light-switch" example, how do I see the notifications from the device (when button pressed) in the chip-tool. In my application, going to implement multiple switches to control multiple loads. I just want to understand whether my implemented switches are sending notifications or not.

you need to subscribe to them

Also want to know how do I enable the multiple endpoints to the multiple switches in the examples code.

You can repeat code like this as many times as you want to....
```
        dimmer_switch::config_t switch_config;
        endpoint_t *endpoint2 = dimmer_switch::create(node, &switch_config, ENDPOINT_FLAG_NONE, (void *)i);
        endpoint::add_device_type(endpoint2, on_off_switch::get_device_type_id(), on_off_switch::get_device_type_version());
```

VivekPalle, 2022-10-25:
@jonsmirl Thanks for your response.

I worked on "lighting-App" example and subscribed to "on-off" events from Chip-tool. Verified the same in chip-tool, it is working.

But in "light-switch-app" example, when I try to subscribe to onoff / switch, the response is "unsupported cluster". 
Below the subscriptions I tried.
- switch subscribe number-of-positions 2 5 789456 1
- onoff subscribe on-off 2 5 789456 1

Please help with which subscription works for "light-switch-app" example?

In which file I can find the endpoint declaration?

"Unpair" is not working for esp32 examples from the chip-tool. How to do this? Is there any other procedure to delete or unpair the device?
Below the command I used.
- ./chip-tool pairing unpair node-id.

jonsmirl, 2022-10-25:
The esp-matter light-switch-app is an example of the OnOff switch cluster. The OnOff switch is designed to be bound to a light bulb. You are probably looking for the generic Switch cluster which is a different device type.

``
        generic_switch::config_t generic_config;
        endpoint_t *endpoint3 = generic_switch::create(node, &generic_config, ENDPOINT_FLAG_NONE, (void *)i);
``

dhrishi, 2022-10-26:
@VivekPalle I will first talk about the unpair part. I assume that you want to uncommission or remove a device from the fabric. If that is the case then please see the below steps:

1. Commission the device using pairing ble-wifi command
$ ./out/debug/chip-tool pairing ble-wifi <node-id> TESTSSID TESTPASSWD 20202021 3840

2. Read current fabric index:
$ ./out/debug/chip-tool operationalcredentials read current-fabric-index <node-id> 0
See the print `CurrentFabricIndex: <fabric-index>` in the output of chip-tool. Use that in the below command

3. Remove device from the fabric you want to
$ ./out/debug/chip-tool operationalcredentials remove-fabric <fabric-index> <node-id> 0

4. The device is now not in Matter fabric but it is connected to the IP network. Reboot the device (to start commissioning mode again - can be implemented on a button press too) and then commission the device using on-network command
$ ./out/debug/chip-tool pairing onnetwork <new-node-id> 20202021

Another option is to reset to factory the device (rather than erase flash every time) using the Boot button on ESP32-C3. Long press it for around 7-8 seconds. You will see the print:
`I (7622) app_reset: Factory reset triggered. Release the button to start factory reset.`

And then commission using pairing ble-wifi command (step 1)

dhrishi, 2022-10-26:
@VivekPalle There are 2 different clusters (and hence device types) here:
1. On/Off client cluster (On/Off Light Switch device type) - That can send commands or subscribe to attributes of On/Off server cluster to control them.
2. Switch server cluster (Generic Switch device type) - That has attributes and events (switch long press, switch short press etc.) of its own.

The light_switch example has 1. above. Do you want to achieve the same ie. an on/off light switch?

VivekPalle, 2022-10-26:
@dhrishi 

Thanks for your response. I am clear with the uncommission part.

- In the "lighting-app" example we were able to subscribe to "onoff" cluster with on-off attribute and got the events in the chip-tool.
- In the "light-switch-app", which cluster and attribute helps to see the switch press notifications in the chip-tool.?
This helps us in implementing multiple switches application to check all my code changes are working.

Do you have any code flow documentation that helps to implement the custom codes with the available examples?
We want to add endpoints and other functions which are working in the current switch example code. Your guide will help us to accomplish this.

dhrishi, 2022-10-26:
>In the "lighting-app" example we were able to subscribe to "onoff" cluster with on-off attribute and got the events in the chip-tool.

In lighting-app example upstream, the device is On/Off server cluster. And hence the chip-tool which has the On/Off client cluster can subscribe to OnOff attribute on the device and get notified when it is updated.

>In the "light-switch-app", which cluster and attribute helps to see the switch press notifications in the chip-tool.?

In light-switch-app there is no server cluster (and hence no attribute). For that, you will need the switch device type (2. in my above comment). So that, the chip-tool can commission the device and then subscribe to events like switch pressed/released etc. Please check:

To create the switch device type - https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_endpoint.cpp#L333 
To trigger events on button press - https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_event.cpp#L191

VivekPalle, 2022-10-26:
@dhrishi 

I just tested the uncommission part.

I will first talk about the unpair part. I assume that you want to uncommission or remove a device from the fabric. If that is the case then please see the below steps:

> - It is working fine.

Another option is to reset to factory the device (rather than erase flash every time) using the Boot button on ESP32-C3. Long press it for around 7-8 seconds. You will see the print:
I (7622) app_reset: Factory reset triggered. Release the button to start factory reset.

> - It's not working for me. I tried with 'lighting-app' and 'light-switch-app' examples. No debug found in the long press of the 'boot' button. Please help on this. 
This is what required for me or else we need to erase flash every time to commission the device with ble-wifi.

dhrishi, 2022-10-26:
@VivekPalle 
>It's not working for me. I tried with 'lighting-app'

Which lighting-app are you talking about? If it is the upstream connectedhomeip lighting-app then it won't work.

> 'light-switch-app' examples. No debug found in the long press of the 'boot' button.

Ideally, it should work. See this: https://github.com/espressif/esp-matter/blob/main/examples/light_switch/main/app_main.cpp#L89. 

Alternatively, you can use the CLI `matter esp factoryreset` to reset the device to factory settings using the console.

VivekPalle, 2022-10-27:
@dhrishi 

I am using upstream connectedhomeip examples from the start.  Just moved to work with the examples available in the esp-matter/examples folder.
[image]

- I just tried to check binding and acl with light-switch and light examples, but it is not working. The Switch is unable to connect to the light. I tried many times, find the below error for your reference.

Error:
I (888264) chip[EM]: Received message of type 0x6 with protocolId (0, 1) and MessageCounter:89193962 on exchange 8221r
I (888284) chip[IN]: Prepared secure message 0x3fc98814 to 0x000000000001B669 (1)  of type 0x7 and protocolId (0, 1) on exchange 8221r with MessageCounter:97556521.
I (888294) chip[IN]: Sending encrypted msg 0x3fc98814 with MessageCounter:97556521 to 0x000000000001B669 (1) at monotonic time: 00000000000D8DE2 msec
I (888344) chip[EM]: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:89193963 on exchange 8221r
I (888364) chip[DIS]: UDP:[FE80::36B4:72FF:FE71:D5B4%st1]:5540: new best score: 3
I (888364) chip[DIS]: UDP:192.168.98.50%st1:5540: score has not improved: 2
I (888374) chip[DIS]: Checking node lookup status after 101 ms
I (888374) chip[DIS]: Keeping DNSSD lookup active
I (888474) chip[DIS]: Checking node lookup status after 200 ms
I (888474) chip[SC]: Initiating session on local FabricIndex 1 from 0x0000000000007283 -> 0x0000000000005164
I (888654) chip[IN]: Prepared unauthenticated message 0x3fc98814 to 0x0000000000000000 (0)  of type 0x30 and protocolId (0, 0) on exchange 53866i with MessageCounter:35324845.
I (888664) chip[IN]: Sending unauthenticated msg 0x3fc98814 with MessageCounter:35324845 to 0x0000000000000000 at monotonic time: 00000000000D8F58 msec
E (888684) chip[SVR]: Failed to establish connection to node 0x0000000000005164

Suggest, what's wrong here. The same worked for me with upstream connectedhomeip examples.

Thanks.

VivekPalle, 2022-10-27:
Also the button long press reset is not working with light-switch example and working fine with light example.

VivekPalle, 2022-10-30:
@dhrishi 
Any updates on this?

And also want to check, any recent examples available with energy metering (voltage, current, power and Energy) cluster support?

jonsmirl, 2022-10-30:
Energy metering is not supported in the first Matter standard. It has been pushed in to the next version. If you want it, you have to do it with custom non-standard clusters that Amazon, Google, Apple, etc are not going to understand. Only your own app will be able to read it.

dhrishi, 2022-11-01:
>I just tried to check binding and acl with light-switch and light examples, but it is not working. 

@VivekPalle Can you confirm if you've followed the steps mentioned [here](https://github.com/espressif/esp-matter/tree/main/examples/light_switch#21-bind-light-to-switch)

>Also the button long press reset is not working with light-switch example and working fine with light example.

We will check this

VivekPalle, 2022-11-01:
@dhrishi 

@VivekPalle Can you confirm if you've followed the steps mentioned [here](https://github.com/espressif/esp-matter/tree/main/examples/light_switch#21-bind-light-to-switch)

Yes, I followed the same. It worked for me with upstream connectedhomeip examples, but not worked with the espressif specific examples.

dhrishi, 2022-11-01:
>Yes, I followed the same. It worked for me with upstream connectedhomeip examples, but not worked with the espressif specific examples.

Ok. We will test with the latest codebase and update. It will be helpful if you can upload the entire console logs on both the device as well as the chip-tool side.

jadhavrohit924, 2022-11-01:
@VivekPalle 
Light and light-switch if working fine on commit id `3401f142664c76172365a49503c83c47ca568051`.
Upstream submodule commit id `87bee4de25886ba00f1a177cd0f8791a48528a1f`

jadhavrohit924, 2022-11-01:
> Also the button long press reset is not working with light-switch example and working fine with light example.

Reset is also working on above commits

dhrishi, 2022-11-01:
@VivekPalle Please see the above comments. Can you try with the latest code? If it doesn't work, please feel free to upload the logs on the device as well as chip-tool side. Where does it fail for you? Commissioning, ACL or Binding?

dhrishi, 2022-11-18:
I will close this now. Please create a new issue, with all the details, if it is still failing.

closed 2022-11-18 by dhrishi