espressif/esp-matter #61: [Question] ESP32 All Clusters App BLE Advertising but Not Visible on Phone?

loganbenda, 2022-09-23:
Hi, 

I've been testing out Matter in an ESP32 with the All Clusters App. I've been able to build and flash properly as well as commission through the chip-tool and change a light state. However, when the ESP32 says it's advertising, I cannot see that possible connection in anything. I've tried with the Google Home app and nRF Connect BLE sniffer (on Android) to view with no success.

`I (3029) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name MATTER-3840)
I (3039) chip[DL]: Device already advertising, stop active advertisement and restart
I (3049) NimBLE: GAP procedure initiated: stop advertising.

I (3059) NimBLE: GAP procedure initiated: advertise; 
I (3059) NimBLE: disc_mode=2
I (3059) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (3069) NimBLE: 

I (3079) chip[SVR]: Cannot load binding table: a0
I (32949) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable, device name MATTER-3840)
I (32949) chip[DL]: Device already advertising, stop active advertisement and restart
I (32959) NimBLE: GAP procedure initiated: stop advertising.

I (32959) NimBLE: GAP procedure initiated: advertise; 
I (32979) NimBLE: disc_mode=2
I (32979) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (32989) NimBLE: `

I'm confused on why that would be, any thoughts based on the log above? Does it have to do with the Phone's OS?

Thanks!

dhrishi, 2022-09-23:
@loganbenda BLE advertisement is active before commissioning and hence the chip-tool can commission the ESP32. Once commissioning is complete, BLE (and hence the advertisement) is turned off in the default case.

Is the above log before commissioning? Do you mean that before commissioning too the BLE scanner app cannot see it? How do you try to find the device in the BLE scanner app? I notice a bug in the print statement which might cause the confusion -

> device name MATTER-3840

The BLE advertisement does not use name at all. So, may be, there is an ESP32 with MAC address instead of the above name and that is your device. You can open the details and verify if the following are the contents of raw advertisement data for the default all-clusters-app: `0x0201060B16F6FF00000FF1FF008000`

We will fix the print statement. Also, the Matter specification does not mention the name to be mandatory in the BLE advertisement and hence it is not added.

jonsmirl, 2022-09-23:
Once you commission a board you have to full-erase the ESP flash before the BLE device will appear again.

loganbenda, 2022-09-23:
> @loganbenda BLE advertisement is active before commissioning and hence the chip-tool can commission the ESP32. Once commissioning is complete, BLE (and hence the advertisement) is turned off in the [...]

@dhrishi Thanks for the comment. Yes, this is a log before commissioning. I was looking for MATTER-3840, so that was part of the confusion. I think I see the device now, if the UUID is specified as 0xFFF6? Do you know how we could enable the name as part of the advertising data to make it easier to search for with a phone app?

@jonsmirl I was thinking that was the case after commissioning was complete when reading through the example code. Thanks!

jonsmirl, 2022-09-23:
When you deploy your products Apple, Google and Amazon are providing commissioning libraries.
https://developers.home.google.com/mobile-sdk

Those libraries will take care of all of the QRCode and BLE stuff. Once Apple, Google and Amazon have commissioned the device onto wifi there is an option in their Home apps to open a commissioning window. Opening the commissioning window allows your app access. In most cases you won't bother with directly commissioning the device.

loganbenda, 2022-09-23:
@jonsmirl After commissioning, is the intent to run all of the matter devices' data through one of the respective hubs (Apple, Google, Amazon) to the cloud, or could it also be through a custom application to the cloud without necessarily relying on a hub? I've been trying to find details on how AWS might integrate the cloud portion, but haven't found a ton of info to date.

jonsmirl, 2022-09-23:
Apple, Google and Amazon apps are going to provide remote control of standard devices, you don't need a cloud piece if you have a standard device.  If your device is non-standard and you want to use the cloud, then you have to do everything yourself like before Matter.

dhrishi, 2022-09-27:
>I think I see the device now, if the UUID is specified as 0xFFF6?

@loganbenda That's right

> Do you know how we could enable the name as part of the advertising data to make it easier to search for with a phone app?

You will have to add local name into the BLE advertisement on ESP32 [here](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/nimble/BLEManagerImpl.cpp#L705)

Feel free to close the issue if you think your questions are answered.

loganbenda, 2022-09-27:
That answers my questions on this, thanks!

closed 2022-09-27 by loganbenda