espressif/esp-matter #62: ERROR: Failed building wheel for chip-rpc

longnv97, 2022-09-29:
I got error when run install.sh, please help me check it

this log in terminal:
 Copying chip_rpc.egg-info to build/bdist.linux-x86_64/wheel/chip_rpc-0.0.1+20220930005055-py3.8.egg-info
      running install_scripts
      error: [('build/bdist.linux-x86_64/wheel/chip_rpc-0.0.1+20220930005055-py3.8.egg-info', 'build/bdist.linux-x86_64/wheel/chip_rpc-0.0.1+20220930005055.dist-info', "[Errno 13] Permission denied: 'build/bdist.linux-x86_64/wheel/chip_rpc-0.0.1+20220930005055.dist-info'")]
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for chip-rpc
Failed to build chip-rpc
ERROR: Could not build wheels for chip-rpc, which is required to install pyproject.toml-based projects

ninja: build stopped: subcommand failed.
['ninja', '-C', '/mnt/d/matter/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']

dhrishi, 2022-10-07:
@longnv97 If this is still an issue, can you please provide the details about the host? And also the OS version

longnv97, 2022-10-08:
I using Linux Subsystem ver 20.04 LTS on Windows 11, I got this error (related to chip-rpc). My laptop using intel core i5 gen 11.
I success when install esp_idf, but this error happen when run ./install.sh for esp_matter.
Please help me check it and share to me your experience.

jonsmirl, 2022-10-27:
it looks like a permissions problem...
```
error: [('build/bdist.linux-x86_64/wheel/chip_rpc-0.0.1+20220930005055-py3.8.egg-info', 'build/bdist.linux-x86_64/wheel/chip_rpc-0.0.1+20220930005055.dist-info', "[Errno 13] Permission denied: 'build/bdist.linux-x86_64/wheel/chip_rpc-0.0.1+20220930005055.dist-info'")]
```

dhrishi, 2022-11-18:
@longnv97 Were you able to resolve this issue? https://github.com/espressif/esp-matter/issues/88#issuecomment-1313440875 shows setting up esp-matter on Windows using WSL. Can you please check if it is helpful?

alibukharai, 2022-11-24:
@longnv97 if the error still presists try this [method](https://github.com/alibukharai/Blogs/blob/main/ESP-Matter(Demo)/Getting_Started/getting_started_with_matter_esp.md)

dhrishi, 2022-12-04:
Closing this now. Feel free to re-open in case of any issue.

closed 2022-12-04 by dhrishi