espressif/esp-matter #63: Matter-Andriod github build is not working with ESP examples

VivekPalle, 2022-10-05:
Hi,

We were able building the Espressif examples and also able to commission and control from the chip-tool (linux) as well.
Now our team built the Android APP from github and trying to test the same with ESP light example, but it is failing with below error.

Please help in resolving  the same to move further.

Error:
Failed to establish BLE connection:./../src/ble/BLEEndPoint.cpp:1533: Ble Error 0x00000415: Connect handshake timed out

Android APP github link:
https://github.com/project-chip/connectedhomeip/tree/master/examples/android/CHIPTool

APP log:
Commission called for node ID 0x0000000000000001
2022-10-04 18:47:05.596 28182-28196/com.google.chip.chiptool D/chip.BluetoothManager: null.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 0
2022-10-04 18:47:05.597 28182-28196/com.google.chip.chiptool D/DL: Received SubscribeCharacteristic
2022-10-04 18:47:05.597 28182-28196/com.google.chip.chiptool D/BluetoothGatt: setCharacteristicNotification() - uuid: 18ee2ef5-263d-4559-959f-4f9c429f9d12 enable: true
2022-10-04 18:47:05.615 28182-28196/com.google.chip.chiptool D/chip.BluetoothManager: null.onDescriptorWrite: 00002902-0000-1000-8000-00805f9b34fb -> 6
2022-10-04 18:47:05.615 28182-28196/com.google.chip.chiptool E/AndroidBleManager: onDescriptorWrite for 00002902-0000-1000-8000-00805f9b34fb failed with status: 6
2022-10-04 18:47:05.615 28182-28196/com.google.chip.chiptool D/BLE: subscribe complete, ep = 0x74402b1dc8
2022-10-04 18:47:05.644 28182-28196/com.google.chip.chiptool D/BluetoothGatt: onConnectionUpdated() - Device=C6:03:4E:C7:C0:59 interval=36 latency=0 timeout=500 status=0
2022-10-04 18:47:20.584 28182-32301/com.google.chip.chiptool E/BLE: connect handshake timed out, closing ep 0x74402b1dc8
2022-10-04 18:47:20.584 28182-32301/com.google.chip.chiptool E/IN: Failed to establish BLE connection: ../../src/ble/BLEEndPoint.cpp:1533: Ble Error 0x00000415: Connect handshake timed out
2022-10-04 18:47:20.584 28182-32301/com.google.chip.chiptool D/IN: Clearing BLE pending packets.
2022-10-04 18:47:20.584 28182-32301/com.google.chip.chiptool D/DL: Received UnsubscribeCharacteristic
2022-10-04 18:47:20.586 28182-32301/com.google.chip.chiptool D/BluetoothGatt: setCharacteristicNotification() - uuid: 18ee2ef5-263d-4559-959f-4f9c429f9d12 enable: false
2022-10-04 18:47:20.734 28182-28196/com.google.chip.chiptool D/chip.BluetoothManager: null.onDescriptorWrite: 00002902-0000-1000-8000-00805f9b34fb -> 0
2022-10-04 18:47:20.735 28182-28196/com.google.chip.chiptool D/BLE: Auto-closing end point's BLE connection.
2022-10-04 18:47:20.735 28182-28196/com.google.chip.chiptool D/DL: Received CloseConnection
2022-10-04 18:47:20.735 28182-28196/com.google.chip.chiptool D/BluetoothGatt: close()
2022-10-04 18:47:20.735 28182-28196/com.google.chip.chiptool D/BluetoothGatt: unregisterApp() - mClientIf=10
2022-10-04 18:47:20.737 28182-28196/com.google.chip.chiptool D/chip.BluetoothManager: onCloseBleComplete
2022-10-04 18:47:35.601 28182-32301/com.google.chip.chiptool E/SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
2022-10-04 18:47:35.602 28182-32301/com.google.chip.chiptool D/DeviceProvisioningFragment: Pairing status update: 1
2022-10-04 18:47:35.602 28182-32301/com.google.chip.chiptool D/DIS: Closing all BLE connections
2022-10-04 18:47:35.602 28182-32301/com.google.chip.chiptool D/DeviceProvisioningFragment: onPairingComplete: 50
2022-10-04 18:47:35.603 28182-28182/com.google.chip.chiptool I/DeviceProvisioningFragment: showMessage:Pairing failed

dhrishi, 2022-10-05:
@VivekPalle Can you confirm if the Android app is built with the latest master branch connectedhomeip? 

From the Android logs, it looks like the Android code is trying to enable BLE notifications (from the API name - setCharacteristicNotification) on the ESP32 device and the ESP32 returns an error. This is because, with the latest code, we've moved from BLE notifications to BLE indications.
```
2022-10-04 18:47:05.597 28182-28196/com.google.chip.chiptool D/BluetoothGatt: 
setCharacteristicNotification() - uuid: 18ee2ef5-263d-4559-959f-4f9c429f9d12 enable: true
```
Is it possible to attach/upload the device side logs? That may confirm the above speculation.

VivekPalle, 2022-10-06:
@dhrishi Thanks for your reply.

Can you confirm if the Android app is built with the latest master branch connectedhomeip?
- Yes.

Find the below device log for your reference:

I (22025) light-app: memTask - Free 120156
I (22775) chip[DL]: BLE GAP connection established (con 1)
I (22775) chip[DL]: CHIPoBLE advertising stopped
I (22775) app-devicecallbacks: Current free heap: 120156

I (23025) light-app: memTask - Free 120156
I (23735) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (23745) chip[BLE]: local and remote recv window sizes = 5
I (23745) chip[BLE]: selected BTP version 4
I (23745) chip[BLE]: using BTP fragment sizes rx 244 / tx 244.
I (24035) light-app: memTask - Free 118632
I (25045) light-app: memTask - Free 118632
I (26055) light-app: memTask - Free 118632
I (27055) light-app: memTask - Free 118632
I (28065) light-app: memTask - Free 118632
I (29075) light-app: memTask - Free 118632
I (30075) light-app: memTask - Free 118632
I (31085) light-app: memTask - Free 118632
I (32095) light-app: memTask - Free 118632
I (33105) light-app: memTask - Free 118632
I (34115) light-app: memTask - Free 118632
I (35125) light-app: memTask - Free 118632
I (36135) light-app: memTask - Free 118632
I (37135) light-app: memTask - Free 118632
I (38145) light-app: memTask - Free 118632
E (38755) chip[BLE]: receive handshake timed out, closing ep 0x3fc9cb88
I (38755) chip[BLE]: Releasing end point's BLE connection back to application.
I (39155) light-app: memTask - Free 120156
I (39925) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (39925) app-devicecallbacks: CHIPoBLE disconnected
I (39925) app-devicecallbacks: Current free heap: 120156

I (39935) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable, device name MATTER-3840)
I (39945) NimBLE: GAP procedure initiated: advertise; 
I (39945) NimBLE: disc_mode=2
I (39955) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (39965) NimBLE: 

I (39965) chip[DL]: CHIPoBLE advertising started
I (39975) app-devicecallbacks: Current free heap: 120152

I (40165) light-app: memTask - Free 120152
I (41165) light-app: memTask - Free 120152
I (42175) light-app: memTask - Free 120152

--Please help in how to make it work. Or share if you have any Android and IOS builds working for Espressif examples.

uiabhishek15, 2022-10-10:
@dhrishi We are unable to build the code if we keep some jar files which we got from the internet then we can able to build the code otherwise if we follow the instructions how to build Matter code in Android studio we were unable to build. can you guide us how to fix the issues we are facing or is there any possibility to provide jar files ?

khushbushah2302, 2022-10-10:
Hi @uiabhishek15  ,

By following the instruction given on [this link](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md), I am able to build Matter code without using any jar files. I am building an app using script given in repo (without opening project in Android studio). 

I am following below steps to build app.

1. Set environment variables with your values
- ANDROID_HOME
- ANDROID_NDK_HOME
- JAVA_HOME

For Mac OS, I am running below commands to set environment variables.
- `export ANDROID_HOME=~/Library/Android/sdk`
- `export ANDROID_NDK_HOME=~/Library/Android/sdk/ndk/24.0.8215888`
- `export JAVA_HOME=/Applications/Android\ Studio.app/Contents/jre/Contents/Home`

Note : 
- In my system, I also need to set PKG_CONFIG_PATH variable as below.
`export PKG_CONFIG_PATH=/usr/local/opt/openssl@3/lib/pkgconfig/`

2. Run below command to prepare for build.
`source scripts/bootstrap.sh`

3. Run command to build APK.
`./scripts/build/build_examples.py --target android-arm64-chip-tool build`

The debug Android package app-debug.apk will be generated at out/android-$TARGET_CPU-chip-tool/outputs/apk/debug/

Hope this will help.

dhrishi, 2022-10-17:
@uiabhishek15 @VivekPalle Did the above work? Please close the issue if it did. Thanks!

dhrishi, 2022-10-26:
@uiabhishek15 @VivekPalle Did you get this to work?

uiabhishek15, 2022-10-26:
@dhrishi No we are still unable to generate the library classes which are useful for the main application.

jonsmirl, 2022-10-27:
Google has abandoned the Android app in the Matter tree. Only Amazon is still using it.

The new Google Matter support is here:
https://developers.home.google.com/matter

dhrishi, 2022-11-18:
@VivekPalle @uiabhishek15 Is there any update you would want to share?

dhrishi, 2022-11-22:
Closing this issue now. Please create a new one in case you are still facing problems.

uiabhishek15, 2023-01-17:
@dhrishi we are facing this issue in the latest code in master branch
[image]

jonsmirl, 2023-01-17:
You source the activate script while in the connecthomeip/connectedhomeip directory in esp-matter

uiabhishek15, 2023-01-17:
@jonsmirl yes i did the same it is same as in the document right?

dhrishi, 2023-01-17:
@uiabhishek15 Are you using the examples from connectedhomeip repository directly or from esp-matter? 

If using connectedhomeip directly, then please try creating a new environment:
```
$ cd /path/to/connectedhomeip
$ rm -rf .environment
$ source scripts/bootstrap.sh
$ source scripts/activate.sh
```

If using esp-matter then please follow: https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#development-setup

uiabhishek15, 2023-01-17:
@dhrishi I am using the project-chip/connectedhomeip (https://github.com/project-chip/connectedhomeip)

do you want me to use esp-matter?

dhrishi, 2023-01-17:
@uiabhishek15 Ohh. My bad. For building the Android app you would have to use the connectedhomeip itself.

Just make sure that you use [v1.0.0.2 tag of connectedhomeip](https://github.com/project-chip/connectedhomeip/releases/tag/v1.0.0.2) as the device/firmware side code is based on it. It is always better to be in sync.

uiabhishek15, 2023-01-17:
@dhrishi after using v1.0.0.2 tag of connectedhomeip i am getting the below error when using the command "source scripts/bootstrap.sh"

ERROR at //build/config/linux/pkg_config.gni:116:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /Users/sairam/Desktop/Arandroid/connectedhomeip/.environment/gn_out/
Command: python3 /Users/sairam/Desktop/Arandroid/connectedhomeip/build/config/linux/pkg-config.py openssl
Returned 1.
stderr:

Package openssl was not found in the pkg-config search path.
Perhaps you should add the directory containing `openssl.pc'
to the PKG_CONFIG_PATH environment variable
No package 'openssl' found
Could not run pkg-config.

See //src/crypto/BUILD.gn:84:3: whence it was called.
  pkg_config("openssl_config") {
  ^-----------------------------
See //BUILD.gn:152:7: which caused the file to be included.
      "${chip_root}/src/crypto",
      ^------------------------
['gn', 'gen', '/Users/sairam/Desktop/Arandroid/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/sairam/Desktop/Arandroid/connectedhomeip/third_party/pigweed/repo"']

Traceback (most recent call last):
  File "/Users/sairam/Desktop/Arandroid/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 335, in install_packages
    subprocess.check_call(
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/sairam/Desktop/Arandroid/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/sairam/Desktop/Arandroid/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/sairam/Desktop/Arandroid/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 857, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/sairam/Desktop/Arandroid/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 849, in main
    return EnvSetup(**vars(parse())).setup()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/sairam/Desktop/Arandroid/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 501, in setup
    result = step(spin)
             ^^^^^^^^^^
  File "/Users/sairam/Desktop/Arandroid/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 661, in virtualenv
    if not virtualenv_setup.install(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/sairam/Desktop/Arandroid/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 370, in install
    install_packages(gn_target)
  File "/Users/sairam/Desktop/Arandroid/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 343, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/sairam/Desktop/Arandroid/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/sairam/Desktop/Arandroid/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

jonsmirl, 2023-01-17:
It is very difficult to correctly build the chip library for Android. It much easier to base your code off from Google's sample app. Use the Google pre-built libs in https://github.com/google-home/sample-app-for-matter-android/tree/main/app/third_party/connectedhomeip/libs

https://github.com/google-home/sample-app-for-matter-android

uiabhishek15, 2023-01-17:
@jonsmirl But as we got the documentation from Matter it has to build in Android too right?

Is there any possibility to build Android app using connectedhomeip?

dhrishi, 2023-01-17:
@uiabhishek15 Please see the detailed steps here: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#installing-prerequisites-on-macos

jonsmirl, 2023-01-17:
I have built the Android apps directly from the Chip repository. I do not recommend doing that.  The version of the CHIP library supplied by Google is integrated with GPS (Google Play Services) and the version you build won't be. The code integrating CHIP with GPS is not public. You can getting the Chip repository version working but all of the GPS integration features are not going to work.

uiabhishek15, 2023-01-18:
@jonsmirl sure i will check the CHIP library provided by Google. Thanks for the reply.

uiabhishek15, 2023-01-18:
@jonsmirl By using the Google Matter sample app i am unable to add erf32, may I know do you have any solution for adding the ERF32?

jonsmirl, 2023-01-18:
The Matter running on your ERF32 is likely too old.

VivekPalle, 2023-01-18:
@jonsmirl 
We are using the latest Matter SDK from Silabs. And we were able to add our thread device to Apple Home APP through HomePod Mini. But unable to add with Google sample APP through Nest Hub (Gen2).

The module used is EFR32MG24.

jonsmirl, 2023-01-18:
Try adding them through Google Home.
Also, inside Google Home select the Nest Hub. From that UI tell it to restart. That will make it download the latest software.

uiabhishek15, 2023-01-19:
@jonsmirl After Rebooting the device also I am unable to add the device.
Below are versions for my Nest 2
Software version : 47.9.4.447810048
Chromecast firmware version: 1.56.324896