espressif/esp-matter #64: OSError: [Errno 101] Network is unreachable

sidhoo-root, 2022-10-09:
Hi  espressif team:

    I hv meet a problem when i run ./install.sh , log as fllow:
```
xxxx:~/Work/esp-matter$ ./install.sh 

Running Matter Setup

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...Traceback (most recent call last):
  File "/home/sid/Work/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 795, in <module>
    sys.exit(main())
  File "/home/sid/Work/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 787, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/sid/Work/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 457, in setup
    result = step(spin)
  File "/home/sid/Work/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 546, in cipd
    cipd_client = cipd_wrapper.init(
  File "/home/sid/Work/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 357, in init
    bootstrap(client, silent, rosetta=rosetta)
  File "/home/sid/Work/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 311, in bootstrap
    tmp.write(client_bytes(rosetta))
  File "/home/sid/Work/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 240, in client_bytes
    conn.request('GET', path)
  File "/usr/lib/python3.8/http/client.py", line 1256, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1302, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1251, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1011, in _send_output
    self.send(msg)
  File "/usr/lib/python3.8/http/client.py", line 951, in send
    self.connect()
  File "/usr/lib/python3.8/http/client.py", line 1418, in connect
    super().connect()
  File "/usr/lib/python3.8/http/client.py", line 922, in connect
    self.sock = self._create_connection(
  File "/usr/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/usr/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
OSError: [Errno 101] Network is unreachable
```
1. the Prerequisites for Matter and espif (4.4.2) is ready. 
2. and i found that issue(https://github.com/project-chip/connectedhomeip/issues/2893) had beed fixed
3. ping www.google.com is ok
>~/Work/esp-matter$ ping www.google.com
PING www.google.com (142.250.204.36) 56(84) bytes of data.
64 bytes from hkg07s38-in-f4.1e100.net (142.250.204.36): icmp_seq=1 ttl=128 time=7.73 ms
64 bytes from hkg07s38-in-f4.1e100.net (142.250.204.36): icmp_seq=2 ttl=128 time=6.96 ms
64 bytes from hkg07s38-in-f4.1e100.net (142.250.204.36): icmp_seq=3 ttl=128 time=6.79 ms
64 bytes from hkg07s38-in-f4.1e100.net (142.250.204.36): icmp_seq=4 ttl=128 time=6.63 ms

hope you give me some advise, thank you

dhrishi, 2022-10-10:
@sidhoo-root  This seems to be an issue with the upstream connectedhomeip repo. Can you please check: https://github.com/project-chip/connectedhomeip/issues/8810

sidhoo-root, 2022-10-10:
@dhrishi ok ,thanks. and Is it related to using a virtual machine environment to install ?

dhrishi, 2022-10-10:
@sidhoo-root Yes, it seems so

sidhoo-root, 2022-10-12:
@dhrishi i had installed it on the Ubuntu machine, it still report this error.   and the host : chrome-infra-packages.appspot.com can be ping, does it means the CIPD_HOST = 'chrome-infra-packages.appspot.com'  works in  my env ?
>PING chrome-infra-packages.appspot.com (199.193.116.105) 56(84) bytes of data.
64 bytes from 199-193-116-105.static.hvvc.us (199.193.116.105): icmp_seq=1 ttl=128 time=220 ms
64 bytes from 199-193-116-105.static.hvvc.us (199.193.116.105): icmp_seq=2 ttl=128 time=218 ms
64 bytes from 199-193-116-105.static.hvvc.us (199.193.116.105): icmp_seq=3 ttl=128 time=219 ms
64 bytes from 199-193-116-105.static.hvvc.us (199.193.116.105): icmp_seq=4 ttl=128 time=220 ms
64 bytes from 199-193-116-105.static.hvvc.us (199.193.116.105): icmp_seq=5 ttl=128 time=218 ms

sidhoo-root, 2022-10-13:
@dhrishi the problem was solved,  because the dns can't be parsed correctly in my office network

closed 2022-10-13 by sidhoo-root

dhrishi, 2022-10-13:
@sidhoo-root Good to know that the issue is identified and resolved.

wivin001, 2023-03-14:
我也遇到这个问题了，使用虚拟机vmware15.0  ubuntu 22.10系统，已经翻墙了，但是依然会报这个错误，有没有解决办法？

chuckrui, 2023-03-15:
> @dhrishi问题解决了，因为在我的办公网络中dns无法正确解析

您好，想问一下是如何解决的，我也遇到了这个问题，ping youtube 跟google都没有问题

chuckrui, 2023-03-15:
> > @dhrishi问题解决了，因为在我的办公网络中dns无法正确解决
> 
> 您好，想问下是如何解决的，我也遇到了这个问题，ping youtube跟google都没有问题

[image]
泪目了~终于搞定了，我就说怎么Ubuntu里面明明都可以访问外网，但是就是一直提示我有问题，不应该的啊，然后在命令行里面定义了一下IP就行了

temp4gh, 2023-03-15:
> > > @dhrishi问题解决了，因为在我的办公网络中dns无法正确解决
> > 
> > 
> > 您好，想问下是如何解决的，我也遇到了这个问题，ping youtube跟google都没有问题
> 
> [image] 泪目了~终于搞定了，我就说怎么Ubuntu里面明明都可以访问外网，但是就是一直提示我有问题，不应该的啊，然后在命令行里面定义了一下IP就行了

Same issue,Could you please explain  how to '命令行里面定义了一下IP'?Thanx!

chuckrui, 2023-03-16:
> > > > @dhrishi问题解决了，因为在我的办公网络中dns无法正确解决
> > > 
> > > 
> > > 您好，想问下是如何解决的，我也遇到了这个问题，ping youtube跟google都没有问题
> > 
> > 
> > [image] 泪目了~终于搞定了，我就说怎么Ubuntu里面明明都可以访问外网，但是就是一直提示我有问题，不应该的啊，然后在命令行里面定义了一下IP就行了
> 
> Same issue,Could you please explain how to '命令行里面定义了一下IP'?Thanx!

sure，u can open the sofeware and the click here just as show in the picture as below.
[image]
[image]
copy that in terminal command line it will connect normally.

temp4gh, 2023-03-16:
> sure，u can open the sofeware and the click here just as show in the picture as below. [image] [image] copy that in terminal command line it will connect normally.

Thanx for your reply!I will take a try.

zplmcu, 2025-07-03:
@temp4gh, do you have a fix for it? Currently, I have encountered the same issue. Also, exported the proxy; 
Unfortunately, it doesn't work. (wifi network can go to Google, YouTube etc)

[image]