espressif/esp-matter #65: Compilation error with CHIP_STACK_LOCK_TRACKING_ENABLED unset

ryan-ma, 2022-10-12:
The compilation fail In file esp_matter_core.cpp
```
status_t chip_stack_lock(uint32_t ticks_to_wait)
{
    if (PlatformMgr().IsChipStackLockedByCurrentThread()) {
        return ALREADY_TAKEN;
    }
```
`PlatformMgr().IsChipStackLockedByCurrentThread()` is not available if `CHIP_STACK_LOCK_TRACKING_ENABLED` unset

We need to disable this flag to build the production firmware.
Otherwise, the binary file will contain file name path string generated by
```
#define assertChipStackLockedByCurrentThread() ::chip::Platform::Internal::AssertChipStackLockedByCurrentThread(__FILE__, __LINE__)
```

jadhavrohit924, 2022-10-17:
How to disable `CHIP_STACK_LOCK_TRACKING_ENABLED`

ryan-ma, 2022-10-24:
I disable it by modify BUILD.gn in local platform. The BUILD.gn is copied from in connectedhomeip/src/platform

chshu, 2022-11-09:
@ryan-ma Could you check has https://github.com/espressif/esp-matter/commit/f8fec4e92f4f1b26f8b2d2322c1b0833853472ca fixed the compile issue?

dhrishi, 2022-11-18:
Closing this. Feel free to reopen in case of any issue.

closed 2022-11-18 by dhrishi