espressif/esp-matter #66: [HELP] How to connect Matter device with Apple Home

longnv97, 2022-10-14:
I want to connect Matter device to Apple home, but I do not find a guide to connect.
Please share with me any information about this. Thanks.

jakubdybczak, 2022-10-15:
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md

longnv97, 2022-10-16:
> https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md

Thanks for your sharing. Will you try to connect the Matter device with Apple Homekit? 
I added the Matter device to Apple Homekit app, but I did not get/set the status for the Matter device from Apple Homekit app, the status of the Matter device is "no reply".
If possible, could you share your opinion?

jakubdybczak, 2022-10-16:
Yeah, I have tried it and had the same issue. In my case the reason was that I don’t have an Apple TV / HomePod. It’s required as it acts as home hub. (https://github.com/project-chip/connectedhomeip/issues/13296)

longnv97, 2022-10-16:
> 

I have success on ios 16 with developer mode and I have Homepod to the home center. 
Did you try connecting Matter device with Amazon Alexa? I did not find any guide for this. If possible, could you share your experience?

longnv97, 2022-10-16:
> Yeah, I have tried it and had the same issue. In my case the reason was that I don’t have an Apple TV / HomePod. It’s required as it acts as home hub.

I have success on ios 16 with developer mode and I have Homepod to the home center.
Did you try connecting Matter device with Amazon Alexa? I did not find any guide for this. If possible, could you share your experience?

jakubdybczak, 2022-10-16:
Nice to hear it!
Nope, I haven't tried connecting Matter device to anything more than Homekit ;).

longnv97, 2022-10-16:
> Nice to hear it! Nope, I haven't tried connecting Matter device to anything more than Homekit ;).

Thank you!
I trying connecting Matter device to anything Home hub, but Matter is not ready for connect to Google home and Alexa.
There's not much documentation for that. If you find any guide, please share it with me.

In the link to guide Pair a Matter device with Google Home, I trying it: https://developers.home.google.com/matter/integration/pair#espressif

jakubdybczak, 2022-10-16:
I have tried to add Matter support to Google Home. I did it by:
1. Updating Google Play services: https://developers.home.google.com/matter/verify-services
2. Veryfing if Matter service is working (https://developers.home.google.com/matter/build/troubleshooting)
I had to restart my phone in order to have this service started.

Finally I have possibility to add Matter device (there is an option in Google Home app after autosearch fails). However, I haven't tried connecting it with real Matter device. Google Nest is needed for it as it acts as home hub same like on Apple HomeKit.

longnv97, 2022-10-16:
> I have tried to add Matter support to Google Home. I did it by:
> 
> 1. Updating Google Play services: https://developers.home.google.com/matter/verify-services
> 2. Veryfing if Matter service is working (https://developers.home.google.com/matter/build/troubleshooting)
>    I had to restart my phone in order to have this service started.
> 
> Finally I have possibility to add Matter device (there is an option in Google Home app after autosearch fails). However, I haven't tried connecting it with real Matter device. Google Nest is needed for it as it acts as home hub same like on Apple HomeKit.
> 
> PS. These guides are so badly organized, I'm not suprised that you could not find it easily grin

Thanks for your useful information!
I will try it.
Hope to get help from you in the future :)

dhrishi, 2022-10-17:
@jakubdybczak Thank you for providing the pointers

@longnv97 Please close this issue if you are now able to connect to Apple HomeKit

jakubdybczak, 2022-10-18:
@longnv97 I have tried connecting Matter device to Google Home but also haven't success. During pairing it says that could not add device to Google account. I have looked to adb logs and found this message:
```
m.eag: PC InsertCHIPDevice returned code NOT_FOUND; Sherlog for InsertCHIPDevice is needed to diagnose
[...]
Caused by: m.kaf: NOT_FOUND: Requested entity was not found.
[...]
Commissioning failed with state m.dlwa6d968a7 [CONTEXT service id=288 ]
```

`Requested entity was not found.` sounds like some HTTP error? I'm not sure. Maybe its a bug in Play Services. It looks like everything is buggy and not mature enough to work flawless yet.

longnv97, 2022-10-19:
@jakubdybczak Thank you for your information.
I think I got the same issues as you. Please update if you have any new information.
I will do that too.

jakubdybczak, 2022-10-19:
@longnv97 I don't feel like it's best place to have a discussion, I think we are off-toping. Can you reach me at jakub.dybczak@gmail.com and close this issue, as there is no longer a problem with HomeKit? Regarding problem with connecting Matter with Google I would suggest opening an issue in main Matter repo or writing to Google.

longnv97, 2022-10-19:
@jakubdybczak ok, I got it, I will connect to you by gmail

closed 2022-10-19 by longnv97