espressif/esp-matter #67: [HELP] How to connect Matter device with Amazon Alexa

longnv97, 2022-10-16:
I want to connect Matter device to the Amazon Alexa, but I do not find a guide to connect.
Please share with me any information about this. Thanks.

jonsmirl, 2022-10-23:
I have been signed up on the Amazon Matter beta list for a year and they have never sent me anything. I do know they are supporting Matter in their Echos, they just don't seem to want to share their code.  Whenever they decide to release it, it should work similarly to the Google and Apple support.

longnv97, 2022-10-27:
@chshu @jonsmirl Thanks for your information.

jonsmirl, 2022-10-27:
Practice right now making your devices work with "Siri" and "Ok, Google". If they work right with those two, then they will work right with Alexa whenever Amazon decides to release their support. All three vendors are using the same mechanism for voice control.

https://developers.home.google.com/matter

jonsmirl, 2022-10-27:
You are probably hitting this issue

https://github.com/google-home/sample-app-for-matter-android/issues/29

dhrishi, 2022-11-03:
@longnv97 Please close this issue if your original question about Amazon Alexa is answered

jonsmirl, 2022-11-04:
https://www.engadget.com/amazon-echo-device-matter-support-compatibility-141558771.html
Amazon will bring Matter smart home support to 17 devices this year

They never let me into their beta despite having signed up almost two years ago.  I guess I will test my stuff on their production release.

dhrishi, 2022-11-14:
Closing as the original question is answered.

closed 2022-11-14 by dhrishi