espressif/esp-matter #68: blemesh_bridge commissioning failure with kCertificationDeclarationNoCertificateFound

s-mcclain, 2022-10-17:
Using the blemesh_bridge example being built out-of-box and using the opensource chip-tool example, I'm consistently getting a commissioning failure with the following error:

```
[1666019233936] [15251:85465] CHIP: [DMG] Received Command Response Data, Endpoint=0 Cluster=0x0000_003E Command=0x0000_0001
[1666019233936] [15251:85465] CHIP: [CTL] Received Attestation Information from the device
[1666019233936] [15251:85465] CHIP: [CTL] Successfully finished commissioning step 'SendAttestationRequest'
[1666019233936] [15251:85465] CHIP: [CTL] Commissioning stage next step: 'SendAttestationRequest' -> 'AttestationVerification'
[1666019233936] [15251:85465] CHIP: [CTL] Performing next commissioning step 'AttestationVerification'
[1666019233936] [15251:85465] CHIP: [CTL] Verifying attestation
[1666019233942] [15251:85465] CHIP: [CTL] Failed in verifying 'Attestation Information' command received from the device: err 601. Look at AttestationVerificationResult enum to understand the errors
[1666019233942] [15251:85465] CHIP: [CTL] Error on commissioning step 'AttestationVerification': '../../src/controller/CHIPDeviceController.cpp:1018: CHIP Error 0x000000AC: Internal error'
[1666019233942] [15251:85465] CHIP: [CTL] Failed to perform commissioning step 8
[1666019233942] [15251:85465] CHIP: [CTL] Going from commissioning step 'AttestationVerification' with lastErr = '../../src/controller/CHIPDeviceController.cpp:1018: CHIP Error 0x000000AC: Internal error' -> 'Cleanup'
[1666019233942] [15251:85465] CHIP: [CTL] Performing next commissioning step 'Cleanup' with completion status = '../../src/controller/CHIPDeviceController.cpp:1018: CHIP Error 0x000000AC: Internal error'
[1666019233942] [15251:85465] CHIP: [CTL] Expiring failsafe on proxy 0x7feef7808600
```

On the ESP32 side, these are the logs which appear to not report any errors as far as not having any valid CDs?

```
I (44488) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (44488) NimBLE: GATT procedure initiated: indicate;
I (44488) NimBLE: att_handle=14

I (44558) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (44558) NimBLE: GATT procedure initiated: indicate;
I (44558) NimBLE: att_handle=14

I (44598) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (44598) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (44608) chip[EM]: >>> [E:8430r M:187172288] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
I (44628) esp_matter_command: Received command 0x0000 for enpoint 0x0000's cluster 0x0000003E
I (44628) chip[ZCL]: OpCreds: Received an AttestationRequest command
I (44898) chip[ZCL]: OpCreds: AttestationRequest successful.
I (44908) chip[EM]: <<< [E:8430r M:252579693] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
I (44908) chip[IN]: (S) Sending msg 252579693 on secure session with LSID: 21973
I (44918) NimBLE: GATT procedure initiated: indicate;
I (44918) NimBLE: att_handle=14

E (44928) chip[DL]: Long dispatch time: 321 ms, for event type 7
I (44978) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (44988) NimBLE: GATT procedure initiated: indicate;
I (44988) NimBLE: att_handle=14

I (45048) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (45048) NimBLE: GATT procedure initiated: indicate;
I (45058) NimBLE: att_handle=14
```

chiragatal, 2022-10-17:
Can you update the chip-tool and try again? I have faced similar `AttestationVerification` errors and updating the chip-tool fixed it.

s-mcclain, 2022-10-17:
Hey @chiragatal  I just updated chip-tool to the latest v1.0-branch commit and rebuilt it. Unfortunately I am still seeing the same errors.

The strange thing is, I do not see these if I use the 'light' example application.

s-mcclain, 2022-10-17:
Looks like this may be a similar issue as https://github.com/project-chip/connectedhomeip/issues/23164

Looks like I should probably try to checkout v1.0-branch on the connectedhomeip submodule that esp-matter pulled from. Will test this out

s-mcclain, 2022-10-17:
That was it! Checking out v1.0-branch in the connectedhomeip submodule led to successful commissioning of the blemesh_bridge.

closed 2022-10-17 by s-mcclain