espressif/esp-matter #69: Matter Client as well as a Thread Boarder Router

JonathanWitthoeft, 2022-10-19:
There is an example of a thread boarder router code here:
https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br

There is an example of a Matter Client here:
https://github.com/espressif/esp-matter/tree/main/examples/light_switch

I am wondering if these 2 can co-exist. If I wanted a thread boarder router that can also control a specific thread Matter Server device, how would I go about doing that? Would the light_switch (WiFi Matter Client) firmware messages be routed by the ot_br firmware if they are both running in different threads on the same ESP32 chip? Would the firmware see the outgoing messages on the WiFi network from it's own IP address and route them to the thread network?

chshu, 2022-10-20:
Yes, the two can co-exist. It will be a Matter device with Thread Border Router capacity.

The Thread Border Router simply forwards IP layer messages between Wi-Fi and Thread networks, it doesn't handle any application layer (Matter in this case) logic. In the case that a Matter Client runs on a Thread Border Router (1st device) sends a message to another Thread Matter client (2nd device), the 1st device's IP stack should be able to figure out the correct routing, and send the message to Thread interface, then reach to the Thread device.

jonsmirl, 2022-10-23:
This is a common configuration, every Echo and Google Home is doing this. You typically use a larger CPU to build this.

JonathanWitthoeft, 2022-10-24:
Thanks @jonsmirl and @chshu. I made the assumption I would need to be a boarder router to create a thread network, but I think that was a misunderstanding of Matter and Thread fundamentals on my part. I know this question likely doesn't belong here, as it is more a design question but my requirements are as follows. 
1. The device needs to be a thread Matter controller
2. If commissioned to a thread network that does not exists (via Matter QR code), the device needs to create the thread network (Leader capabilities)
3. A Wi-Fi/Ethernet interface

**-Option 1-**
Use an OpenThread RCP module and do thread network commissioning and a thread Matter controller in the ESP32-C firmware. I want the IP stack to routing matter traffic through the thread RCP rather than the Wi-Fi/Ethernet interface.
With the ESP32-C Matter SDK, Is it easily possible to have the device come up as a **thread** commisionable Matter device rather than a Wi-Fi/Ethernet commisionable Matter device?

**-Option 2-**
Develop the Matter controller code on the thread module (nRF52840 or ESP32-H2) and create my own serial command protocol between the ESP32-C MCU and the thread module (using standard esp-idf).

Would either of you have advise on which option would make most since or be easiest to do with the Espressif SDKs as they stand today?

jonsmirl, 2022-10-24:
There is an Espressif border router demo...
https://github.com/espressif/esp-thread-br

chshu, 2022-10-26:
@JonathanWitthoeft Firstly I want to clarify that:

A Matter Controller who hopes to control Matter Thread devcies, doesn't have to be a Border Router by itself. It could be just a Wi-Fi based controller, then a Thread Border Router in the same local network could connect and rounting the messages between the Controller and Thread devices. The Thread Border Router helps with the Bi-directional Connectivity and Service Discovery.

Then comes to your requirements, if it's designed to be both Matter Controller and Thread Border Router, then Option-1 seems more resonable. But a question: Will the device also acts as the Matter Commissioner (Admin)?

If yes, It's currelty a common method in Amazon/Apple/Google's Smart speakers which act as Matter Commissioner/Controller and Thread Border Router. This device is actually a Matter Amdin, it's provisioned via the ecosystem's own flow (Not Matter), then form a Thread network and become a Border Router.

But If it's not a Commissioner, the device could be either a Matter Wi-Fi device or a Matter Thread device, if it's configured as thread commisionable Matter device, there should be a Thread network in the Matter fabric already, otherwise the commissioner doesn't know the Thread network datasets for commissioning.

dhrishi, 2022-11-03:
@JonathanWitthoeft Please close the issue if your questions are answered

dhrishi, 2022-11-14:
Closing this now. Feel free to file a new issue if you need more details

closed 2022-11-14 by dhrishi