espressif/esp-matter #70: Can BLE Mesh device connect to Matter Network?

xiaolongba, 2022-10-22:
Hello Espressif Experts,

I checked the esp-matter SDK and found that there is **blemeshbridge** example project. based on the description, it means that the BLE mesh device can connect to the Matter network via a ble mesh bridge. However, I checked the matter v1.0 spec, it only supports the Matter over Thread and Matter over WiFi. so I have the following issues, as follows:

1. if it is against the matter spec if BLE device connects to the matter network? is it reasonable?
2. unauthorized BLE devices can connect to Matter Network via an authorized ble mesh bridge if item1 is possible? any solutions for this scenario?
3. if item2 is impossible, all matter devices need to do the matter certification. but how to do the matter certification for BLE-only devices as Matter v1.0 spec only support Thread and WiFi at this moment?

regarding the issues above, could you give me a hand?

thanks in advance.

jonsmirl, 2022-10-23:
The blemeshbridge has to be Matter certified.  When the BLE bridge connects to a BLE mesh device (not normal BLE) it typically connects using BLE mesh encryption, so you have to have the QR code for the device with the key in it. This is similar as how Matter works. The bridge then constructs a Matter proxy for the BLE mesh device. It is this Matter proxy which has to be Matter certified. So the Matter network only ever sees certified devices -- the bridge is certified and the proxy is certified. It the the blemesgbridge code which is controlling which authorized BLE devices can connect. You control this by giving the bridge the QR code for the device.  Of course since you have source code you can defeat that.

Note that is is possible to bridge anything you want into a Matter network, you just need a certified bridge. For example the Zigbee bridge works like this. You could write a Zwave and Insteon bridge too if you wanted.

xiaolongba, 2022-10-24:
Hello Jonsmirl,

I am clear about the functionality of the matter bridge now, however as you said above

> So the Matter network only ever sees certified devices

Matter app how to control the non-matter devices? in my understanding, the bridge should map the non-matter devices to the matter network, and then non-matter devices can be controlled or communicate with other matter devices via Matter App directly. the more details about it, pls refer to this [link](https://blog.espressif.com/matter-bridge-for-non-matter-devices-d3b7f003a004)

regarding the doubt, can you help me double-check?

thanks in advance.

Can

jonsmirl, 2022-10-24:
The Matter app manipulates the Matter device proxy in the certified bridge. Code in the bridge implements the certified device and as you send commands to it, the bridge translates those command and then sends them onto BLE mesh devices.  You can see the bridge doing the mapping in this file... https://github.com/espressif/esp-matter/blob/main/examples/blemesh_bridge/main/blemesh_bridge.cpp

xiaolongba, 2022-10-24:
@jonsmirl 

Many thanks for your replies. so based on the description you said above, the Matter App can also control non-matter devices, but it needs to be translated, right? the diagram as below:
[image]

so it is almost the same as controlling the matter devices

xiaolongba, 2022-10-24:
@jonsmirl OK, thanks. Let me look into the code you shared first. any more further info I will reopen it.

thanks again.

dhrishi, 2022-11-03:
@xiaolongba Please close this issue if your original question about BLE Mesh and Matter is answered.

closed 2022-11-03 by xiaolongba