espressif/esp-matter #71: Google sample app not working with esp-matter

jonsmirl, 2022-10-23:
This is due to a Google update requiring that attestation work, and esp-matter is not implementing an attestation that Google will accept.
https://github.com/google-home/sample-app-for-matter-android/issues/29

Google has not implemented the  "allow uncertified device" popup like Apple has done.

I tried playing the data provider options in menuconfig and could not come up with anything that works.

jonsmirl, 2022-10-24:
Moving esp-matter over to CHIP master will allow the devices to talk to Google Home again.

dhrishi, 2022-10-26:
@jonsmirl Is this with the current connectedhomeip submodule `87bee4de25886ba00f1a177cd0f8791a48528a1f` ?

jonsmirl, 2022-10-26:
I am randomly using  0bff5ad04c696626ce3aa8fcc5c9e229ed8bacf0.

But the problem was faad9e179293775110f09692cd1c28e0d71b792f

jonsmirl, 2022-10-27:
```
commit faad9e179293775110f09692cd1c28e0d71b792f
Author: Evgeny Margolis <emargolis@google.com>
Date:   Wed Oct 5 10:09:59 2022 -0700

    Updated CSA Official CD Signing Certificates (#23027)
```

changes the CHIP certificates. This is a bit of a mess because some Google stuff is on the old certificates and some is on the new ones. They are working on sorting it out.

closed 2022-10-28 by jonsmirl