espressif/esp-matter #72: How to set field on Basic Information cluster

jonsmirl, 2022-10-25:
So how do I set these?
```
        /* Attributes managed internally */
        global::attribute::create_feature_map(cluster, 0);
        attribute::create_data_model_revision(cluster, 0);
        attribute::create_location(cluster, NULL, 0);
        attribute::create_vendor_name(cluster, NULL, 0);
        attribute::create_vendor_id(cluster, 0);
        attribute::create_product_name(cluster, NULL, 0);
        attribute::create_product_id(cluster, 0);
        attribute::create_hardware_version(cluster, 0);
        attribute::create_hardware_version_string(cluster, NULL, 0);
        attribute::create_software_version(cluster, 0);
        attribute::create_software_version_string(cluster, NULL, 0);
        attribute::create_capability_minima(cluster, NULL, 0, 0);
```

jonsmirl, 2022-10-26:
this doesn't work because the fields are read-only.  With ZAP you can set them in as initializers when the structure is created. Add them as parameters on the constructor for Basic Info?
```
    esp_err_t error;
    esp_matter_attr_val_t val = esp_matter_char_str("Lowpan", sizeof("Lowpan"));
    error = attribute::update(0, Basic::Id, Basic::Attributes::VendorName::Id, &val);
    val = esp_matter_int(0xFFF1);
    error = attribute::update(0, Basic::Id, Basic::Attributes::VendorID::Id, &val);
    val = esp_matter_char_str("Rocker", sizeof("Rocker"));
    error = attribute::update(0, Basic::Id, Basic::Attributes::ProductName::Id, &val);
    val = esp_matter_int(0x8001);
    error = attribute::update(0, Basic::Id, Basic::Attributes::ProductID::Id, &val);
    val = esp_matter_char_str("V1.0", sizeof("V1.0"));
    error = attribute::update(0, Basic::Id, Basic::Attributes::HardwareVersionString::Id, &val);
    val = esp_matter_int(0x1);
    error = attribute::update(0, Basic::Id, Basic::Attributes::HardwareVersion::Id, &val);
```

chiragatal, 2022-10-27:
For `Attributes managed internally`, the read, write, default values, all of them were handled in the upstream code itself in their respective `src/app/clusters/` file. So, we do not store/update their values in the esp-matter database, and let the upstream database handle it.

These attributes are typically read from some other source, example, from nvs, or other components like `ConfigurationMgr`, etc.

jonsmirl, 2022-10-27:
I see now that these are set via menuconfig. 
Component config -> CHIP Device Layer -> Device Identification Options -> ....

Menuconfig is missing the ability to set in the strings for these options such as vendorName, productName, hardwareVersionString, etc...

jonsmirl, 2022-10-27:
The Unique field should retrieve the hardware MAC address.

jonsmirl, 2022-11-11:
Bump -- menuconfig is missing a lot of these static settings.

jonsmirl, 2022-11-23:
So somehow these strings need to end up in the initial flash image.  Looking in the code for CHIP chef I see this..
```
message(STATUS "Product ID " ${CONFIG_DEVICE_PRODUCT_ID})
message(STATUS "Vendor ID " ${CONFIG_DEVICE_VENDOR_ID})
message(STATUS "SW Version String" ${CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING})
idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_PRODUCT_ID=${CONFIG_DEVICE_PRODUCT_ID}" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_VENDOR_ID=${CONFIG_DEVICE_VENDOR_ID}" APPEND)
if(NOT ${CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING} STREQUAL "")
    idf_build_set_property(COMPILE_OPTIONS "-DCHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING=\"${CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING}\"" APPEND)
```

That looks like it is taking the values from the ESP Kconfig and moving them over to CHIP. So something similar needs to be added to esp-matter? And then expanded for the missing attributes?

        attribute::create_data_model_revision(cluster, 0);
        attribute::create_location(cluster, NULL, 0);
        attribute::create_vendor_name(cluster, NULL, 0);
        attribute::create_vendor_id(cluster, 0);
        attribute::create_product_name(cluster, NULL, 0);
        attribute::create_product_id(cluster, 0);
        attribute::create_hardware_version(cluster, 0);
        attribute::create_hardware_version_string(cluster, NULL, 0);
        attribute::create_software_version(cluster, 0);
        attribute::create_software_version_string(cluster, NULL, 0);
        attribute::create_capability_minima(cluster, NULL, 0, 0);

The unique ID attribute is going to need some code in the ESP CHIP implementation, it needs to copy the unique ESP chip ID down into the flash where the CHIP code can find it.

closed 2022-12-14 by jonsmirl