espressif/esp-matter #73: How to add a PartsList to Descriptor cluster on EP

jonsmirl, 2022-10-27:
So I make my generic switch EP...
```
        generic_switch::config_t generic_config;
        endpoint_t *endpoint3 = generic_switch::create(node, &generic_config, ENDPOINT_FLAG_NONE, (void *)i);
        cluster::fixed_label::create(endpoint3, NULL, CLUSTER_FLAG_SERVER);
        cluster::user_label::create(endpoint3, NULL, CLUSTER_FLAG_SERVER);
```
Now I need to add endpoint1 and endpoint2 to the PartsList on the Descriptor cluster of endpoint3.
So how do I so that? This is similar problem to the BasicCluster, the PartsList field is Read-Only.

chiragatal, 2022-10-27:
The last argument for `emberAfSetDynamicEndpoint()`, which is being used in esp-matter to create all the endpoints, is parent_endpoint_id. The parts list is created internally based on this.

Currently in esp-matter, this is just being set to kInvalidEndpointId for all the endpoints. This needs to be fixed and handled correctly in esp-matter. Maybe a separate API to add the parent_endpoint_id would be needed.

I did a quick test and the parts_list is correctly reflected, even for multiple levels in the 'endpoint tree'.

jonsmirl, 2022-10-28:
I will test this and soon as it is available. I need to see what Apple and Google Home are going to do when they see the device partslist.

dhrishi, 2022-11-18:
@jonsmirl Can you please check with latest esp-matter main branch. https://github.com/espressif/esp-matter/commit/8183dd3963f1c3f1a78519f4cba911e5ce71a1a9 is added now

jonsmirl, 2022-12-07:
SetParent should be an endpoint member function, not standalone.
You have: set_parent_endpoint(endpoint3, endpoint1);
Should be  endpoint3->set_parent(endpoint1);

Google and Apple have work to do on this, if I enable this both of their apps GPF. Maybe you can set them up with a test case since they don't listen to me.

A light dimmer switch is a composite device that should look like this:

Light Bulb EP
--child--  Light Switch EP
--child--  Generic Switch EP

In their apps the Light Bulb is the primary thing controlled by the UI. The child Light Switch should be ignored. And the child Generic Switch should expose events that can be used to trigger routines.

jonsmirl, 2023-03-24:
This is working now, but adding the EP to the partlist isn't the solution to the problem I was trying to address. The actually solution is that a new device type for Smart Switches is needed.